arkq/bluez-alsa #79: Setting up bluez-alsa to stream audio to two BT speakers from two dongles

tmp6154, 2018-01-07:
I'm trying to setup sort of a multiroom speaker system using two bluetooth speakers and MPD and I've ran into a problem. Whenever I connect a second speaker, either one starts to disconnect. Searching forums on the topic, I've found a little, but someone said that cheap chinese dongles support at most one A2DP stream, so I suppose I need to use two dongles, one dongle per speaker. Fortunately, I have a pair of those:

Bus 010 Device 011: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
Bus 005 Device 007: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)

I've set up my .asoundrc as follows:

```
pcm.btspeaker {
        type plug
        slave.pcm {
                type bluealsa
                interface "hci0"
                device "FC:58:FA:26:21:03"
                profile "a2dp"
        }
        hint {
                show on
                description "MINISO Bluetooth Speaker"
        }
}

pcm.mispeaker {
        type plug
        slave.pcm {
                type bluealsa
                interface "hci1"
                device "00:9E:C8:F4:39:EC"
                profile "a2dp"
        }
        hint {
                show on
                description "Xiaomi Bluetooth Speaker"
        }
}

ctl.btspeaker {
        type bluealsa
}

ctl.mispeaker {
        type bluealsa
}
```

But unfortunately, problems still persists. I can connect to one BT speaker, but connecting to second one doesn't result in it saying "paired" to indicate a successful connection and attempting to aplay with it results in

```
ALSA lib /tmp/portage/media-sound/bluez-alsa-9999/work/bluez-alsa-9999/src/asound/bluealsa-pcm.c:646:(_snd_pcm_bluealsa_open) BlueALSA connection failed: No such file or directory
aplay: main:786: audio open error: No such file or directory
```

Arkq, 2018-01-07:
Hi,

Do you run two bluealsa servers (one per each HCI)?

    bluealsa -i hci0
    bluealsa -i hci1 --disable-hsp --disable-hfp

I'm not sure if disabling HSP and HFP profiles is required (I've never tested bluez-alsa with more than one HCI), however current implementation (BlueZ implementation) does not allow to specify which HCI should be used with HSP and HFP profile. So, I would opt for disabling these profiles in one of your HCI devices.

Also add `interface "hci0"` and `interface "hci1"` to your `ctl` definitions.

tmp6154, 2018-01-07:
Ah, so that was the main problem. I've splitted services, now with two per-HCI servers I can connect both speakers and aplay works:

`aplay -D mispeaker 1.wav & aplay -D btspeaker 1.wav`

Unfortunately, MPD doesn't want to play from those devices now:

```
ALSA lib /tmp/portage/media-sound/bluez-alsa-9999/work/bluez-alsa-9999/src/asound/bluealsa-pcm.c:659:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA transport: No such device
Jan 07 14:08 : alsa_output: Failed to open "MINISO Bluetooth Speaker" [alsa]: Failed to open ALSA device "btspeaker": No such device
ALSA lib /tmp/portage/media-libs/alsa-lib-1.1.2/work/alsa-lib-1.1.2/src/pcm/pcm.c:2450:(snd_pcm_open_noupdate) Unknown PCM mispeaker
```

I wonder, why it doesn't see them, while aplay can play just fine?

tmp6154, 2018-01-08:
@Arkq I've got a moderate success with MPD by specifying devices explicitly via "bluealsa:" parameters. But I have feeling that I'm doing something the wrong way.

```
audio_output {
        type            "alsa"
        name            "MINISO Bluetooth Speaker"
        device          "bluealsa:HCI=hci1,DEV=FC:58:FA:26:21:03,PROFILE=a2dp"
        mixer_type      "software"
}

audio_output {
        type            "alsa"
        name            "Xiaomi Bluetooth Speaker"
        device          "bluealsa:HCI=hci0,DEV=00:9E:C8:F4:39:EC,PROFILE=a2dp"
        mixer_type      "software"
}
```

I've added mpd user to audio group, but it still cannot see "btspeaker" and "mispeaker" PCMs. Even though they're listed:

```
$ aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
btspeaker
    MINISO Bluetooth Speaker
mispeaker
    Xiaomi Bluetooth Speaker
```

tmp6154, 2018-01-08:
After downgrading bluealsa to 1.2.0, MPD no longer gets stuck. Looks like it's a degradation in git version.

mkriegl, 2018-01-08:
Hi guys,
I got a question to a similar issue: I managed to get bluealsa work with mopidy on one speaker with also a Chinese dongle. Out of your experience, is it possible to get the music from mopidy via one dongle to two or more speakers? So, the same sound to several devices? Or is also one dongle needed per speaker? What does the asound.conf look like?
Otherwise, is there a more easy way to switch between MAC addresses? 
Thanks in advance .. Max

Arkq, 2018-01-08:
@tmp6154 Hmm... does playback jams when using aplay as well? If so, please run aplay with gdb, eg. `gdb --args aplay -D mispeaker 1.wav`. When the jam occurs, break execution (Ctrl-C) and switch to pcm-io thread (`info threads` will give you thread listing, then use `thread <NB>` to switch to desirable thread), and produce backtrace, then post it here.

I've got some other reports about jamming, however I'm not able to reproduce it...

@mkriegl 
Yes, it should be possible to stream music to multiple devices. Bare in mind, though, that BT link throughput is limited. If both devices uses SBC as a codec, then there is a great chance, that the audio will be chopped in both speakers and there is no workaround for it except using different (more efficient) codec.

tmp6154, 2018-01-08:
@Arkq No, with aplay I was not able to reproduce it, even with a longer track (about 2 minutes long). With MPD, it jams in 20-30 seconds and reproduces stably for me and only if there's two active speakers.

What's interesting, sometimes, there are short bursts of sound (less than a second long) between the silence, i.e. it's not a complete deadlock. Looks more like a very slow consumption of buffer, and MPD probably jams until buffer is consumed on all outputs, including bluealsa ones.

tmp6154, 2018-01-08:
@Arkq Caught MPD hang with 1.2.0 bluealsa. It doesn't respond to mpc, all commands time out. Most of MPD's threads wait on rwlock. mispeaker seems to wait on sem_close()

```
(gdb) info threads
  Id   Target Id         Frame 
* 1    LWP 14788 "mpd"   0x00007fb13266e466 in pthread_rwlock_timedwrlock () from /lib64/libpthread.so.0
  2    LWP 14789 "io"    0x00007fb1323a62e3 in ?? () from /lib64/libc.so.6
  3    LWP 14790 "player" 0x00007fb13266e466 in pthread_rwlock_timedwrlock () from /lib64/libpthread.so.0
  4    LWP 14791 "decoder:mad" 0x00007fb13266e466 in pthread_rwlock_timedwrlock () from /lib64/libpthread.so.0
  5    LWP 14792 "output:My ALSA " 0x00007fb13266e466 in pthread_rwlock_timedwrlock () from /lib64/libpthread.so.0
  6    LWP 14793 "output:MINISO B" 0x00007fb13266e466 in pthread_rwlock_timedwrlock () from /lib64/libpthread.so.0
  7    LWP 14794 "output:Xiaomi B" 0x00007fb13267238c in sem_close () from /lib64/libpthread.so.0
  8    LWP 14795 "output:UBX Home" 0x00007fb13266e466 in pthread_rwlock_timedwrlock () from /lib64/libpthread.so.0
```

Unfortunately, most of backtraces are useless (looks like debug symbols for MPD are missing). Backtrace for mispeaker output:

```
(gdb) bt
#0  0x00007fb13267238c in sem_close () from /lib64/libpthread.so.0
#1  0x000ff30400000000 in ?? ()
#2  0x0000001000090100 in ?? ()
#3  0x000000010000ac44 in ?? ()
#4  0x0000000000000000 in ?? ()
```

Probably, I'll have to recompile MPD with debug symbols.

tmp6154, 2018-01-08:
@Arkq Ah, yes! I seem to recall one strange thing with aplay: it didn't hang during playback, it hangs sometimes after sending Ctrl+C to it. Hangs completely, to the point, where SIGTERM can do nothing, process can only be killed. That also can explain 1.2.0 hang with MPD, because it's occured during/after stop, it seems. However, in git version, it's probably a different issue that happens during playback.

Arkq, 2018-01-08:
Yeap, debug symbols might be helpful :) However, something seems to be not right. If MPD uses bluealsa PCM plugin, there should have been (at least one) thread with the `pcm-io` name, and I can see none...

As for bluez-alsa-1.2.0, there has been a lot of changes. I would prefer to see backtrace from the current master snapshot, of course if it's not a problem for you.

tmp6154, 2018-01-08:
Yes, I've already has rebuilt MPD without stripping debug symbols, and will get git master snapshot of bluez-alsa. Fortunately, under Gentoo it's just a matter of re-emerging package with another keyword.

Yes, I've looked into commit history, there was indeed a lot of changes. I've even thought it might be useful to bisect bluez-alsa, if problem would be not clear from backtraces.

Yes, thread-safe API is most likely the cause. I didn't disable it, and looks like there's no use flag or something to disable it in Gentoo. Probably will need to edit ebuild.

Arkq, 2018-01-08:
> I didn't disable it, and looks like there's no use flag or something to disable it in Gentoo. Probably will need to edit ebuild.

There is no need for that. Just export `LIBASOUND_THREAD_SAFE=0` before running MPD.

tmp6154, 2018-01-08:
@Arkq Here's better results:

```
(gdb) info threads
  Id   Target Id         Frame 
* 1    Thread 0x7f7ab3321740 (LWP 32402) "mpd" 0x00007f7aad566303 in epoll_wait () from /lib64/libc.so.6
  2    Thread 0x7f7aa8817700 (LWP 32403) "io" 0x00007f7aad566303 in epoll_wait () from /lib64/libc.so.6
  3    Thread 0x7f7aa8016700 (LWP 32404) "player" 0x00007f7aad82e6b6 in pthread_cond_wait@@GLIBC_2.3.2 () from /lib64/libpthread.so.0
  4    Thread 0x7f7aa7815700 (LWP 32405) "decoder:mad" 0x00007f7aad82e6b6 in pthread_cond_wait@@GLIBC_2.3.2 () from /lib64/libpthread.so.0
  5    Thread 0x7f7aa6c04700 (LWP 32406) "output:My ALSA " 0x00007f7aad82e6b6 in pthread_cond_wait@@GLIBC_2.3.2 () from /lib64/libpthread.so.0
  6    Thread 0x7f7aa6403700 (LWP 32407) "output:MINISO B" 0x00007f7aad82e6b6 in pthread_cond_wait@@GLIBC_2.3.2 () from /lib64/libpthread.so.0
  7    Thread 0x7f7aa5c02700 (LWP 32408) "output:Xiaomi B" 0x00007f7aad55abad in poll () from /lib64/libc.so.6
  8    Thread 0x7f7aa5401700 (LWP 32409) "output:UBX Home" 0x00007f7aad82e6b6 in pthread_cond_wait@@GLIBC_2.3.2 () from /lib64/libpthread.so.0
  9    Thread 0x7f7a87dd7700 (LWP 2263) "pcm-io" 0x00007f7aad83279d in write () from /lib64/libpthread.so.0
  10   Thread 0x7f7aa4c00700 (LWP 2264) "pcm-io" 0x00007f7aad8334ef in do_sigwait () from /lib64/libpthread.so.0
(gdb) thread 9
[Switching to thread 9 (Thread 0x7f7a87dd7700 (LWP 2263))]
#0  0x00007f7aad83279d in write () from /lib64/libpthread.so.0
(gdb) bt
#0  0x00007f7aad83279d in write () from /lib64/libpthread.so.0
#1  0x00007f7aa41fb4c9 in io_thread () from /usr/lib64/alsa-lib/libasound_module_pcm_bluealsa.so
#2  0x00007f7aad827867 in start_thread () from /lib64/libpthread.so.0
#3  0x00007f7aad565cdf in clone () from /lib64/libc.so.6
(gdb) thread 10
[Switching to thread 10 (Thread 0x7f7aa4c00700 (LWP 2264))]
#0  0x00007f7aad8334ef in do_sigwait () from /lib64/libpthread.so.0
(gdb) bt
#0  0x00007f7aad8334ef in do_sigwait () from /lib64/libpthread.so.0
#1  0x00007f7aad8335ae in sigwait () from /lib64/libpthread.so.0
#2  0x00007f7aa41fb3b7 in io_thread () from /usr/lib64/alsa-lib/libasound_module_pcm_bluealsa.so
#3  0x00007f7aad827867 in start_thread () from /lib64/libpthread.so.0
#4  0x00007f7aad565cdf in clone () from /lib64/libc.so.6
```

It's git master of bluealsa, and same kind of deadlock that happens only on git version, i.e. not in ALSA. Looks like write() call in 9 thread got stuck?

Arkq, 2018-01-08:
OK, that's something... however it's a pity there are no line numbers.

Could you join bluealsa@freenode? It will be easier to talk, if you've got some time. For me it's still 2-3 hours before sleep :)

tmp6154, 2018-01-10:
I've performed a longer test with only setsockopt/fcntl line. I was afraid that 2 days ago, while discussing on IRC I didn't give it enough time to jam. Result is same, though. Keeping only one line resolves jam for me.

Will perform the test with MTU multiplier now.

Arkq, 2018-01-10:
> Keeping only one line resolves jam for me.

I think that it is reasonable. Because when you remove setsockopt() call, the socket buffer is big enough to prevent jamming, while removing fcntl() resolves jamming, so the size of the buffer doesn't matter. So, that's why I'd like to see a test with a bigger socket buffer.

I was trying to localize the place in the BlueZ where the other end of the socket is (in order to check the blocking logic), however it seems, that the other end of the socket is in the kernel space... I will try to create a fix for it in a next few days (I'm afraid, that the refactoring will be rather extensive). I will let you know, and I hope you will be able to test it with a positive result :)

tmp6154, 2018-01-10:
Multiplier test, fcntl() present:

1 - Jammed after 2:30. Just finished playing 2:29 track, played, started playing next and hanged next second.
2 - Jammed after 3:29. Again, it has hanged right after track has changed and played 1 second of new track.
3 - Played for 2:40. First track played, and second played for 12 seconds, then jammed.
4 - Played 3 tracks, couldn't reproduce jam.
5 - Played 3 tracks, couldn't reproduce jam.

So, basically, there's no direct correlation between time-to-jam and buffer size. But instead, it looks like this time it was more dependent on song switches (it wasn't triggered by song switches and occured faster on non-debug ebuild, not sure why this has changed).

Also, 4 seems to be a barrier of multiplier: starting from it and going higher seems to solve the problem.

Arkq, 2018-01-10:
@tmp6154 many thanks for this test. Right now I'm almost certain, that my deduction is right :)

tmp6154, 2018-01-10:
Tried removing fcntl() and setting multiplier to 1. Looks like you're right. It's a combination of both small buffer and blocking I/O.

Arkq, 2018-01-11:
@tmp6154 please, check if branch [fcntl](https://github.com/Arkq/bluez-alsa/tree/fcntl) resolves this issue for you. I'm going to test it as well, so if there are no problems, this branch will become master in the next week.

tmp6154, 2018-01-11:
Installed bluealsa from fcntl branch. Will report if I encounter any problems with this version.

tmp6154, 2018-01-12:
So far so good, playback is fine, without artifacts and jams.

closed 2018-01-14 by Arkq

vinaykumar2491, 2018-12-18:
> Ah, so that was the main problem. I've splitted services, now with two per-HCI servers I can connect both speakers and aplay works:

How did you split the services. I'm stuck in the error:
```
ALSA lib ../../../src/asound/bluealsa-pcm.c:665:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA transport: No such device
aplay: main:788: audio open error: No such device
```

tmp6154, 2018-12-18:
Since I'm on Gentoo, I've copied /etc/init.d/bluealsa service into /etc/init.d/bluealsatwo. I've changed pidfile, so they wouldn't conflict:

pidfile="/var/run/bluealsatwo.pid"

Next, I've copied /etc/conf.d/bluealsa to /etc/conf.d/bluealsatwo and made a different config. In original file, I used hci0 (first dongle):

BLUEALSA_CONF="-i hci0 -S --profile=a2dp-source"

In copied file, I've used hci1 (second dongle):

BLUEALSA_CONF="-i hci1 -S --profile=a2dp-source"

Then I've enabled and started service.

tmp6154, 2018-12-18:
Also, besides bluealsa, I've configured ALSA like this:

`# Specify the parameters of the Bluetooth connection
pcm.speakerone {
        type plug
        slave.pcm {
                type bluealsa
                interface "hci0"
                device "<Speaker 1 MAC goes there>"
                profile "a2dp"
        }
        hint {
                show on
                description "description - speaker one"
        }
}

pcm.speakertwo {
        type plug
        slave.pcm {
                type bluealsa
                interface "hci1"
                device "<Speaker 1 MAC goes there>"
                profile "a2dp"
        }
        hint {
                show on
                description "description - speaker two"
        }
}

ctl.speakerone {
        type bluealsa
        interface "hci0"
}

ctl.speakertwo {
        type bluealsa
        interface "hci1"
}`

tmp6154, 2018-12-18:
Finally, if you want to use both speakers from MPD, add sections like that

```
audio_output {
        type            "alsa"
        name            "speaker 1"
        device          "bluealsa:HCI=hci0,DEV=<speaker 1 MAC>,PROFILE=a2dp"
        mixer_type      "software"
}

audio_output {
        type            "alsa"
        name            "speaker 2"
        device          "bluealsa:HCI=hci1,DEV=<speaker 2 MAC>,PROFILE=a2dp"
        mixer_type      "software"
}
```

commit 6879322f (2018-01-14): Restore non-blocking mode for BT socket

It seems that commit 79f68a9 has introduced a very peculiar bug. In
that commit I've changed socket mode from non-blocking to blocking
and decreased socket buffer. It has turned out, that changing BT
socket mode to blocking causes some kind of a "dead-lock" during
writing to this socket...

Fixes #79