arkq/bluez-alsa #80: Blualsa support with OpenWRT

PraneetKaur, 2018-01-11:
Hi,
I would like to know if bluealsa is supported on OpenWRT platform. I am trying to get audio streaming with bluez5.37 using A2DP. I want to enable both A2DP source and A2DP sink for audio streaming purpose.

Thanks

diizzyy, 2018-05-05:
https://github.com/openwrt/packages/pull/6010 might be of interest

Rixerx, 2018-12-19:
@Arkq Arek it is any chance of taking maintenance by You of bluez-alsa package in OpenWrt repo?

raphikWasHere, 2020-06-19:
[This](https://foro.seguridadwireless.net/openwrt/der-router-als-bluetooth-lautsprecher/ ) might be of service.

Arkq, 2021-01-16:
Since there is an example of how to use bluez-alsa with OpenWRT, I'm closing this issue, because maintaining distribution packaging is of of the bluez-alsa issues scope.

closed 2021-01-16 by Arkq