arkq/bluez-alsa #82: Bluetooth headphones - aplay "unable to install hw params"

mattdenner, 2018-01-12:
I've been trying to get bluez-alsa to work, on a NixOS machine, with my bluetooth headphones for a few days now with little success.

I've had to build from a specific sha (`99d9d72a5aed22dd755485a2661e209ded697079`) because after this point the build tries to copy files into system locations on install and these paths are protected under NixOS.  To get round that I have added a couple of additional settings to my `~/.asoundrc` as follows:

```
pcm_type.bluealsa {
  lib /nix/store/bhpr800zf8bsphqg37ri0c8hjjvyhj90-bluez-alsa-1.2.0/lib/alsa-lib/libasound_module_pcm_bluealsa.so
}
ctl_type.bluealsa {
  lib /nix/store/bhpr800zf8bsphqg37ri0c8hjjvyhj90-bluez-alsa-1.2.0/lib/alsa-lib/libasound_module_ctl_bluealsa.so
}

defaults.bluealsa.interface "hci0"
defaults.bluealsa.profile "a2dp"
defaults.bluealsa.delay 20000

ctl.headphones {
  type bluealsa
  interface "hci0"
}

pcm.headphones {
  type plug
  slave {
    pcm {
      type bluealsa
      interface "hci0"
      device "F0:85:C1:B5:9D:CD"
      profile "a2dp"
      delay 20000
    }
  }
  hint {
    show on
    description "Bluetooth headphones"
  }
}
```

I start the blue-alsa service with `systemctl start blue-alsa` and see:

```
Jan 12 12:56:19 nixos bluealsa[21228]: /nix/store/yp52q0f7sj987dqlxrc96q82bs96njqp-bluez-alsa-1.2.0/bin/bluealsa: ctl.c:493: Starting controller loop
Jan 12 12:56:19 nixos bluealsa[21228]: /nix/store/yp52q0f7sj987dqlxrc96q82bs96njqp-bluez-alsa-1.2.0/bin/bluealsa: bluez.c:680: Registering endpoint: /A2DP/SBC/Source/1
Jan 12 12:56:19 nixos bluetoothd[717]: Endpoint registered: sender=:1.211 path=/A2DP/SBC/Source/1
Jan 12 12:56:19 nixos bluealsa[21228]: /nix/store/yp52q0f7sj987dqlxrc96q82bs96njqp-bluez-alsa-1.2.0/bin/bluealsa: bluez.c:680: Registering endpoint: /A2DP/SBC/Sink/1
Jan 12 12:56:19 nixos bluetoothd[717]: Endpoint registered: sender=:1.211 path=/A2DP/SBC/Sink/1
Jan 12 12:56:19 nixos bluealsa[21228]: /nix/store/yp52q0f7sj987dqlxrc96q82bs96njqp-bluez-alsa-1.2.0/bin/bluealsa: bluez.c:910: Registering profile: /HSP/Headset
Jan 12 12:56:19 nixos bluealsa[21228]: /nix/store/yp52q0f7sj987dqlxrc96q82bs96njqp-bluez-alsa-1.2.0/bin/bluealsa: bluez.c:910: Registering profile: /HSP/AudioGateway
Jan 12 12:56:19 nixos bluealsa[21228]: /nix/store/yp52q0f7sj987dqlxrc96q82bs96njqp-bluez-alsa-1.2.0/bin/bluealsa: bluez.c:910: Registering profile: /HFP/HandsFree
Jan 12 12:56:19 nixos bluealsa[21228]: /nix/store/yp52q0f7sj987dqlxrc96q82bs96njqp-bluez-alsa-1.2.0/bin/bluealsa: bluez.c:910: Registering profile: /HFP/AudioGateway
Jan 12 12:56:19 nixos bluealsa[21228]: /nix/store/yp52q0f7sj987dqlxrc96q82bs96njqp-bluez-alsa-1.2.0/bin/bluealsa: main.c:225: Starting main dispatching loop
```

I then use bluetoothctl to connect to my headphones and see:

```
Jan 12 12:58:12 nixos bluealsa[21228]: /nix/store/yp52q0f7sj987dqlxrc96q82bs96njqp-bluez-alsa-1.2.0/bin/bluealsa: bluez.c:857: Profile method call: org.bluez.Profile1.NewConnection()
Jan 12 12:58:12 nixos bluealsa[21228]: /nix/store/yp52q0f7sj987dqlxrc96q82bs96njqp-bluez-alsa-1.2.0/bin/bluealsa: transport.c:668: State transition: 0 -> 2
Jan 12 12:58:12 nixos bluealsa[21228]: /nix/store/yp52q0f7sj987dqlxrc96q82bs96njqp-bluez-alsa-1.2.0/bin/bluealsa: io.c:1184: Starting IO loop: HFP Audio Gateway
Jan 12 12:58:12 nixos bluealsa[21228]: /nix/store/yp52q0f7sj987dqlxrc96q82bs96njqp-bluez-alsa-1.2.0/bin/bluealsa: transport.c:104: Created new IO thread: HFP Audio Gateway
Jan 12 12:58:12 nixos bluealsa[21228]: /nix/store/yp52q0f7sj987dqlxrc96q82bs96njqp-bluez-alsa-1.2.0/bin/bluealsa: bluez.c:797: HFP Audio Gateway configured for device F0:85:C1:B5:9D:CD
Jan 12 12:58:12 nixos bluealsa[21228]: /nix/store/yp52q0f7sj987dqlxrc96q82bs96njqp-bluez-alsa-1.2.0/bin/bluealsa: ctl.c:575: +-+-
Jan 12 12:58:12 nixos bluealsa[21228]: /nix/store/yp52q0f7sj987dqlxrc96q82bs96njqp-bluez-alsa-1.2.0/bin/bluealsa: transport.c:668: State transition: 0 -> 2
Jan 12 12:58:12 nixos bluealsa[21228]: /nix/store/yp52q0f7sj987dqlxrc96q82bs96njqp-bluez-alsa-1.2.0/bin/bluealsa: rfcomm.c:553: Starting RFCOMM loop: HFP Audio Gateway
Jan 12 12:58:12 nixos bluealsa[21228]: /nix/store/yp52q0f7sj987dqlxrc96q82bs96njqp-bluez-alsa-1.2.0/bin/bluealsa: transport.c:104: Created new IO thread: HFP Audio Gateway
Jan 12 12:58:12 nixos bluealsa[21228]: /nix/store/yp52q0f7sj987dqlxrc96q82bs96njqp-bluez-alsa-1.2.0/bin/bluealsa: at.c:156: AT message: SET: command:+BRSF, value:63
Jan 12 12:58:12 nixos bluealsa[21228]: /nix/store/yp52q0f7sj987dqlxrc96q82bs96njqp-bluez-alsa-1.2.0/bin/bluealsa: rfcomm.c:95: Sending AT message: RESP: command:+BRSF, value:992
Jan 12 12:58:12 nixos bluealsa[21228]: /nix/store/yp52q0f7sj987dqlxrc96q82bs96njqp-bluez-alsa-1.2.0/bin/bluealsa: rfcomm.c:95: Sending AT message: RESP: command:(null), value:OK
Jan 12 12:58:12 nixos bluealsa[21228]: /nix/store/yp52q0f7sj987dqlxrc96q82bs96njqp-bluez-alsa-1.2.0/bin/bluealsa: rfcomm.c:113: HFP state transition: 0 -> 2
[... 53 lines cut ...]
CHANNELS: 2
RATE: 44100
PERIOD_TIME: (92879 92880)
PERIOD_SIZE: 4096
PERIOD_BYTES: 16384
PERIODS: 5
BUFFER_TIME: (464399 464400)
BUFFER_SIZE: 20480
BUFFER_BYTES: 81920
TICK_TIME: 0
```

And the journal says:

```
Jan 12 12:59:42 nixos bluealsa[21228]: /nix/store/yp52q0f7sj987dqlxrc96q82bs96njqp-bluez-alsa-1.2.0/bin/bluealsa: ctl.c:554: New client accepted: 13
Jan 12 12:59:42 nixos bluealsa[21228]: /nix/store/yp52q0f7sj987dqlxrc96q82bs96njqp-bluez-alsa-1.2.0/bin/bluealsa: ctl.c:575: +-+-
Jan 12 12:59:42 nixos bluealsa[21228]: /nix/store/yp52q0f7sj987dqlxrc96q82bs96njqp-bluez-alsa-1.2.0/bin/bluealsa: ctl.c:575: +-+-
Jan 12 12:59:42 nixos bluealsa[21228]: /nix/store/yp52q0f7sj987dqlxrc96q82bs96njqp-bluez-alsa-1.2.0/bin/bluealsa: ctl.c:527: Invalid request length: -1 != 20
Jan 12 12:59:42 nixos bluealsa[21228]: /nix/store/yp52q0f7sj987dqlxrc96q82bs96njqp-bluez-alsa-1.2.0/bin/bluealsa: ctl.c:575: +-+-
```

I've also tried `--disable-hfp` when starting the bluealsa service with no change.

I'm sure it's something stupid I'm either doing or not but it would be great to get this working.  Any ideas?

Arkq, 2018-01-12:
Hi,

> I've had to build from a specific sha 99d9d72a5aed22dd755485a2661e209ded697079 because after this point the build tries to copy files into system locations on install and these paths are protected under NixOS.

Does, `pkg-config --variable=libdir alsa` outputs the right place where ALSA plugins are stored? I don't know how NixOS works, however if this directory is somehow protected (assuming it is correct), I'm afraid there is no fix for this. You can specify `--with-alsaplugindir=PATH` during configuration, though. Type `./configure --help` for all options.

> And then I try aplay -D headphones foo.wav
>     Jan 12 12:59:42 [...] in/bluealsa: ctl.c:527: Invalid request length: -1 != 20

Ups... There was an error during IPC communication. And it seems, that I'm not checking for errors. Please, add `debug("ERROR: %d %s", errno, strerror(errno));` below the line 527 in the `src/ctl.c` file and post the results here (only this extra debug message).

mattdenner, 2018-01-27:
Sorry it’s taken a while to get back to you but I managed to get the latest code to compile & install on NixOS. My headphones connect and work with aplay. When I use `alsa_in` from jack I have issues with the quality of the audio and see messages about drift (not at my computer at the moment so don’t have exact message but will get it). 

So a good couple of steps forward with a little one backwards.

borine, 2021-06-03:
It seems the initial issue reported in the subject line is resolved. `alsa_in` and other jack applications have some issues with non-hardware devices such as bluealsa. Please feel free to open a new issue if you are still experiencing issues with jack when using the latest bluealsa code.

closed 2021-06-03 by borine