arkq/bluez-alsa #84: No termination when phone disconnected

georgefst, 2018-02-04:
I believe bluealsa-aplay is meant to terminate after the device is disconnected (if streaming has occurred).

This worked for me when I last tried a few months ago, but no longer works. As a result, the process sometimes needs to be manually killed in order to release the audio card.

This is somewhat frustrating as I'm trying to get to a headless setup on my RPi - I'd rather not have to ssh in and kill the process from time to time. (Out of curiosity, does bluealsa-aplay use any significant amount of power if the device isn't connected, or is it virtually idle?)

Arkq, 2018-02-04:
> I believe bluealsa-aplay is meant to terminate after the device is disconnected (if streaming has occurred)

No, it is not. It should work without termination. However, it will terminate if one kills bluealsa server. One can run e.g.: `bluealsa-aplay 00:00:00:00:00:00` and Linux powered device should change into the bluetooth speaker (assuming that BlueZ is configured in a way that allows connection without human intervention - trusting device/entering PIN code).

> This worked for me when I last tried a few months ago, but no longer works. As a result, the process sometimes needs to be manually killed in order to release the audio card.

Can you post here how you are using `bluealsa-aplay`?

> Out of curiosity, does bluealsa-aplay use any significant amount of power if the device isn't connected, or is it virtually idle?

Power consumption should be almost zero. There is a negligence amount of work done by kernel when `bluealsa-aplay` is running - governing poll() call :). When I'm incorporating new features into the bluez-alsa my number one priority is performance - sometimes it might cause side-effects (I haven't seen any, though), e.g. I'm trying no to use mutexes for thread synchronization :)

georgefst, 2018-02-05:
> No, it is not.

Ok. Ignore me. I must have got confused with something else.

> Can you post here how you are using bluealsa-aplay?

I'm using a RPi 3 (with a HifiBerry DAC+). I was previously ssh-ing into the pi to connect the bluetooth and start bluealsa-aplay each time I wanted to play music. But now I've realised that by running bluealsa-aplay with 'nohup' and trusting the bluetooth device through bluetoothctl I can avoid doing any setup - I just turn bluetooth on on my phone and play.

Thanks for replying, been using as above for the past few days with no issues. Haven't tested power consumption but I'll take your word that it's low. :)

Arkq, 2018-02-05:
> But now I've realised that by running bluealsa-aplay with 'nohup' and trusting the bluetooth device through bluetoothctl I can avoid doing any setup - I just turn bluetooth on on my phone and play.

Yeap, that's the idea behind `bluealsa-aplay`. I've started it (`bluealsa-aplay`) as a proof-of-concept player, however due to submitted issues, right now it is a solid (I hope) BT player :).

> Haven't tested power consumption but I'll take your word that it's low. :)

To be honest, I don't know how to make it any lower :D without extreme profiling and inlined assembler code blocks (I'm talking about `bluealsa-aplay`, because in the `bluealsa` code there is a room for some optimizations, e.g. [poll() with timeout](https://github.com/Arkq/bluez-alsa/blob/5a13bdb2baf0652a061ff0314f72b0f774aaf88e/src/io.c#L1240) :/).

> been using as above for the past few days with no issues

So, I'm assuming, that this issue is resolved, therefore I'm closing it.

closed 2018-02-05 by Arkq

nicokaiser, 2018-02-05:
You can start `bluealsa-aplay` as system service with systemd:

/etc/systemd/system/bluealsa-aplay.service:
```
[Unit]
Description=BlueALSA player
After=bluealsa.service

[Service]
Type=simple
User=root
ExecStartPre=/bin/sleep 2
ExecStart=/usr/bin/bluealsa-aplay 00:00:00:00:00:00
Restart=on-failure

[Install]
WantedBy=graphical.target
```

The `sleep` was necessary, because the `After` seems to not really work with the Raspbian supplied `bluealsa.service`. Maybe someone with better systemd skills can improve this ;-)