arkq/bluez-alsa #86: Bluealsa and bluealsa-aplay merge with convenience switching

George-ou812, 2018-02-19:
Hi,

I've been using bluealsa with bluealsa-aplay for some time now and in all works well. I would like your advice on two things, 1) Currently I've hooked into transport.c to spin up bluealsa-aplay automatically on transport construction and freeing; passing the related deviceaddr to bluealsa-aplay accordingly all works fine.
So.. Any thoughts or advice on the best way to integrate the bluealsa-aplay code into bluealsa:transport.c therefore making things more seamless and less clunky?

2) The main reason for this post is I'm in desperate need to convenience switch the audio hardware. So one music render on board will pause bluealsa ( if playing ), grab the output hardware ( hw:0.0 ) and play. Likewise starting music on bluealsa will pause the other app, grab the card and output via bluealsa-aplay as per normal. Command line wise, in a very primitive way I can do this pausing both ways. BT audio I can do via a dbus-send  MediaPlayer1.Play/Pause. This releases the audio hardware from bluealsa/aplay after a few seconds so the other app can grab it and go. Regarding the other music application I can send a simple SIGUSR pausing the app/stream, grabbing the audio hardware for BT.
Based on keeping the change over signalling simple, if I use another SIGUSR for bluealsa / bluealsa-aplay to pause it and teardown its sound hardware where would be the optimal place to add this hook?

Thanks.

George-ou812, 2018-02-20:
Hi,

If you have any advice great but otherwise this "issue" probably could be closed. I've amalgamated bluealsa--aplay into bluealsa via its own thread and it works really well. So bluealsa now can play audio from any connected device e.g mob, ipad etc straight to the hardware.
Just have to sort my convenience switching which now won't be too bad....
Cheers.

Arkq, 2018-02-20:
Hi,

> So.. Any thoughts or advice on the best way to integrate the bluealsa-aplay code into bluealsa:transport.c therefore making things more seamless and less clunky?

To be honest, I have never thought about incorporating bluealsa-aplay into bluealsa server and probably never will. It doesn't seem right (maybe because I'm a true believer in Unix philosophy of doing one thing and doing it right :D).

> The main reason for this post is I'm in desperate need to convenience switch the audio hardware.

Hmmm... IMHO, to do things right, you need another application, which will do mixing for you. Of course, you can develop "this mixer" in the bluealsa server codebase, but if you really need writing one application inside another one, why not to write it as a kernel module, then you can simply run Linux kernel and voila (trolling mode off :D).

> [...] where would be the optimal place to add this hook?

Really, I don't know. But if you really need custom version of bluez-alsa, you can try to put it into the IO thread. Register SIGUSR handler in the bluealsa, and when it hits you might change (^=) "stop" flag and send event to all transports. Then, in the IO thread (in the t->event_fd dispatcher) you might check if stop/pause flag is set and change pcm_fd to -1, so poll() will block. Then when SIGUSR will hit again, you will restore pcm_fd = pcm->fd ... something along those lines.

Arkq, 2018-02-20:
> If you have any advice great but otherwise this "issue" probably could be closed.

OK, I'm closing it as "WON'T FIX".

closed 2018-02-20 by Arkq

George-ou812, 2018-02-20:
Thanks. Yes it is a bit proprietary, but a headless stand-alone music renderer that lives its life in a box! 
Last question, for a general stop of pause internal to bluealsa, where's a good place?

Arkq, 2018-02-20:
> Last question, for a general stop of pause internal to bluealsa, where's a good place?

You could check for a global "pause" variable in here (in every sink IO thread):
https://github.com/Arkq/bluez-alsa/blob/3e600b874d4424e01fe3d803a993f4982f8f33df/src/io.c#L205

George-ou812, 2018-02-20:
I need to do the pause linux side, e.g. the equivalent of this: dbus-send --system --print-reply --dest=org.bluez /org/bluez/hci0/dev_A4_E9_75_CD_87_85/player2 org.bluez.MediaPlayer1.Pause

Doing this while bluealsa is playing stops/pauses the iPad and stops the transport within bluealsa-aplay exactly what I want; however I'd like to "send" the pause/stop to bluez from within bluealsa/bluealsa-aplay. This pause signal will come from the other app not the iPad. Could I do this using g_bus commands similar to what's in bluez.c? Questions is what calls do I need and where?

Arkq, 2018-02-20:
> Could I do this using g_bus commands [...]?

Yes. Look ad `aplay.c` file. I'm already using `org.bluez.MediaPlayer1.Pause()` D-Bus call.

> Questions is what calls do I need and where?

Wherever your pausing logic is. I think, that `g_dbus_connection_send_message_with_reply_sync()` is thread-safe, anyway you can check it in the glibc documentation.

George-ou812, 2018-02-20:
Hmm. not in my /utils/aplay,c Just went an grabbed the lasted zip too. If you could send me the function or a link to a file I can take it from there.

Thanks for all you help.

Arkq, 2018-02-20:
There: https://github.com/Arkq/bluez-alsa/blob/3e600b874d4424e01fe3d803a993f4982f8f33df/utils/aplay.c#L187

PS.

> grabbed the lasted zip

LOL Isn't it easier to clone git repo? :D Then you can just pull changes from the remote (github) server. Just sayin'... :)

George-ou812, 2018-02-20:
git repo, true, bit lazy in my old age.. All sorted now! Thanks again.