arkq/bluez-alsa #87: Small consultation need please

Pro-pra, 2018-02-20:
Dear Arkq, maybe you know how to create a serial connection (make /dev/rfcomm0 port) without using rfcomm, using only dbus command. (i use dbus-send from my scripts).

Bluez5 code examples are written on a python, I can not understand them.

Arkq, 2018-02-20:
It seems, there is a `rfcomm` utility, which should be used for managing RFCOMM. Unfortunately, I've got no such a program installed on my box... Maybe, BlueZ needs to be configured somehow specially during compilation...

infirit, 2018-02-20:
We have some c code to create and release rfcomm devices. See https://github.com/blueman-project/blueman/blob/05eea0db0cacf888629d7d23ca62c73d663e98f2/module/libblueman.c#L412

Arkq, 2018-02-20:
Thx @infirit for this code. And what are the benefits of creating `/dev/rfcomm0`? Maybe bluealsa should create it after HFP/HSP connection... what do you think?

Pro-pra, 2018-02-21:
Yes, I know about the utility rfcomm, but now it is deprecated. Now i use rfcomm in my bash script.

I want to learn how to make the interface through Dbus, in the future it will be necessary to create interfaces for Ethernet interface. Make Com device is example for other work with bluez5 via Dbus.

Pro-pra, 2018-04-25:
i send message to linux-bluetooth mailinglist

closed 2019-01-11 by Pro-pra