arkq/bluez-alsa #89: Please explain a2dp_volume

RJPoelstra, 2018-02-28:
Hi,

I can use BlueAlsa just fine and volume controls works. I noticed a --a2dp-volume options though and I wonder what it does? Can you please provide some documentation/explanation about this option?

Thanks!

Kind regards,

Remco Poelstra

Arkq, 2018-03-02:
See this commit message: d567b84

With this options (if you are lucky), volume will be controlled by the headset. However, to be honest, I don't really know who should control volume.... source or sink. This feature (and few other) is not documented because it does not work properly.

RJPoelstra, 2018-03-02:
So if I understand it correctly it has no use when I'm using BlueAlsa as the 'headset'?

________________________________

Met vriendelijke groet / With kind regards,
Remco Poelstra
Product Developer Electronics
.
Vogel’s Products BV
Hondsruglaan 93 | 
5628 DB
Eindhoven | 
The Netherlands
vogels.com
Get in touch with Vogel's:
youtube.com/VogelsExperience
© Vogel's Products BV 2017
 | 
facebook.com/VogelsExperience | 
linkedin.com/company/vogels

Arkq, 2018-03-02:
Yes, you are correct. Controlling volume from connected A2DP source (e.g. smartphone) does not require this option, because there is no alternative way of controlling volume. If you are able to control volume from smartphone everything is OK, and if you are not, there is nothing I can do.... But with other devices order (bluealsa acts as a A2DP source), if native volume control does not work (with this `--a2dp-volume` flag), I'm able to control volume within bluealsa. Since, native control is not reliable, the default is a software volume control (internally by bluealsa).

Arkq, 2018-08-01:
I'm closing this issue, since it is not an issue per se.

closed 2018-08-01 by Arkq