arkq/bluez-alsa #91: ALSA module installation path issue when cross-compiling

joerg-krause, 2018-03-02:
I am preparing a bluez-alsa package for [Buildroot](https://buildroot.org) and I came across the ALSA module installation path issue. The issue was already noticed in https://github.com/Arkq/bluez-alsa/issues/34#issuecomment-283623237 and https://github.com/Arkq/bluez-alsa/issues/82.

The problem when building bluez-alsa in a cross-compilation context is that pkg-config prepends the sysroot path to all the paths it returns. For example `pkg-config --variable=libdir alsa` returns `buildroot/output/host/bin/../arm-buildroot-linux-musleabihf/sysroot/usr/lib` in Buildroot. Using the sysroot path for the ALSA modules the installation path created by automake in this case is $(DESTDIR)/$(SYSROOT), which is wrong.

These are the corresponding parts of the code.

configure.ac
```
alsaplugindir=$($PKG_CONFIG --variable=libdir alsa)/alsa-lib

AC_SUBST([ALSA_DATA_DIR], [$alsadatadir])
AC_SUBST([ALSA_PLUGIN_DIR], [$alsaplugindir])
```

Makefile.in:
```
asound_module_ctldir = @ALSA_PLUGIN_DIR@
asound_module_pcmdir = @ALSA_PLUGIN_DIR@
asound_module_datadir = @ALSA_DATA_DIR@/alsa.conf.d
```

Makefile.in:
```
am__installdirs = "$(DESTDIR)$(asound_module_ctldir)" \
	"$(DESTDIR)$(asound_module_pcmdir)" \
	"$(DESTDIR)$(asound_module_datadir)"
```

That's why pkg-config should not be used to set the ALSA module installation path, but should default to `/usr/lib/alsa-lib`.

infirit, 2018-03-02:
I don't see how buildroot mangling the pkgconfig paths is a problem of bluez-alsa. pkgconfig is designed to not hardcode paths.

joerg-krause, 2018-03-02:
Right, I just figured that Buildroot is patching pkgconf to sysroot-prefix the libdir variable, because some packages use (misuse?) includedir and libdir for linking. So, I am forced to set the configure options to:

```
--with-alsaplugindir=/usr/lib/alsa-lib
--with-alsadatadir=/usr/share/alsa
```

closed 2018-03-02 by joerg-krause