arkq/bluez-alsa #93: Audio goes silent

RJPoelstra, 2018-03-21:
Hi,

Sometimes I've a problem where the audio goes silent. My phone believes it's still playing, but (in debug mode) I get a lot of message from bluealsa/bluealsa-aplay.
What's happening? And how can I prevent it?
I've attached a log of the messages:
[bluealsa.txt](https://github.com/Arkq/bluez-alsa/files/1833128/bluealsa.txt)

It starts with messages that I connect my phone and hit play.
I then start bluealsa-aplay. After a while I get a 'Connection reset by peer'
bluealsa-aplay then gets into a loop of 'fetching available transports'. During this time audio is silent, but my phone keeps playing.
When I then disconnect the bluetooth from my phone, bluealsa-aplay spits out the 'Bad message' error and quits.

Any help is highly appreciated.

Kind regards,

Remco Poelstra

Arkq, 2018-04-09:
> bluealsa-aplay then gets into a loop of 'fetching available transports'

Yeap, I've seen this by myself however I've got no time to debug it.

However, I doubt that this bug causes silence. Try to start bluealsa without HFP/HSP support (using current master snapshot, run bluealsa as follows: `bluealsa -p a2dp-sink`. The problem might be caused by very naive implementation of HFP connection establishment (in the bluealsa code).

RJPoelstra, 2018-04-17:
I've been testing the new daemon for a while now and it seems to run fine.
Thanks.

borine, 2021-06-03:
Issue appears to have been resolved in later code.

closed 2021-06-03 by borine