arkq/bluez-alsa #94: aplay works, but mplayer hangs

raomin, 2018-03-25:
Hi
I've configured my asoundrc like this:
```
pcm.!default {
 type plug
  slave {
    pcm {
      type bluealsa
      interface hci0
      device "FC:58:FA:42:54:44"
      profile "a2dp"
    }
  }
  hint {
    show on
    description "BT Headset"
  }
}
ctl.!default {
  type bluetooth
  interface "FC:58:FA:42:54:44"
}

```
Playing a WAV with mplayer (ans aplay) works fine:
```
# mplayer /usr/lib/libreoffice/share/gallery/sounds/train.wav
MPlayer 1.3.0 (Debian), built with gcc-6.2.1 (C) 2000-2016 MPlayer Team
do_connect: could not connect to socket
connect: No such file or directory
Failed to open LIRC support. You will not be able to use your remote control.

Playing /usr/lib/libreoffice/share/gallery/sounds/train.wav.
libavformat version 57.56.101 (external)
Mismatching header version 57.56.100
Audio only file format detected.
Load subtitles in /usr/lib/libreoffice/share/gallery/sounds/
==========================================================================
Opening audio decoder: [pcm] Uncompressed PCM audio decoder
AUDIO: 11025 Hz, 1 ch, s16le, 176.4 kbit/100.00% (ratio: 22050->22050)
Selected audio codec: [pcm] afm: pcm (Uncompressed PCM)
==========================================================================
AO: [pulse] Init failed: Connection refused
Failed to initialize audio driver 'pulse'
../../../src/asound/../shared/ctl-client.c:102: Connecting to socket: /var/run/bluealsa/hci0
../../../src/asound/../shared/ctl-client.c:214: Getting transport for FC:58:FA:42:54:44 type 1
../../../src/asound/bluealsa-pcm.c:547: Setting constraints
../../../src/asound/bluealsa-pcm.c:295: Initializing HW
../../../src/asound/../shared/ctl-client.c:341: Requesting PCM open for FC:58:FA:42:54:44
../../../src/asound/bluealsa-pcm.c:318: FIFO buffer size: 4096
../../../src/asound/bluealsa-pcm.c:324: Selected HW buffer: 16 periods x 6000 bytes == 96000 bytes
../../../src/asound/bluealsa-pcm.c:339: Initializing SW
../../../src/asound/bluealsa-pcm.c:339: Initializing SW
../../../src/asound/bluealsa-pcm.c:339: Initializing SW
../../../src/asound/bluealsa-pcm.c:355: Prepared
../../../src/asound/bluealsa-pcm.c:339: Initializing SW
AO: [alsa] 48000Hz 1ch s16le (2 bytes per sample)
Video: no video
Starting playback...
../../../src/asound/bluealsa-pcm.c:226: Starting
../../../src/asound/../shared/ctl-client.c:409: Requesting PCM resume for FC:58:FA:42:54:44
../../../src/asound/bluealsa-pcm.c:121: Starting IO loop
../../../src/asound/bluealsa-pcm.c:430: BlueALSA delay: 15.1 ms (724 frames)
A:   2.1 (02.1) of 2.0 (02.0)  0.9%

../../../src/asound/../shared/ctl-client.c:434: Requesting PCM drain for FC:58:FA:42:54:44
../../../src/asound/bluealsa-pcm.c:132: IO thread paused: 4
../../../src/asound/bluealsa-pcm.c:266: Stopping
../../../src/asound/bluealsa-pcm.c:266: Stopping
../../../src/asound/bluealsa-pcm.c:331: Freeing HW
../../../src/asound/../shared/ctl-client.c:383: Closing PCM for FC:58:FA:42:54:44
../../../src/asound/bluealsa-pcm.c:284: Closing plugin
```

But when trying to play an MP3 file, it hangs before starting playing:
```
# mplayer SoundHelix-Song-1.mp3
MPlayer 1.3.0 (Debian), built with gcc-6.2.1 (C) 2000-2016 MPlayer Team
do_connect: could not connect to socket
connect: No such file or directory
Failed to open LIRC support. You will not be able to use your remote control.

Playing SoundHelix-Song-1.mp3.
libavformat version 57.56.101 (external)
Mismatching header version 57.56.100
Audio only file format detected.
Load subtitles in ./
==========================================================================
Opening audio decoder: [mpg123] MPEG 1.0/2.0/2.5 layers I, II, III
AUDIO: 44100 Hz, 2 ch, s16le, 192.0 kbit/13.61% (ratio: 24000->176400)
Selected audio codec: [mpg123] afm: mpg123 (MPEG 1.0/2.0/2.5 layers I, II, III)
==========================================================================
AO: [pulse] Init failed: Connection refused
Failed to initialize audio driver 'pulse'
../../../src/asound/../shared/ctl-client.c:102: Connecting to socket: /var/run/bluealsa/hci0
../../../src/asound/../shared/ctl-client.c:214: Getting transport for FC:58:FA:42:54:44 type 1
../../../src/asound/bluealsa-pcm.c:547: Setting constraints
../../../src/asound/bluealsa-pcm.c:295: Initializing HW
../../../src/asound/../shared/ctl-client.c:341: Requesting PCM open for FC:58:FA:42:54:44
../../../src/asound/bluealsa-pcm.c:318: FIFO buffer size: 4096
../../../src/asound/bluealsa-pcm.c:324: Selected HW buffer: 16 periods x 6000 bytes == 96000 bytes
../../../src/asound/bluealsa-pcm.c:339: Initializing SW
../../../src/asound/bluealsa-pcm.c:339: Initializing SW
../../../src/asound/bluealsa-pcm.c:355: Prepared
(((Hangs here until i ^C)))
MPlayer interrupted by signal 2 in module: ao2_init
```
I also tried with `delay 2000` and `delay -2000` .. no luck.
Any idea?

Arkq, 2018-04-09:
It looks like ALSA deadlock. See the troubleshooting section in the readme file.

sphakka, 2019-01-21:
`mpv` exhibits similar behaviour with `mp3` and `ape` files. See attached strace output.

[mpv-ape.txt](https://github.com/Arkq/bluez-alsa/files/2780041/mpv-ape.txt)
[mpv-mp3.txt](https://github.com/Arkq/bluez-alsa/files/2780042/mpv-mp3.txt)

With `bluez-alsa-1.3.1` and `alsa-lib-1.1.7` on Gentoo.
Any idea?

Arkq, 2021-01-16:
Strange... I'm almost always testing with `mpv` (on Gentoo) and I've never observed any issues with mp3 files. Does it still occur with `bluez-alsa-3.0.0` (and `alsa-lib >= 1.1.4`)?

borine, 2021-01-17:
@sphakka do the wav, ogg and mp3 files all have the same sample format, rate and channels? The difference in behaviour may be due to different ALSA plugins being used. (Although I personally have not seen any locking issues with alsa-lib 1.1.4 or later)

sphakka, 2021-01-18:
All fine since a while with `alsa-lib-1.2.3` + `bluez-alsa-3.0.0` :-)

Arkq, 2021-01-18:
Great, so I'm closing this issue then :)

closed 2021-01-18 by Arkq