arkq/bluez-alsa #96: bluealsa-aplay cannot connect to slave

ssipos90, 2018-03-29:
Hello, I was following the guide here: https://volumio.org/forum/volumio-bluez-alsa-a2dp-bluetooth-support-t6130.html and I got this error in the output of bluealsa-aplay:
```
$ sudo systemctl status bluealsa-aplay@94\:65\:2D\:7A\:28\:55.service -n 100
● bluealsa-aplay@94:65:2D:7A:28:55.service - BlueAlsa-Aplay 94:65:2D:7A:28:55 -dhw:1,0
   Loaded: loaded (/lib/systemd/system/bluealsa-aplay@.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2018-03-29 12:49:28 EEST; 3min 24s ago
 Main PID: 15223 (bluealsa-aplay)
    Tasks: 4 (limit: 4915)
   CGroup: /system.slice/system-bluealsa\x2daplay.slice/bluealsa-aplay@94:65:2D:7A:28:55.service
           └─15223 /usr/bin/bluealsa-aplay 94:65:2D:7A:28:55 -dhw:1,0

systemd[1]: Started BlueAlsa-Aplay 94:65:2D:7A:28:55 -dhw:1,0.
bluealsa-aplay[15223]: /usr/bin/bluealsa-aplay: ../../utils/../src/shared/ctl-client.c:102: Connecting to socket: /var/run/bluealsa/hci0
bluealsa-aplay[15223]: /usr/bin/bluealsa-aplay: ../../utils/../src/shared/ctl-client.c:126: Subscribing for events: 101
bluealsa-aplay[15223]: /usr/bin/bluealsa-aplay: ../../utils/aplay.c:575: Starting main loop
bluealsa-aplay[15223]: /usr/bin/bluealsa-aplay: ../../utils/aplay.c:597: Fetching available transports
bluealsa-aplay[15223]: /usr/bin/bluealsa-aplay: ../../utils/aplay.c:661: Creating PCM worker 94:65:2D:7A:28:55
bluealsa-aplay[15223]: /usr/bin/bluealsa-aplay: ../../utils/../src/shared/ctl-client.c:102: Connecting to socket: /var/run/bluealsa/hci0
bluealsa-aplay[15223]: /usr/bin/bluealsa-aplay: ../../utils/../src/shared/ctl-client.c:341: Requesting PCM open for 94:65:2D:7A:28:55
bluealsa-aplay[15223]: /usr/bin/bluealsa-aplay: ../../utils/aplay.c:328: Starting PCM loop
bluealsa-aplay[15223]: /usr/bin/bluealsa-aplay: ../../utils/aplay.c:344: Device marked as inactive: 94:65:2D:7A:28:55
bluealsa-aplay[15223]: /usr/bin/bluealsa-aplay: ../../utils/aplay.c:391: An underrun has occurred

```

Output from bluealsa.service
```
$ sudo systemctl restart bluealsa.service 
$ sudo systemctl status bluealsa.service 
● bluealsa.service - BluezAlsa proxy
   Loaded: loaded (/lib/systemd/system/bluealsa.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2018-03-29 12:47:45 EEST; 1s ago
 Main PID: 15136 (bluealsa)
    Tasks: 4 (limit: 4915)
   Memory: 1.1M
      CPU: 8ms
   CGroup: /system.slice/bluealsa.service
           └─15136 /usr/bin/bluealsa

systemd[1]: Started BluezAlsa proxy.
../../src/ctl.c:504: Starting controller loop
../../src/bluez.c:680: Registering endpoint: /A2DP/MPEG24/Source/1
../../src/bluez.c:680: Registering endpoint: /A2DP/MPEG24/Sink/1
../../src/bluez.c:680: Registering endpoint: /A2DP/SBC/Source/1
../../src/bluez.c:680: Registering endpoint: /A2DP/SBC/Sink/1
../../src/bluez.c:910: Registering profile: /HSP/Headset
../../src/bluez.c:910: Registering profile: /HSP/AudioGateway
../../src/bluez.c:910: Registering profile: /HFP/HandsFree
../../src/bluez.c:910: Registering profile: /HFP/AudioGateway
../../src/main.c:249: Starting main dispatching loop
../../src/bluez.c:857: Profile method call: org.bluez.Profile1.NewConnection()
../../src/transport.c:675: State transition: 0 -> 2
../../src/transport.c:104: Created new IO thread: HFP Hands-Free
../../src/bluez.c:797: HFP Hands-Free configured for device 94:65:2D:7A:28:55
../../src/transport.c:675: State transition: 0 -> 2
../../src/ctl.c:586: +-+-
../../src/io.c:1151: Starting IO loop: HFP Hands-Free
../../src/transport.c:104: Created new IO thread: HFP Hands-Free
../../src/rfcomm.c:568: Starting RFCOMM loop: HFP Hands-Free
../../src/rfcomm.c:96: Sending AT message: SET: command:+BRSF, value:244
../../src/at.c:156: AT message: RESP: command:+BRSF, value: 2919
../../src/rfcomm.c:114: HFP state transition: 0 -> 1
../../src/at.c:156: AT message: RESP: command:, value:OK
../../src/rfcomm.c:114: HFP state transition: 1 -> 2
../../src/rfcomm.c:96: Sending AT message: SET: command:+BAC, value:1
../../src/at.c:156: AT message: RESP: command:, value:OK
../../src/rfcomm.c:114: HFP state transition: 2 -> 3
../../src/rfcomm.c:96: Sending AT message: TEST: command:+CIND, value:(null)
../../src/at.c:156: AT message: RESP: command:+CIND, value: ("call",(0,1)),("callsetup",(0-3)),("service",(0-1)),("signal",(0-5)),("roam",(0,1)),("battchg",
../../src/rfcomm.c:114: HFP state transition: 3 -> 4
../../src/at.c:156: AT message: RESP: command:, value:OK
../../src/rfcomm.c:114: HFP state transition: 4 -> 5
../../src/rfcomm.c:96: Sending AT message: GET: command:+CIND, value:(null)
../../src/at.c:156: AT message: RESP: command:+CIND, value: 0,0,1,3,0,1,0
../../src/rfcomm.c:114: HFP state transition: 5 -> 6
../../src/ctl.c:586: +-+-
../../src/at.c:156: AT message: RESP: command:, value:OK
../../src/rfcomm.c:114: HFP state transition: 6 -> 7
../../src/rfcomm.c:96: Sending AT message: SET: command:+CMER, value:3,0,0,1,0
../../src/at.c:156: AT message: RESP: command:, value:OK
../../src/rfcomm.c:114: HFP state transition: 7 -> 8
../../src/rfcomm.c:114: HFP state transition: 8 -> 9
../../src/bluez.c:609: Endpoint method call: org.bluez.MediaEndpoint1.SetConfiguration()
../../src/bluez.c:541: A2DP Sink (AAC) configured for device 94:65:2D:7A:28:55
../../src/bluez.c:543: Configuration: channels: 2, sampling: 44100
../../src/transport.c:675: State transition: 0 -> 0
../../src/ctl.c:586: +-+-
../../src/bluez.c:663: Endpoint already registered: /A2DP/MPEG24/Source/1
../../src/bluez.c:680: Registering endpoint: /A2DP/MPEG24/Sink/2
../../src/bluez.c:663: Endpoint already registered: /A2DP/SBC/Source/1
../../src/bluez.c:663: Endpoint already registered: /A2DP/SBC/Sink/1
../../src/ctl.c:565: New client accepted: 14
../../src/ctl.c:586: +-+-
../../src/ctl.c:565: New client accepted: 15
../../src/ctl.c:586: +-+-
../../src/ctl.c:318: PCM requested for 94:65:2D:7A:28:55 type 1 stream 1
../../src/ctl.c:586: +-+-
../../src/bluez.c:1028: Signal: PropertiesChanged: org.bluez.MediaTransport1
../../src/bluez.c:1028: Signal: PropertiesChanged: org.bluez.MediaTransport1
../../src/transport.c:675: State transition: 0 -> 1
../../src/transport.c:821: New transport: 16 (MTU: R:672 W:1008)
../../src/bluez.c:1028: Signal: PropertiesChanged: org.bluez.MediaTransport1
../../src/transport.c:675: State transition: 1 -> 2
../../src/transport.c:104: Created new IO thread: A2DP Sink (AAC)
../../src/io.c:565: Starting IO loop: A2DP Sink (AAC)
../../src/bluez.c:1028: Signal: PropertiesChanged: org.bluez.MediaTransport1
../../src/transport.c:675: State transition: 2 -> 0
../../src/transport.c:847: Releasing transport: A2DP Sink (AAC)
../../src/transport.c:875: Closing BT: 16
../../src/transport.c:999: Exiting IO thread
```

```$ cat /lib/systemd/system/bluealsa-aplay@.service
[Unit]
Description=BlueAlsa-Aplay %I -dhw:1,0
Requires=bluetooth.service bluealsa.service
[Service]
Type=simple
User=root
Group=audio
Environment=LIBASOUND_THREAD_SAFE=0 
ExecStart=/usr/bin/bluealsa-aplay %I -dhw:1,0
[Install]
WantedBy=multi-user.target
```

My hunch is that the issue is related to permissions / ownership of files or the owner of the alsa process.

---
I've compiled this package from source with AAC enabled. I plan to recompile with aptX.

Some env data:
```
$ uname --kernel-release 
4.13.0-37-generic

$ lsb_release -id
Distributor ID:	Ubuntu
Description:	Ubuntu 17.10
```

Arkq, 2018-04-09:
It looks like termination requested by connected device (the `bluez.c:1028:  Signal:` line):
```
../../src/io.c:565: Starting IO loop: A2DP Sink (AAC)
../../src/bluez.c:1028: Signal: PropertiesChanged: org.bluez.MediaTransport1
../../src/transport.c:675: State transition: 2 -> 0
```

You can debug it further using `dbus-monitor`. Just connect your device and before starting playback run `dbus-monitor --system` as root (running it earlier will gather too much unrelated messages).

ssipos90, 2018-06-15:
Found the problem, forgot to RTFM properly and end to end. Everything works fine. You can close this.

Arkq, 2018-06-15:
OK, great that it's working fine for you. Closing, then.

closed 2018-06-15 by Arkq