arkq/bluez-alsa #97: A2DP and HFP same time

Vitalic66, 2018-03-30:
Hallo,

I use my Pi as headunit for openauto ( Android Auto client ). Listening to music on A2DP audio stream is working without problems. But can I use HFP at the same time?

HFP is necessary to use mic input over usb soundcard to send "OK Google" and other commands to AA running on the connected phone.

Or can only one run at the same time? How can I start HFP?

Arkq, 2018-04-09:
It depends on the headset you have. E.g. my headset does not allow A2DP and HFP to be "enabled" at the same time - A2DP is muted when HFP profile is activated (SCO transport is acquired), which makes sense, because you would rather want to hear incoming call than music. So to be quite honest it is rather not possible to run A2DP and HFP at the same time.

pali, 2018-07-29:
There are some A2DP codecs (like FastStream or aptX Low Latency) which provide backchannel support for microphone voice. So you can use both high quality audio playback and microphone input in A2DP profile. But you need headset with this support.

See: https://www.spinics.net/lists/linux-bluetooth/msg76649.html

closed 2019-11-25 by Arkq