arkq/bluez-alsa #98: Not Pausing on Samsung Tab2 but does on everything else

George-ou812, 2018-04-05:
Hi,

Funny issue with pausing as described in the title. The dbus pause() method request  is being sent to the device but no response is coming back. I have certified that the address is correct, the Player0/1/2 is correct and the Method exists using gdbus introspect --system --dest=org.bluez --object-path /org/bluez --recurse. 

To expand, using dbus-monitor --system I can see bluealsa pausing iPads and other devices correctly, using dbus-send --system --print-reply --dest=org.bluez /org/bluez/hci0/dev_F0_6B_CA_D9_24_DE/player0 org.bluez.MediaPlayer1.Pause from terminal I can do the equivalent - works fine. However on the Sansumg Tab2 follow the same as above absolutely nothing comes back.  dbus-monitor says it's been sent with no errors ( in comparison to the iPad for example ) but after that nothing - hence it doesn't pause.

Thanks,

George-ou812, 2018-04-25:
Adding lots of debug and dbus monitor it seems to have connected via hfp on this device as opposed to A2DP as per normal. A dbus dump shows no pause method for hfp on the tab2. Any idea how to force it's hand? I guess disabling a few profiles in bluetooth in general?

Thanks.

borine, 2021-06-03:
This issue has been idle for 3 years now, so closing it as no longer relevant. Please re-open is this is still an issue with the latest bluealsa code

closed 2021-06-03 by borine