arkq/bluez-alsa #99: Could not restart bluealsa

phaoost, 2018-04-11:
Hello,

I am getting sound stopped sometimes and could not get it restart.
I am trying to kill bluealsa, then cannot start it back:

bluealsa: Couldn't initialize controller thread: Address already in use

There is no bluealsa process running. Can anyone help?

drumfreak, 2018-04-11:
Try to stop and then start the hciuart service. Also make sure permissions are correct on /var/run/bluealsa dir and files.

Arkq, 2018-04-11:
> I am trying to kill bluealsa, then cannot start it back

If `bluealsa` hangs (due to bug), it is not possible to kill it without repercussions :) After `pkill bluealsa` make sure that there is no "files" in the /var/run/bluealsa/ directory (if there is any, remove it).

Also this "I  am getting sound stopped sometimes and could not get it restart" might require a little bit of debugging, because it seems like a nasty bug :/ .

phaoost, 2018-04-18:
Removing files under  /var/run/bluealsa helps alot. How would I help with debugging?

Arkq, 2019-06-01:
Now, the /var/run/bluealsa directory is gone. BlueALSA uses D-Bus for communication.

closed 2019-06-01 by Arkq