arkq/bluez-alsa #103: Raspberry Pi 3: Audio from Kodi 18.3 via bluez-alsa possible?

stahlwollschaf, 2018-04-24:
Hallo,

Don't know whether to adress this issue here or in a OSMC thread, but I'll give it a try:

I'd like to use my OSMC Kodi (17.6.0-24 which is up to date) with a bluetooth JBL Flip 3 speaker. I couldn't manage to get Pulseaudio running properly, so I switched to bluez-alsa which works great, but only with aplay.
my /etc/asound.rc is:
```
# B8:D5:0B:DE:ED:84 MAC JBL-Flip 3
# 00:30:91:40:05:5E MAC BT-dongle
pcm.BT-Sound {
  type plug
  slave.pcm {
    type bluealsa
    device "B8:D5:0B:DE:ED:84"
    profile "a2dp"
  }
  hint {
    show on
    description "BT-Sound"
  }
}

ctl.BT-Sound {
  type bluetooth
  interface "00:30:91:40:05:5E"
}

```

after `sudo systemctl stop mediacenter && sudo systemctl start mediacenter` (but only with speaker connected i.e. switched on!) I'm able to choose my audio output device "ALSA: BT-Sound" in Kodi settings->system->audio. But playing any sound will fail.

Is anybody here who managed to get it run in Kodi on a RBP3? Any help will be appreciated very much.
Maybe important: I didn't use bluealsa-aplay since I coundn't figure out how to use it :/

Regards.

graham8, 2018-04-24:
fzinken on the OSMC forum has managed to get pulse working again.  OSMC would like to adopt bluez-alsa but there are few things to do yet.

esmaeilpour, 2019-11-20:
> Hallo,
>
> Don't know whether to adress this issue here or in a OSMC thread, but I'll give it a try:
>
> I'd like to use my OSMC Kodi (17.6.0-24 which is up to date) with a bluetooth JBL Flip 3 speaker. I [...]
 see how it works herer
https://www.raspberrypi.org/forums/viewtopic.php?t=206898#p1325864

Arkq, 2021-01-16:
Closing stale issue.

closed 2021-01-16 by Arkq