arkq/bluez-alsa #105: Disable profiles with no pause ability

George-ou812, 2018-04-25:
Hi,

I'm having trouble pausing the connected client ( iPad ) as it appears to have connected using the hfp profile which doesn't seem to have a pause method - at least in my case. Is there any harm in disabling most profiles in bluealsa e.g. headset, hands-free, audio sink leaving only A2DP due to the fact my device is pretty much only a BT speaker?

Thanks.

Arkq, 2018-04-25:
No, it's not a problem. In fact, that's why I've pushed [this](https://github.com/Arkq/bluez-alsa/commit/c3a1c3520b1ec8bb9856492d26a73805dfa6b19c) commit :) Compile fresh master and see `bluealsa -h` for more details.

I'm closing this issue, because it's not a bug report.

closed 2018-04-25 by Arkq

George-ou812, 2018-04-25:
Thanks for the advice. One pain is I'm forced to keep ofono as well which
is hfp, taking it out my iPad just won't see BT at all...