arkq/bluez-alsa #107: bluealsa hangs on stream shutdown/teardown

diizzyy, 2018-04-30:
There seems to be some issue with bluealsa when a stream gets shutdown.

When you use aplay (the one included with alsa-utils) it'll occasionally hang during shutdown without any error messages. Hitting ctrl-c doesn't help (no output at prompt) and the only way to get back to the prompt is to kill the bluealsa process. I can also produce this using mpg123. It's not very frequent, about 1 in 10 times or so. I've been able to reproduce this both with and without --disable-thread-safety.

Hardware:
Linksys WRT3200ACM, Bluetooth builtin (Marvell)

Software:
OpenWrt master, Linux 4.14.32, musl 1.1.19
alsa 1.1.6 (with --disable-thread-safety), bluez 5.49, bluez-alsa master branch (commit 2faa68f8278d85a6a104882f75c39e7e50d3c0cd)

I can provide a package file for bluez-alsa if you want to test for yourself, it's not in the packages repo for now.

Have in mind that I'm no programmer so please be verbose if you want me to provide logs etc.

diizzyy, 2018-05-04:
bluealsa is also prone to crash if you interrupt (ctrl-c) aplay and mpg123 in particular during playback.

niwa3836, 2018-11-02:
Hello, sorry if this is the wrong place, I'm really new to this, however I have this issue and wondering if there is anything I can do. Right now I have to reset lots of things to get audio back. It often crashes if I control C out of mpg123 and sometimes crashes on next play back even if mpg123 finished ok. The typical error I get is:
[src/libout123/modules/alsa.c:126] error: initialize_device(): cannot set hw params
[src/libout123/libout123.c:562] error: failed re-opening of device after pause
main: [src/mpg123.c:822] error: Deep trouble! Cannot flush to my output anymore!
[src/libout123/modules/alsa.c:126] error: initialize_device(): cannot set hw params
[src/libout123/libout123.c:562] error: failed re-opening of device after pause
Any thoughts please...

Arkq, 2018-11-06:
Unfortunately, I don't know what might cause such a trouble. I do not own any mips nor arm based linux system, so it is impossible for me to debug. It will be required to see why this `cannot set hw params` error has occurred starting from mpg123... :/ Seems like lots of debugging.

ElDavoo, 2019-02-08:
Same on x86_64. I listen to playlists with mpv, so the stream gets interrupted at every song change. Seemingly randomly mpv hangs while trying to play a song, and so does alsamixer and speaker-test. The only way to let everything work again is to restart bluealsa (I do it with the init.d script). If I get annoyed by this issue I will try to further debug.

Arkq, 2019-02-08:
@ElDavoo could you please check whether PR #184 solves this issue?

ElDavoo, 2019-02-08:
Ok. It did hang one more time before I applied the PR. Thing is it doesn't happen so often so it will take some days to understand if the issue is fixed.
If you don't hear from me in the next 7 days, you can assume the fix does the job

ElDavoo, 2019-02-09:
Nope, apparently it didn't change anything.

Arkq, 2020-01-25:
Do you still experience this issue with new bluealsa version (>= 2.0.0)? If so, please reopen.

closed 2020-01-25 by Arkq