arkq/bluez-alsa #111: Raspberry 3, Volumio, Bluetooth Plugin - blue alsa fails

drindal82, 2018-05-18:
I'm trying to get my JBL Xtreme Speaker working with Raspberry 3 and Volumio. I've installed the [Bluetooth Speeker Plugin](https://volumio.org/forum/bluetooth-speaker-plugin-t7432.html) but there are some problems with blue alsa.

`volumio@poolparty:~$ sudo systemctl status bluezalsa.service
● bluezalsa.service - BluezAlsa proxy
   Loaded: loaded (/lib/systemd/system/bluezalsa.service; enabled)
   Active: failed (Result: exit-code) since Wed 2018-05-16 11:36:27 UTC; 13min ago
  Process: 867 ExecStart=/usr/bin/bluealsa --disable-hfp (code=exited, status=1/FAILURE)
 Main PID: 867 (code=exited, status=1/FAILURE)

May 16 11:36:27 poolparty systemd[1]: Starting BluezAlsa proxy...
May 16 11:36:27 poolparty systemd[1]: Started BluezAlsa proxy.
May 16 11:36:27 poolparty systemd[1]: bluezalsa.service: main process exited, code=exited, status=1/FAILURE
May 16 11:36:27 poolparty systemd[1]: Unit bluezalsa.service entered failed state.
May 16 11:36:27 poolparty bluealsa[867]: /usr/bin/bluealsa: unrecognized option '--disable-hfp'
May 16 11:36:27 poolparty bluealsa[867]: Try '/usr/bin/bluealsa --help' for more information.`

The bluetooth speaker connects to the raspberry, but there is no sound playing.

Can anyone help mi, how to fix this problem. I'm not that experienced with linux to get this working myself.

thanks!

Arkq, 2018-06-04:
Hi,

I don't know how you've installed bluez-alsa (via package manager or manually) but the problem is with the bluezalsa.service file (wherever it is). There has been a change in the bluez-alsa, which is not backward compatible, and it seems that you've got installed bluez-alsa from master snapshot. The simplest solution to your problem is to change line `ExecStart=/usr/bin/bluealsa --disable-hfp` to something like: `ExecStart=/usr/bin/bluealsa -p a2dp-source` (if you need A2DP source profile only, otherwise run `bluealsa -h` and enable needed Bluetooth profiles).

borine, 2021-06-04:
This issue has been idle for 3 years now. I believe the previous comment correctly identified cause, so am now closing.

closed 2021-06-04 by borine