arkq/bluez-alsa #112: Can't get BlueAlsa to work with Kodi

lukestu, 2018-05-26:
Firstly, this is exactly what I was looking for, because I don't want to use PulseAudio on my htpc. Thanks so much for your hard work on this.

I realize this might be a kodi issue, but posting here in case it's an issue with my bluealsa configuration. I find Kodi cannot use bluealsa as a playout device. Aplay to the pcm configured in .asoundrc works perfectly, but kodi is unable to play out, fails with a bunch of errors

My setup:
Ubuntu 18.04 (yeah I know, a bit new)
Kodi 17.6 (also tried 18.0 [beta)](url)
Alsa 1.0.25 (default ubuntu package)
Pulseaudio not installed.

My .asoundrc:

```
pcm.btout {
        type plug
        slave.pcm {
                type bluealsa
                device "E9:7B:E6:01:48:DF"
                profile "a2dp"
        }
        hint {
                show on
                description "Bluetooth Hub"
        }
}
```

Using systemd bluealsa.service to start bluealsa -S.

aplay -D btout xxxx.wav works perfectly"

```
$aplay -D btout 01-TomJoad.WAV
../../../src/asound/../shared/ctl-client.c:102: Connecting to socket: /var/run/bluealsa/hci0
../../../src/asound/../shared/ctl-client.c:214: Getting transport for E9:7B:E6:01:48:DF type 1
../../../src/asound/bluealsa-pcm.c:547: Setting constraints
Playing WAVE '11-LoveMeJustLeaveMeAlone.WAV' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo
../../../src/asound/bluealsa-pcm.c:295: Initializing HW
../../../src/asound/../shared/ctl-client.c:341: Requesting PCM open for E9:7B:E6:01:48:DF
../../../src/asound/bluealsa-pcm.c:318: FIFO buffer size: 4096
../../../src/asound/bluealsa-pcm.c:324: Selected HW buffer: 6 periods x 16380 bytes <= 98284 bytes
../../../src/asound/bluealsa-pcm.c:339: Initializing SW
../../../src/asound/bluealsa-pcm.c:339: Initializing SW
../../../src/asound/bluealsa-pcm.c:339: Initializing SW
../../../src/asound/bluealsa-pcm.c:355: Prepared
../../../src/asound/bluealsa-pcm.c:339: Initializing SW
../../../src/asound/bluealsa-pcm.c:226: Starting
../../../src/asound/../shared/ctl-client.c:409: Requesting PCM resume for E9:7B:E6:01:48:DF
../../../src/asound/bluealsa-pcm.c:121: Starting IO loop
```

Selecting "Bluetooth Hub" as the output device in Kodi renders kodi unable to play anything. Kodi log shows:

```
10:13:57.080 T:140289840568064   ERROR: ActiveAE::InitSink - failed to init
10:13:59.580 T:140289840568064   ERROR: ActiveAE::FlushEngine - failed to flush
10:14:04.581 T:140289840568064   ERROR: ActiveAE::InitSink - failed to init
10:14:05.007 T:140289840568064 WARNING: CActiveAE::StateMachine - signal: 1 from port: OutputDataPort not handled for state: 1
10:14:07.081 T:140289840568064   ERROR: ActiveAE::FlushEngine - failed to flush
10:14:12.081 T:140289840568064   ERROR: ActiveAE::InitSink - failed to init
10:14:14.582 T:140289840568064   ERROR: ActiveAE::FlushEngine - failed to flush
10:14:19.582 T:140289840568064   ERROR: ActiveAE::InitSink - failed to init
10:14:19.582 T:140289840568064 WARNING: CActiveAE::StateMachine - signal: 1 from port: OutputDataPort not handled for state: 1
10:14:22.083 T:140289840568064   ERROR: ActiveAE::FlushEngine - failed to flush
```

Anything obvious I've missed? Thanks for any help!

lukestu, 2018-05-26:
Vlc also works perfectly:

```
vlc --alsa-audio-device btout ....
```

gb160, 2019-12-31:
@lukestu 
Did u ever find a solution mate?
I have this working perfectly on a Pi4 running Kodi, but have the exact same issues u on Kodi on a Vero4K+, playback using aplay from the cli works great.

borine, 2021-06-21:
Kodi appears to be working fine with latest bluez-alsa code

closed 2021-06-21 by borine