arkq/bluez-alsa #113: Omxplayer : no sound with mutiple alsa speakers ans video is too fat

MthatG, 2018-05-31:
Hey !

MthatG, 2018-05-31:
Here is my .asoundrc (quite long, I know) :

```
pcm.!default {
    type plug;
    slave.pcm "surround";
}

pcm.surround {
    type route;
    slave.pcm {
        type multi;
        slaves {
	    a { channels 1 pcm "left" }
            b { channels 1 pcm "right" }
            c { channels 1 pcm "center" }
            d { channels 1 pcm "subwoofer" }
            e { channels 1 pcm "leftSur" }
            f { channels 1 pcm "rightSur" }
        }

        bindings {
         0 { slave a channel 0 }
         1 { slave b channel 0 }
         2 { slave c channel 0 }
         3 { slave d channel 0 }
         4 { slave e channel 0 }
         5 { slave f channel 0 }
        }

    }
    
    ttable [
        [ 1 0 0 0 0 0 ]
        [ 0 1 0 0 0 0 ]
        [ 0 0 1 0 0 0 ]
        [ 0 0 0 1 0 0 ]
        [ 0 0 0 0 1 0 ]
        [ 0 0 0 0 0 1 ]
    ]

}

pcm.stereo {
    type route;
    slave.pcm {
        type multi;
        slaves {
            a { channels 1 pcm "left" }
            b { channels 1 pcm "right" }
        }

        bindings {
         0 { slave a channel 0 }
         1 { slave b channel 0 }
        }
    }

    ttable [
        [ 1 0 ]
        [ 0 1 ]
    ]
}

pcm.one {
    type route;
    slave.pcm "left";
    slave.channels 2;
    ttable.0.0 1;
    ttable.1.1 1;
}

pcm.left {
    type plug;
    slave.pcm {
        type bluealsa;
        interface "hci0";
        device "00:E0:4C:3E:7E:C0";
        profile "a2dp";
    }
}

pcm.center {
    type plug;
    slave.pcm {
        type bluealsa;
        interface "hci0";
        device "00:E0:4C:0E:69:A5"
        profile "a2dp";
    }
}

pcm.right {
    type plug;
    slave.pcm {
        type bluealsa;
        interface "hci0";
        device "00:E0:4C:D7:57:80";
        profile "a2dp";
    }
}

pcm.rightSur {
    type plug;
    slave.pcm {
        type bluealsa;
        interface "hci1";
        device "00:E0:4C:7C:1A:62";
        profile "a2dp";
    }
}

pcm.leftSur {
    type plug;
    slave.pcm {
        type bluealsa;
        interface "hci1";
        device "00:E0:4C:8B:19:72";
        profile "a2dp";
    }
}

pcm.subwoofer {
    type plug;
    slave.pcm {
        type bluealsa;
        interface "hci1";
        device "40:EF:4C:C7:DF:95";
        profile "a2dp";
    }
}

ctl.left {
    type bluealsa;
    interface "hci0";
}

ctl.center {
    type bluealsa;
    interface "hci0";
}

ctl.right {
    type bluealsa;
    interface "hci0";
}

ctl.rigthSur {
    type bluealsa;
    interface "hci1";
}

ctl.leftSur {
    type bluealsa;
    interface "hci1";
}

ctl.subwoofer {
    type bluealsa;
    interface "hci1";
}
```

gpmaxx, 2018-11-16:
Hi.   Is there an update, resolution or confirmation of this issue?  I'm facing the same problem.  Video and sound work fine when playing over HDMI audio. However, when trying to play via bluetooth using the -o alsa option the video plays too fast with no sound. Slowing the video down (hitting 1) corrects the fast play and the sound starts.  Even more confusing is that perhaps 1 out of every 5 times everything works fine from startup.  Using the alsa option to play via i2s works fine.  This is on Raspberry Pi Stretch.

gpmaxx, 2018-11-16:
Not sure if this is the correct place for this issue on second thought?  Is it more a omxplayer problem?

MthatG, 2018-11-16:
Hi!
I finally used gstreamer playbin and it worked very fine.

pinguinpfleger, 2018-12-06:
Same issue here on a pi zero w, thanks for reporting.
I tried gst123 -a alsa myvideo.mp4 but no luck

borine, 2021-06-21:
After 30 months it is clear there are no omxplayer experts following this list. This looks like a raspberrypi specific issue to me, as the upstream omxplayer sources do not support the option `-o alsa`. There have been very many improvements to bluealsa since this issue is raised, so I am closing this issue. 

If it turns out the omxplayer still has bluealsa-specific issues with the latest bluealsa code please feel free to re-open.

closed 2021-06-21 by borine

MelleNi, 2022-07-26:
Still having this exact same issue. Looks like omxplayer is deprecated, but vlc is not a replacement. Has anyone found the fix?

borine, 2022-07-27:
> Has anyone found the fix?

The "fix", if one is possible, would have to be in omxplayer. And as you say, omxplayer is now deprecated by its own maintainer, so any update to it is now most unlikely.

It says on the omxpayer site:
```
Please try using vlc. If t[here](https://github.com/RPi-Distro/vlc/issues) are features of omxplayer that vlc does not handle then try reporting here.
```
So perhaps reporting your issues there is a way forward.

MelleNi, 2022-07-27:
Yep, I've reported my issue there. VLC doesn't seem to work at all with bluealsa, so yeah ...

borine, 2022-07-27:
> VLC doesn't seem to work at all with bluealsa

Please open a new issue here describing the problem you are having with VLC and bluealsa. Perhaps that one can be fixed

MelleNi, 2022-07-27:
I found a workaround that somewhat works. Using python I open a subprocess with the video and then immediately send a keystroke to it. I haven't been able to send multiple keystrokes to it, so I suppose the video doesn't run at 100% speed (but 97.5%?). However, the sound works over bluetooth every time now.

```
from subprocess import Popen, PIPE

p = Popen(["omxplayer -o alsa:bluealsa video.mp4"], stdin=PIPE, shell=True)
p.communicate(input=b'1')
```

I also tried using omxplayer as a python module, but it seems like you cannot send audio parameters to it, so that's not an option.