arkq/bluez-alsa #114: SCO simultaneous aplay arecord not working

truthz03, 2018-06-02:
Hi,

yesterday I found this very nice software. 
I'm trying to build a automotive radio with a rpi3 since a long time and now I want to implement a A2DP and a HFP functionality. The A2DP thing already works but with the HFP I have some problems.

For the A2DP thing I called (this works): 
arecord -f S16_LE -c 2 -r 48000 -D bluealsa:HCI=hci0,DEV=XX:XX:XX:XX:XX:XX,PROFILE=a2dp | aplay

For the HFP thing I called (they don't work simultaneously):
arecord -c 1 -r 8000 -D bluealsa:HCI=hci0,DEV=XX:XX:XX:XX:XX:XX,PROFILE=sco | aplay
arecord -c 1 -r 8000 | aplay -D bluealsa:HCI=hci0,DEV=XX:XX:XX:XX:XX:XX,PROFILE=sco
If I call one of this commands it works, but if I call them in two console windows it doesn't????

I also tried (this works):
arecord -c 1 -r 8000 -D bluealsa:HCI=hci0,DEV=XX:XX:XX:XX:XX:XX,PROFILE=sco | aplay -D bluealsa:HCI=hci0,DEV=XX:XX:XX:XX:XX:XX,PROFILE=sco
Therefore I don't know if this is a problem with the bluealsa software or maybe its a problem with aplay and arecord or alsa itself.

Does anybody have experience with two directional audio through bluealsa?

Thanks
Thomas

Arkq, 2018-06-07:
Hi,

HFP profile is not fully functional yet, it's got some quirks unfortunately. I've got limited testing possibilities, however it seems, that it works quite well with headsets (bluealsa is an audio gateway), but when I'm trying to use it as a headset (smartphone is an audio gateway) it does not work smoothly. The problem is with SCO acquisition in the right time. You might try to use HSP, though. This profile is much simpler and it is a great chance it will work.

ssipos90, 2018-09-07:
@Arkq what would you need to test HFP? Maybe I have some hardware I can send to you...

tbultel, 2018-12-08:
Hi,
I am encountering the same issue, and (and another one related to this that deserves a separate thread), @Arkq, did you get an opportunity to investigate this a little ? Have you got some hints 
for I start to  look on my side ?
Knowing that HSP is not offered by many phones, that means that we currently do not have a working
solution for full-duplex voice
Cheers
Thierry

Arkq, 2018-12-11:
@tbultel does this issue affects ofono-powered HFP as well? I'll try to reproduce this with my setup, however, it will have to wait till weekend.

tbultel, 2018-12-12:
Hi,

The behavior I can see is independent of Ofono or not, I have checked that.

However, I am also facing rate/format issue in the microphone sampling, that
prevent me to isolate the full-duplex issue properly.
I have to try with .wav playback to HFP while listening at the same 
time. Hopefully I can do that by today.

Arkq, 2020-01-25:
Since 2018, there was a lot of work put into the SCO support. I think, that right now SCO profile should work as expected. If the problem still occurs, please reopen this issue, or create new one.

closed 2020-01-25 by Arkq