arkq/bluez-alsa #116: Is it possible to get track details?

redspike, 2018-06-07:
I have a working version of this as a sink for my phone, just wondering if there's anyway to get the current track details at all?

Arkq, 2018-06-07:
Yes, it should be possible, however not via bluez-alsa. You will have to tap into the d-bus and see "Player" property of the `org/bluez/hci0/dev_XX_XX_XX_XX_XX_XX/org.bluez.MediaControl1`. However, I've just checked this property and it seems that my phone does not send media information... but I believe, that I've seen this property with my old phone (which was running vanilla Android).

redspike, 2018-06-07:
Ah okay, thanks for that. Will give it a go!

closed 2018-06-07 by redspike

Arkq, 2018-06-07:
I've checked my old phone and right now it does not expose song information... something must have changed since the last time I've seen it. Maybe it is required to subscribe for such information... dunno. Anyway, this feature (retrieving song info) is not a part of bluez-alsa. Maaaaaybe someday it will be incorporated in the `bluealsa-aplay`.

redspike, 2018-06-07:
Hi @Arkq it seems to be working with my phone here. I've added the code below incase anyone else needs it.

I co-opted a Bluez test script from here: 
[https://github.com/pauloborges/bluez/blob/master/test/test-manager](url)
To create a quick and *very* dirty print out of the DBus data:
[https://gist.github.com/redspike/2d7d51d54c66301a552f696c54a8d681](url)

infirit, 2018-06-07:
I know you said very dirty but I can't help but comment :smiley_cat:.

People should stop using dbus-glib for this and instead use Gio's proxy classes. In blueman we have classes for several objects exported by bluez, see https://github.com/blueman-project/blueman/tree/master/blueman/bluez.

I created a small experimental class and plugin that does the basics of playback, of interest to you is the class at https://github.com/infirit/blueman/blob/mediaplayer/blueman/bluez/AVRemote.py.

On the AVRemote class you can query the Track property under AVRemoteInstance['Track'] than listen for changes by connecting to the `properties-changed` signal, an example can be found in the actual plugin https://github.com/infirit/blueman/blob/mediaplayer/blueman/plugins/applet/AVRemote.py#L180.

gbrault, 2018-12-29:
sudo dbus-monitor --system "type='signal',sender='org.bluez'" let you know the events. Depending upon the client, you may have different behavior
On android
- youtube gives some time twice the track information
- spotify gives it right some time even providing the advertisement content (not always) but "free account" so...

One of my phone gives the volume, the other not (but rather new and cheap)

On windows: vlc nor web gives track

...