arkq/bluez-alsa #121: [Off topic] automatically trust BT device

ssipos90, 2018-06-16:
Hello,

This is off topic but I've been trying to setup my RPi 3B as a BT speaker (a2dp sink) and the PoC works great.

1. After pairing, I always have to go into bluetoothctl and "trust" the newly paired device or it won't connect. Any way to disable/automate this? For what's worthn I'm connecting from an Android 8 phone and I know they changed some stuff in connectivity.

2. Also, I can't seem to be able to make the script run at startup. I tried using systemctl and I've abandoned it and now I'm trying udev but it doesn't work.
```root@raspberrypi:~# cat /etc/udev/rules.d/99-input.rules 
KERNEL=="hci0", SUBSYSTEM=="bluetooth", ACTION=="add", RUN+="/usr/local/bin/bluealsa-a2dp"

root@raspberrypi:~# cat /usr/local/bin/bluealsa-a2dp
#!/bin/bash
/usr/bin/bluealsa --profile=a2dp-sink --profile=hsp-hs --a2dp-volume 2>&1 >> /var/log/bluealsa.log &
/usr/bin/bluealsa-aplay -dhw:0,0 00:00:00:00:00:00 --profile-a2dp    2>&1 >> /var/log/bluealsa.log &
```
If I take those two commands, they work fine.

_The grander scheme of things: this PI will go into my car with some SPI magic as a CD changer emulator and maybe as a BT hands-free if Arkq manages to stabilize HSF/HPF. I've disabled the onboard sound and added an USB soundcard and started working on SPI_

ssipos90, 2018-06-16:
I know realize that the ACTION is no good, because it would duplicate the commands after multiple connections.

Pro-pra, 2018-06-17:
Use ATTRS{idProduct} or ATTRS{idVendor} or check udev info for device and parce bluetooth address.

Or may be bluealsa-a2dp check you device and launch bluealsa if needed.

ssipos90, 2018-07-05:
@Pro-pra thanks for the reply, however I'm not sure on what I'm supposed to do. I have almost no experience with this part of linux.

ssipos90, 2018-09-07:
Meanwhile, I've managed to setup a2dp sink on RPi3 with the integrated BT adapter using systemd for services and a BT python agent.
If any of you guys want, I can compile a tutorial for this because I will reproduce the procedure from my "lab" Pi to my deployment Pi Zero W (using an old integrated USB headphone soundcard - MS LX-3000).

besi, 2018-10-24:
@ssipos90: I would be interested in this. Maybe you could add a link to the tutorial here.

ssipos90, 2018-10-30:
@besi I'll try to compile a gist.

closed 2019-11-25 by Arkq