arkq/bluez-alsa #123: Unable to get transport after 5e0d8a8

pxlbrnd, 2018-06-26:
I am using BlueZ 5.49, alongside with alsa-lib 1.1.4.1 and bluez-alsa (for which I can't thank you enough for maintaining!!).
Both audio source and sink scenarios have been realised, and I am working to improve stability and audio delay regarding some remote devices. Everything works fine until commit 5e0d8a8, which sadly breaks the audio sink functionality, because the lib I'm writing suddenly can't seem to get the transport anymore, for e.g. reading the sample rate in order to set up the audio stream accordingly.

Is there anything you can recommend, may it be in general or specifically?
Let me know if you require additional information.
Kind regards!

Arkq, 2018-06-26:
> Everything works fine until commit 5e0d8a8, which sadly breaks the audio sink functionality, because the lib I'm writing suddenly can't seem to get the transport anymore, for e.g. reading the sample rate in order to set up the audio stream accordingly.

That's strange because there is a tool `bluealsa-aplay` which is used for connecting BT sink with ALSA output PCM, and it works fine. What do you mean, that you can't get the transport anymore? If BT device is connected, then you should be able to get the transport with the [`bluealsa_get_transport()`](https://github.com/Arkq/bluez-alsa/blob/4a81de1d60643c68ef62aced0a003d23d23c0123/src/shared/ctl-client.c#L198) call.

pxlbrnd, 2018-06-26:
Thank you very much for your quick reply!
I am using bluealsa-aplay, indeed.
When calling `bluealsa_get_transport()` with the respective parameters after setting up a connection, I seem to just get a -1 returned back. Using the previous commit, [0ef747e](https://github.com/Arkq/bluez-alsa/commit/e731a1a961b47b3e9064f3e08b21906eef5e182e), it works fine without problems. The remote speaker remains the same in both cases.

Arkq, 2018-10-11:
It seems that master works fine. I'm closing this issue, because no one else reports similar issue.

PR #730 (2024-09-30): Fix potential temporary deadlock in A2DP codec selection procedure.
If a dbus client thread locks the codec_id_mtx mutex while Bluez is processing a SetConfiguration request, then that can lead to the main thread being blocked if a client calls any method that needs to read the current codec. However, the SetConfiguration request needs the main thread to be active to respond to the ClearConfiguration call that Bluez makes during its own handling of SetConfiguration. So there is a risk of deadlock if the client attempts to read the codec id while SetConfiguration is in progress. This deadlock i [... cut]
closed 2018-10-11 by Arkq