arkq/bluez-alsa #124: Delay after disconnections

Ulrar, 2018-06-27:
Hi,

That bug seems to have existed in pulseaudio too and, according to the bug report, was solved. Right now everytime I have small connection problems, the audio starts lagging behind. It can be fixed by hitting pause in mpv and unpausing, so I assume mpv must keep writing in a buffer which for some reason doesn't get flushed after a reconnect, or something like that. Using 1.2.0 from portage (gentoo).

Arkq, 2018-06-27:
Yes, I know about this issue. I've been trying to fix (pinpoint) it for almost a year now... but without any success. Could you past here the link to the bug for PA?

Ulrar, 2018-06-27:
Sure, here : https://bugs.freedesktop.org/show_bug.cgi?id=58746
Sounds like the same issue to me, they seem to have fixed it by messing with the buffer size if I understand it correctly

Arkq, 2018-06-28:
Thanks for the link, there is some valuable information there :)

> Using 1.2.0 from portage (gentoo).

Try **9999. There are A LOT of changes since 1.2 (I will release 1.3 in the near future, though :D). In the master there is also the same workaround (decreased transport buffer) as in the (proposed) patch for PA - 79f68a9. I hope, that this will work for you, however do not close this issue yet, because it is not solved entirely.

I experience out of sync audio-video when I'm using mpv with a large video file (>1G) stored on a suspended hard drive (e.g USB disk). There is no problem when using hardware ALSA PCM, so this issue is for sure bluez-alsa related.

borine, 2021-06-04:
This looks like the same underlying problem as issue #15

borine, 2021-07-27:
It seems this issue is most likely fixed when using latest bluez-alsa code. Given that there has been no update here for a long time, and the related issue #15 is now closed, I am now closing this one too. Please open new issue with debug logs from latest bluez-alsa if same or similar issue occurs now.

closed 2021-07-27 by borine