arkq/bluez-alsa #125: Bluealsa as a sink device

badbat75, 2018-06-27:
When bluealsa-aplay is started volume of reproduction device go to 100%.
There's no way to control it by remote device and volume control can't be disabled.

badbat75, 2018-07-25:
It is possible to implement a parameter in blue-alsa command line to disable volume control when used as a sink device?

Arkq, 2018-07-26:
What do you mean by "disable volume control"? Could you specify exact usecase?

Also, I'm not able to test it properly, because my smarphone does not communicate well with bluez and volume control works fine on the smartphone site (AVRCP volume command is not sent to PC). Anyway, AVRCP communication is beyond the scope of bluez-alsa.

Take a look at this fork: https://github.com/hmbedded/bluez-alsa

badbat75, 2018-07-26:
I'm sorry, I understand I was not so clear.
The problem is that in the BT sink host, independently of how the master volume is set, when blualsa-aplay is started, the master volume of the default alsa device is set to 100%. So if I want to hear at soft volume a bluetooth streaming I have to reduce the bluetooth volume on the source device but the quality of reproduction is very bad because in every device I'm using the blootooth volume is a software volume.
I supposed the behaviour in bluealsa is because the host can manage the volume parameter sent from client devices, but I really don't know.
Is it a bug that volume go to 100% when bluealsa-aplay start, or is there a reason for this behaviour?
If it is the right behaviour, is there a way to disable it keeping the volume at its own value without changing it when bluealsa-aplay is started?

joerg-krause, 2018-07-26:
I have implemented a volume control based on @hmbedded 's fork: https://github.com/joerg-krause/bluez-alsa/tree/volume.

The volume is set at the beginning of a transport session to the current volume of the Mixer element passed via the `--volume-control` option, e.g. "Master":

```
bluealsa -p a2dp-sink --a2dp-volume &
bluealsa-aplay --profile-a2dp --volume-a2dp --volume-control Master 00:00:00:00:00:00
```
With volume control, if the transport volume is changed, e.g. on the iPhone, the ALSA mixers control volume is changed. If the volume is changed on the device, e.g. by hardware buttons or via amixer, the transport value is updated, too.

badbat75, 2018-08-01:
Hi Joerg,
I tested the volume branch with parameter you suggested but:

In this way:
`
/usr/bin/bluealsa -p a2dp-source -p a2dp-sink
/usr/bin/bluealsa-aplay --pcm-buffer-time=20000 --profile-a2dp -d plughw:0,0 --volume-control Master 98:10:E8:2A:E9:70
`
The behaviour remain the same as the standard bluealsa branch
In this way:
`
/usr/bin/bluealsa -p a2dp-source -p a2dp-sink --a2dp-volume
/usr/bin/bluealsa-aplay --pcm-buffer-time=20000 --profile-a2dp -d plughw:0,0 --volume-control Master 98:10:E8:2A:E9:70
`
The hardware volume on sink device go to 100% as the previous way:
`Simple mixer control 'Master',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 248
  Mono: 248 [100%]
Simple mixer control 'Channels',0
  Capabilities: volume
  Playback channels: Front Left - Front Right
  Capture channels: Front Left - Front Right
  Limits: 0 - 248
  Front Left: 218 [88%]
  Front Right: 218 [88%]
`
but software volume control on my iPhone stopped completely to work. Go from mute at mute step to 100% at first step of volume until to last step.

joerg-krause, 2018-08-01:
@badbat75 

`bluealsa` needs to be run with the `--a2dp-volume` command line parameter, otherwise the volume scale is calculated by bluealsa.

So your `Master` device is at 100% in your log. Was it at 100% before you started `bluealsa-aplay`? Furthermore, it is mono. Maybe this isn't handled correctly.

Is your `Channels` mixer usable? If so, could you try to start bluealsa-play with `--volume-control Channels`?

badbat75, 2018-08-01:
Ok I tried also this setup:
/usr/bin/bluealsa -p a2dp-source -p a2dp-sink --a2dp-volume
/usr/bin/bluealsa-aplay --pcm-buffer-time=20000 --profile-a2dp -d plughw:0,0 --volume-control Channels 98:10:E8:XX:XX:XX

Always Master to 100% (with bluealsa-aplay shut down Master was is 57%), no changes on Channels channel when I change volume on my Iphone
And the issue remains as I stated in my previous post, 2nd case.

joerg-krause, 2018-08-01:
I will create a separate branch with includes more debug messages in the next days. Maybe we can find why it is not working for you.

joerg-krause, 2018-08-28:
@badbat75 I've reworked the volume handling. Feel free to test it using the separate volume2 branch of my fork: https://github.com/joerg-krause/bluez-alsa/tree/volume2

badbat75, 2018-08-31:
Hi @joerg-krause , I have just tested, but the situation is the same as my previous post...

joerg-krause, 2019-03-26:
I've reworked the volume control again.

If anybody is interested in testing please try: https://github.com/joerg-krause/bluez-alsa/tree/volume. Note, that the branch is not rebased on Arkqs master branch, yet.

badbat75, 2019-03-31:
Sadly I have to say that behaviour didn't change. Hardware volume 100% and no control on it.
Thank you anyway :-)

joerg-krause, 2019-04-02:
@badbat75 Can you please provide your alsa.conf and how you start bluealsa and bluealsa-aplay?

badbat75, 2019-04-03:
`/usr/bin/bluealsa -p a2dp-source -p a2dp-sink --a2dp-volume
/usr/bin/bluealsa-aplay --pcm-buffer-time=20000 --profile-a2dp -d plughw:0,0 --volume-control Master 98:10:E8:XX:XX:XX`

`#
#  ALSA library configuration file
#

# pre-load the configuration files

@hooks [
        {
                func load
                files [
                        {
                                @func concat
                                strings [
                                        { @func datadir }
                                        "/alsa.conf.d/"
                                ]
                        }
                        "/etc/asound.conf"
                        "~/.asoundrc"
                ]
                errors false
        }
]

# load card-specific configuration files (on request)

cards.@hooks [
        {
                func load
                files [
                        {
                                @func concat
                                strings [
                                        { @func datadir }
                                        "/cards/aliases.conf"
                                ]
                        }
                ]
        }
        {
                func load_for_all_cards
                files [
                        {
                                @func concat
                                strings [
                                        { @func datadir }
                                        "/cards/"
                                        { @func private_string }
                                        ".conf"
                                ]
                        }
                ]
                errors false
        }
]

#
# defaults
#

# show all name hints also for definitions without hint {} section
defaults.namehint.showall on
# show just basic name hints
defaults.namehint.basic on
# show extended name hints
defaults.namehint.extended on
#
[... 738 lines cut ...]
# There are four sets of four numbers (Enable/Disable, Frequency, Bandwidth, Gain). The final number is overall gain.
# - Enable/disable:     1/0
# - Frequency:          20 - 14000 Hz (Center Frequency)
# - Bandwidth:          0.12 - 8 (Q factor) must have leading 0 if < 1
# - Gain:                       -24 - 24 dB
# - Master Gain         -24 - 24 dB
#

pcm.eqfa4p {
    type plug
    slave.pcm "plug_eqfa4p"
}

pcm.plug_eqfa4p {
    type ladspa
        slave.pcm "plughw:0,0";
    path "/usr/lib/ladspa";
    plugins [ {
                id 2609
                label EqFA4p
                input {
                        controls [ 1 20 1 0  1 180 1 0  1 2000 1 0  1 11700 1 0  0 ]
                }
    } ]
}`

gearhead, 2019-04-08:
@badbat75 
What audio card are you using? I have a couple PCM5012s which require software volume control and a number of PCM5122s which have hardware volume control. The 5102, I believe, is always 100% and needs to be adjusted by the amplitude of the PCM signal. The 5122, though has a hardware volume control

The 5122 shows up like this:
```
# amixer
Simple mixer control 'DSP Program',0
  Capabilities: enum
  Items: 'FIR interpolation with de-emphasis' 'Low latency IIR with de-emphasis' 'High attenuation with de-emphasis' 'Fixed process flow' 'Ringing-less low latency FIR'
  Item0: 'Ringing-less low latency FIR'
Simple mixer control 'Analogue',0
  Capabilities: pvolume
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 1
  Mono:
  Front Left: Playback 1 [100%] [0.00dB]
  Front Right: Playback 1 [100%] [0.00dB]
Simple mixer control 'Analogue Playback Boost',0
  Capabilities: volume
  Playback channels: Front Left - Front Right
  Capture channels: Front Left - Front Right
  Limits: 0 - 1
  Front Left: 1 [100%] [0.80dB]
  Front Right: 1 [100%] [0.80dB]
Simple mixer control 'Auto Mute',0
  Capabilities: pswitch
  Playback channels: Front Left - Front Right
  Mono:
  Front Left: Playback [on]
  Front Right: Playback [on]
Simple mixer control 'Auto Mute Mono',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [on]
Simple mixer control 'Auto Mute Time Left',0
  Capabilities: enum
  Items: '21ms' '106ms' '213ms' '533ms' '1.07s' '2.13s' '5.33s' '10.66s'
  Item0: '21ms'
Simple mixer control 'Auto Mute Time Right',0
  Capabilities: enum
  Items: '21ms' '106ms' '213ms' '533ms' '1.07s' '2.13s' '5.33s' '10.66s'
  Item0: '21ms'
Simple mixer control 'Clock Missing Period',0
  Capabilities: enum
  Items: '1s' '2s' '3s' '4s' '5s' '6s' '7s' '8s'
  Item0: '1s'
Simple mixer control 'Deemphasis',0
  Capabilities: pswitch pswitch-joined
  Playback channels: Mono
  Mono: Playback [on]
Simple mixer control 'Digital',0
  Capabilities: pvolume pswitch
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 207
  Mono:
  Front Left: Playback 122 [59%] [-42.50dB] [on]
  Front Right: Playback 122 [59%] [-42.50dB] [on]
Simple mixer control 'Max Overclock DAC',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 40
  Mono: 0 [0%]
Simple mixer control 'Max Overclock DSP',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 40
  Mono: 0 [0%]
Simple mixer control 'Max Overclock PLL',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 20
  Mono: 0 [0%]
Simple mixer control 'Volume Ramp Down Emergency Rate',0
  Capabilities: enum
  Items: '1 sample/update' '2 samples/update' '4 samples/update' 'Immediate'
  Item0: '1 sample/update'
Simple mixer control 'Volume Ramp Down Emergency Step',0
  Capabilities: enum
  Items: '4dB/step' '2dB/step' '1dB/step' '0.5dB/step'
  Item0: '4dB/step'
Simple mixer control 'Volume Ramp Down Rate',0
  Capabilities: enum
  Items: '1 sample/update' '2 samples/update' '4 samples/update' 'Immediate'
  Item0: '1 sample/update'
Simple mixer control 'Volume Ramp Down Step',0
  Capabilities: enum
  Items: '4dB/step' '2dB/step' '1dB/step' '0.5dB/step'
  Item0: '1dB/step'
Simple mixer control 'Volume Ramp Up Rate',0
  Capabilities: enum
  Items: '1 sample/update' '2 samples/update' '4 samples/update' 'Immediate'
  Item0: '1 sample/update'
Simple mixer control 'Volume Ramp Up Step',0
  Capabilities: enum
  Items: '4dB/step' '2dB/step' '1dB/step' '0.5dB/step'
  Item0: '1dB/step'
```
The 5102 shows no mixer capability. No listing on alsamixer or amixer and will require the phone to modulate volume. I want to have the phone volume control to directly control the mixer volume control on my i2s chip. Will this allow this? I do not know how this would work... This is how shairport-sync and upmpdcli interacts with the i2s cards.

joerg-krause, 2019-04-08:
I am using the TI PCM5102 as well. I can control the volume using the ALSA softvol plugin. Amixer is showing this:

```
# amixer 
Simple mixer control 'Master',0
  Capabilities: volume
  Playback channels: Front Left - Front Right
  Capture channels: Front Left - Front Right
  Limits: 0 - 255
  Front Left: 128 [50%]
  Front Right: 128 [50%]
```

joerg-krause, 2019-04-08:
@badbat75 I do not see any ALSA mixer control configuration in your configuration. What is the output of `cat /etc/asound.conf`?

gearhead, 2019-04-08:
# cat /etc/asound.conf
cat: /etc/asound.conf: No such file or directory

Don't have an asound.conf. Never had one or needed one. I run mpd/upmpdcli and shairport-sync on these pis and have no problem controlling the volume as long as it is set properly. 

With the PCM5102, (in Shairport-sync) the access to the alsa device requires:
output_device="hw:0";

With the PCM5122, (alsin in Shairport-sync):
output_device="hw:0"; // alsa_output_device
mixer_control_name="Digital"; // alsa_mixer_control_name
mixer_device="hw:0";

I'm not that knowledgeable about how this all works, but this is what I have learned, that I need to specify the mixer device to get the volume to work correctly with the 5122 and the 5102 requires that I comment out all references to the mixer to get it to work.

badbat75, 2019-06-10:
> 
> 
> @badbat75 I do not see any ALSA mixer control configuration in your configuration. What is the output of `cat /etc/asound.conf`?

Joerg I replied in the 3rd April post.

joerg-krause, 2019-06-10:
@badbat75 Sorry for not replying earlier...

From what I can see I guess you are using Moode Audio. What DAC does your system has? What is the output of `alsamixer` and `aplay -L`?

tuomashamalainen, 2020-03-28:
@joerg-krause I happened across this issue and your fork when searching for a solution to this very problem. How much would it take to merge new bluez-alsa changes to your fork? I'm pretty interested in testing this.

joerg-krause, 2020-04-01:
Hi @tuomashamalainen , thanks for your interest! I rebase my branch from time to time. My latested private branch is somewhere between v2.1.0 and master. I will publish it soon.

joerg-krause, 2020-04-01:
This is my volume branch based on v2.1.0 including some follow up fixes: https://github.com/joerg-krause/bluez-alsa/tree/volume-v2.1.0.

tuomashamalainen, 2020-04-01:
Thanks Jörg, will give this a go with a Beocreate 4-Channel Amplifier.

tuomashamalainen, 2020-04-02:
@joerg-krause, testing this just now and the volume sync works great, both ways. Thanks again. We would love to include this in HiFiBerryOS/Beocreate 2 if you've got no objections.

joerg-krause, 2020-04-03:
I'm glad it works! This encourages me to make a pull request :smile:  I have no objections, you are welcome to add it to HiFiBerryOS.

joerg-krause, 2020-04-07:
@tuomashamalainen I've rebased the commits to latest upstream master. Mind to test the branch https://github.com/joerg-krause/bluez-alsa/tree/alsa-volume-control, so I can prepare a PR? Thanks!

Note, that the ALSA PCM device parameter has changed in bluez-alsa from `-d` to `-D`.

hifiberry, 2020-04-07:
@joerg-krause Latest commit on this branch is from September last year. Is this correct?

joerg-krause, 2020-04-07:
Yes, it is the date of when the commit was created. I rebased the commits on top of master.

paul-1, 2020-04-08:
I was using my own rebased version, but I built yours, and it appears to be working fine.

hifiberry, 2020-04-08:
@joerg-krause Yes, seems to work here.

gearhead, 2020-04-08:
@Arkq Can you implement this in the current git?

joerg-krause, 2020-04-09:
Thanks @paul-1 , @hifiberry for testing. @gearhead I'm preparing a PR soon, so @Arkq can review the code.

joerg-krause, 2020-04-14:
I've rebased the branch again on latest master branch and changed the command line option to `-M` or `--mixer=NAME`.

Arkq, 2021-01-10:
With bluez >= 5.55, there is a fix for reporting volume for headsets, and in the current master of bluez-alsa, there is a fix for correct volume scaling (gauge on smartphone, e.g. iPhone, and volume on host with bluez-alsa should match). Please, check whether right now the volume control is as expected.

As for the issue request for disabling volume scaling. There is a possibility to disable software volume scaling with D-Bus call. The SoftVolume property of the PCM endpoint: https://github.com/Arkq/bluez-alsa/blob/a255e2483319b2eca7d50bd94a2e63dec3f28b9c/doc/bluealsa-api.txt#L110-L115

Because the requested feature has been implemented, I'm closing this issue.

Discussion for volume forwarding between ALSA mixer and Bluetooth device, please continue in the PR #321 or #390.

closed 2021-01-10 by Arkq