arkq/bluez-alsa #126: bluealsa: Couldn't initialize controller thread: Bad file descriptor

Ulrar, 2018-06-29:
Hey,

As advised in a separate issue, I updated to 9999 (git version on gentoo, as you know). Since that, I haven't been able to restart bluealsa with the init script, it just crashes. Before rebooting I could still start it manually, but since I rebooted I can't : 

```bash
 > bluealsa
bluealsa: Couldn't initialize controller thread: Bad file descriptor
 > bluealsa -S
bluealsa: Segmentation fault
```

I'll revert to 1.2.0 for now, but let me know if that's something you know about, or if you'd like me to test stuff.
Thanks

Arkq, 2018-06-30:
O yeah... I forgot, the init script from *9999 is not compatible with the newest master (I will have to report this :D), so the bluealsa service might not start (if there is no custom configuration). In order to start current master you will have to specify exactly which BT profile should be enabled.

Change the `BLUEALSA_CONF` in the `/etc/init.d/bluealsa` (if you haven't modified it by yourself) from
```
BLUEALSA_CONF="--disable-hsp"
```
to 
```
BLUEALSA_CONF="-p a2dp-source"
```

for more info see `bluealsa --help`.

> bluealsa: Couldn't initialize controller thread: Bad file descriptor

Make sure, that there is no other `bluealsa` running. Or (if there was some SIGSEGV before), remove every file from the `/run/bluealsa/` directory (do NOT remove directory itself, though).

> bluealsa: Segmentation fault

Hmm..., that's strange. Could you provide some more info, like backtrace from gdb? Preferably, compile bluez-alsa from git master by yourself (do not install in, just compile and do not unmerge portage's version if it is *9999) with the configuration as follows:
```
../configure CFLAGS="-g -O0" LDFLAGS="-g"
```
then simply run compiled `src/bluealsa` with `gdb`: `gdb --args ./src/bluealsa [your set of options]`.

Ulrar, 2018-07-01:
That does seem to work !
I guess I must have had old leftovers in /run then, but now with the proper BLUEALSA_CONF it seems to work well, thanks

Arkq, 2018-07-02:
Great, I'm closing this issue, then. Also, I've created bug for **9999 ebuild: https://bugs.gentoo.org/show_bug.cgi?id=659838.

closed 2018-07-02 by Arkq