arkq/bluez-alsa #129: Make bluealsa-aplay release the audio device on silence

TheDonDan, 2018-07-21:
Hi,
for my using scenario it would be extremely usefull if the bluealsa-aplay would release the audio device if the audio stream from the bluetooth source turns silent.

I am using an Raspberry Pi as an audio-box. At the software side it combines mpd to play local files and bluealsa-aplay to act as an bluetooth speaker to play music from the mobile phone.

On the hardware side I use the i2s soundcard from  iqaudio. The iqaudio driver has an option to mute the amplifier while the alsa audio device is closed, which it great because it saves power an it turn off any annoying hiss/noise from the speaker while nothing is playing. 

The problem now is that, if my mobile phone is connected via bluetooh to the audio-box  bluealsa-aplay starts automatically and grabs the alsa device unmuting the amplifier even if there is no sound playing. 

The second issue is that  until I turn the bluetooth link from mobile phone off, bluealsa-aplay
blocks the alsa device, not allowing to play audio with mpd although there is no sound coming from the phone.

I know that that "race" condition cannot be solved with alsa, but I am OK with the principle "first come, first served" but not blocking if there is nothing coming... 

By the way: Really big thanks for making bluez with alsa possible, I have been waiting for this a few years know, since my first Raspberry pi! I always thought that pulseaudio for a small "embedded" audio-box is an overcomplicated overkill.

Thanks

DonDan

Arkq, 2018-07-23:
Thanks, for pointing that out. It is indeed good idea not to open PCM if there is no data.

closed 2018-07-23 by Arkq

joerg-krause, 2018-07-24:
@Arkq Thanks for the commit! Works like a charm :smile:

TheDonDan, 2018-07-25:
Updated and testing! Thanks for the (very) fast enhancement!

One thing would make the behavior a bit nicer, if I play some music from an local source in the (bluetooth) silence period the local player plays nicely and blocks the PCM device as expected.

 If in the meantime some sound is send from the bluetooth device the bluealsa service logs as expected the following error "Couldn't open PCM: Device or resource busy, aplay.c:285: Exiting PCM worker". 
So far so good, but if I turn the local player off and the PCM device is freed again it would be nice if  aplay  would try to open the PCM again as soon the bluetooth stream exits the silence and starts sending audio again.

The would make a very viable way of coexisting both audio players. To switch from one to the other one must only press stop on one side and press pause/play on the other!

Thanks

DonDan

Arkq, 2018-07-26:
Test the current master, I hope that right now it will work for you:)

TheDonDan, 2018-07-27:
Realy big thanks for the effort! At this point it works realy great! I really like it!

The only think to note is that there is maybe a small flaw from an  "resource managment" point of view.

If some other process has grabbed the PCM first the aplay process tries to reopen the PCM very aggressively (multiple time per second) and consumes about 15% CPU on an Raspberry Pi trying to reopen PCM every frame it receives. 

It is OK for my using scenario, but I think It would be enough if the aplay process only retries to open the PCM again wenn returning from the "Device marked as inactive" status. 

This would have the side effect that the Bluetooth stream is only resumed on a pause/play cycle on the source device and not immediately as currently, but this way it would not hog the cpu unnecessarily.

Thanks

DonDan

TheDonDan, 2018-08-04:
I saw now that you fixed it a few days ago!   
Thank you! This is getting realy nice!

Now the two bluealsa-aplay processes just consume together under 1.5% CPU (RPi B+) on retrying, before it was almost 15-20% together. Very Nice!
I compared both versions again. What I observed doing this, was that the two "bluealsa --profile-a2dp-sink" processes still consume 16-20% together.   What are they doing? I thought decoding is done by  bluealsa-aplay. Must they compute the incoming stream even if it is not used by the aplay process? 
Would it be possible to reduce it as well?

Cheers 

DonDan

Arkq, 2018-08-05:
> I compared both versions again. What I observed doing this, was that the two "bluealsa --profile-a2dp-sink" processes still consume 16-20% together. What are they doing? I thought decoding is done by bluealsa-aplay. Must they compute the incoming stream even if it is not used by the aplay process?

No, decoding/encoding is performed by `bluealsa` process and raw PCM data is forwarded to client(s). Right now it is not possible to connect more than one client, however general idea is that there is a single process with high load - `bluealsa`.

> Would it be possible to reduce it as well?

No, not really. Generally, it would be possible to change PCM data flow, however in the long run it might generate unnecessary spaghetti code.

commit 65a1a117 (2018-07-23): Release playback PCM when there is no data

Grabbing PCM activates underlying device. If there is no data (data
might not arrive for a long time), there is no need to keep device
in an active state, because this will prevent device to enter power
saving mode.

Fixes #129