arkq/bluez-alsa #130: Using AAC Codec results a complete muted reproduction

badbat75, 2018-07-23:
I'm using the chain bluealsa -> bluealsa-aplay to reproduce music from my phone.
If I use a standard SBC coded everything is working, when I try to use AAC (using the iPhone) there's no audio.
In debug mode I see all the packets flow so everything seems fine.
Something is happening in the fdk-aac library. Have you tried this before?
Thank you

Arkq, 2018-07-23:
I've been testing it with iPhone, however it was quite a long time ago and since then there was a lot of changes in the bluez-alsa project :) Fortunately, right now I've got access to an Android phone with the AAC support. I will check it today in the evening.

Arkq, 2018-07-23:
I hope, that this "fix" will work for you. It seems that iPhone does not follow A2DP AAC specification...

closed 2018-07-23 by Arkq

badbat75, 2018-07-23:
I Ark thank you for the prompt reply. Now the something exits from the loudspeakers but it is like pieces of waveform separated by silence (likes 3 or 4 per seconds). (Imagine someone at the volume that raise to 100% and low to 0% 3 or 4 times per seconds... this is the effect)

joerg-krause, 2018-07-24:
@Arkq I can confirm that the commit fixes the issue on my iPhone. I do not get the fragmented sound as described by @badbat75 .

Arkq, 2018-07-24:
@joerg-krause I do not know what causes sound chopping, but I was able to reproduce it as well. However, I think, that it is not related to this issue. It will be tracked with the #132. I've got two doogee smartphones, and with one of these phones sound streaming does not work....

commit b907f5a3 (2018-07-23): RTP mark bit quirk workaround

It seems, that RTP fragmentation is a rocket science, and not everyone
gets it. A2DP specification states, that the media payload format is
defined in the RFC 3016 (obsoleted by the RFC 6416), and I quote:

The marker bit is set to 1 to indicate that the RTP packet contains
a complete audioMuxElement or the last fragment of an audioMuxElement.

This commit adds a workaround for cases where mark bit is set to zero
for all packets.

Fixes #130