arkq/bluez-alsa #132: Using AAC Codec after b907f5a commit results in fragmented audio reproduction

badbat75, 2018-07-23:
After the commit something exits from the loudspeakers but it is like pieces of waveform separated by silences (likes 3 or 4 per seconds). (Sounds like someone at the volume that raises to 100% and then to 0% 3 or 4 times per seconds... this is the effect)

badbat75, 2018-07-24:
I continue here.
My last test. I applied the b907f5a patch (io.c) on tagged version 1.3.0 skipping fd6e91b, 65a1a11 and bd49d2e commits and everything works flawlessly with AAC and SBC codecs.
So the problem is in one of these 3 commits as you mentioned in the other thread.

Arkq, 2018-07-24:
Hmm, that's strange. I've checked SBC with current master (with doogee smartphone) and it works flawlessly. Then, I've checked AAC and it does not work. In the system log I've got lots of messages:
`kernel: Bluetooth: Dropping L2CAP data: receive buffer overflow`

Quick googling shows that this issue is caused by the connected Bluetooth device (in this case smartphone): http://linux-bluetooth.vger.kernel.narkive.com/obnRK7AD/patch-bluetooth-print-error-when-dropping-l2cap-data

Could you confirm that in your logs there are also messages like this?

My PC setup:
kernel 4.14.52
bluez 5.49

Smartphone 1 (AAC **does not** work):
Android 5.1
DOOGEE X5 2016.02.22
kernel 3.4.5

Smartphone 2 (AAC **does** work):
Android 6.0
DOOGEE X5pro 2016.07.28
kernel 3.18.19

Arkq, 2018-07-24:
Oookaayyy....

Could you cherry-pick commits fd6e91b and bd49d2e on top of your patched v1.3.0 and test whether your setup will work (test iPhone and one of devices with SBC codec)? This will narrow down a little bit further the cause. I'm 99% sure that commit fd6e91b does not cause troubles. If the test will be positive (65a1a11 is the troublesome commit) I will try to prepare some changes which will allow to debug this issue further.

badbat75, 2018-07-24:
Hi Arkq.
Step by step.
Commit fd6e91b + b907f5a
iPhone 6S iOS 11.4 (AAC worked)
Lumia 950 (Windows 10) (SBC worked)

I'm a little bit slow on test commits because I'm manually integrating it in the code.
Here is becaoming late. Tomorrow I will test on top the bd49d2e commit.
Cheers

badbat75, 2018-07-24:
I couldn't resist... :-)
Commit bd49d2e + fd6e91b + b907f5a
iPhone 6S iOS 11.4 (AAC worked)
Lumia 950 (Windows 10) (SBC worked)

Still remain the 65a1a11 commit that at this point should be the problem....

Arkq, 2018-07-26:
I've made some changes (on top of current master) and pushed it into the [test-132](https://github.com/Arkq/bluez-alsa/tree/test-132) branch. It adds some debug lines, and restores previous behaviour (PCM device is opened during bluealsa-aplay startup). If this solution will fork, please check if it will still work when [these](https://github.com/Arkq/bluez-alsa/blob/test-132/utils/aplay.c#L344) lines are uncommented. If bluealsa-aplay still works, please post here the debug lines printed during bluealsa-aplay startup (enable debug during configuration).

badbat75, 2018-07-26:
Ok, unfortunately I can't test today. I will give you some updates the day after tomorrow.

badbat75, 2018-07-27:
Oh my...
I tested the test-132 branch without and with uncommented lines. It works in both cases.
Those are the logs:
``Jul 27 19:52:30 moode systemd[1]: Started BlueAlsa-Aplay.
Jul 27 19:52:30 moode bluealsa-aplay[11150]: /usr/bin/bluealsa-aplay: ../../utils/../src/shared/ctl-client.c:103: Connecting to socket: /var/run/bluealsa/hci0
Jul 27 19:52:30 moode bluealsa-aplay[11150]: /usr/bin/bluealsa-aplay: ../../utils/../src/shared/ctl-client.c:130: Subscribing for events: 101
Jul 27 19:52:30 moode bluealsa-aplay[11150]: /usr/bin/bluealsa-aplay: ../../utils/aplay.c:622: Starting main loop
Jul 27 19:52:30 moode bluealsa-aplay[11150]: /usr/bin/bluealsa-aplay: ../../utils/aplay.c:644: Fetching available transports
Jul 27 19:52:30 moode bluealsa[10569]: /usr/bin/bluealsa: ../../src/ctl.c:614: Received new connection: 10
Jul 27 19:52:30 moode bluealsa[10569]: /usr/bin/bluealsa: ../../src/ctl.c:627: New client accepted: 10
Jul 27 19:52:30 moode bluealsa[10569]: /usr/bin/bluealsa: ../../src/ctl.c:651: +-+-
Jul 27 19:52:30 moode bluealsa[10569]: /usr/bin/bluealsa: ../../src/ctl.c:651: +-+-
Jul 27 19:52:30 moode bluealsa[10569]: /usr/bin/bluealsa: ../../src/ctl.c:651: +-+-
Jul 27 19:52:31 moode bluealsa[10569]: /usr/bin/bluealsa: ../../src/bluez.c:608: Endpoint method call: org.bluez.MediaEndpoint1.SetConfiguration()
Jul 27 19:52:31 moode bluealsa[10569]: /usr/bin/bluealsa: ../../src/ctl.c:645: Emitting notification: 1 => 10
Jul 27 19:52:31 moode bluealsa-aplay[11150]: /usr/bin/bluealsa-aplay: ../../utils/aplay.c:644: Fetching available transports
Jul 27 19:52:31 moode bluealsa[10569]: /usr/bin/bluealsa: ../../src/ctl.c:651: +-+-
Jul 27 19:52:31 moode bluealsa[10569]: /usr/bin/bluealsa: ../../src/bluez.c:540: A2DP Sink (AAC) configured for device 98:10:E8:2A:E9:70
Jul 27 19:52:31 moode bluealsa[10569]: /usr/bin/bluealsa: ../../src/bluez.c:542: Configuration: channels: 2, sampling: 44100
Jul 27 19:52:31 moode bluealsa[10569]: /usr/bin/bluealsa: ../../src/transport.c:691: State transition: 0 -> 0
Jul 27 19:52:31 moode bluealsa[10569]: /usr/bin/bluealsa: ../../src/ctl.c:651: +-+-
Jul 27 19:52:31 moode bluealsa[10569]: /usr/bin/bluealsa: ../../src/bluez.c:658: Endpoint already registered: /A2DP/MPEG24/Source/1
Jul 27 19:52:31 moode bluealsa[10569]: /usr/bin/bluealsa: ../../src/bluez.c:675: Registering endpoint: /A2DP/MPEG24/Sink/2
Jul 27 19:52:31 moode bluealsa[10569]: /usr/bin/bluealsa: ../../src/bluez.c:658: Endpoint already registered: /A2DP/SBC/Source/1
Jul 27 19:52:31 moode bluealsa[10569]: /usr/bin/bluealsa: ../../src/bluez.c:658: Endpoint already registered: /A2DP/SBC/Sink/1
Jul 27 19:52:31 moode bluetoothd[10557]: Endpoint registered: sender=:1.20 path=/A2DP/MPEG24/Sink/2
Jul 27 19:52:31 moode bluealsa-aplay[11150]: /usr/bin/bluealsa-aplay: ../../utils/aplay.c:709: Creating PCM worker 98:10:E8:2A:E9:70
Jul 27 19:52:31 moode bluealsa-aplay[11150]: /usr/bin/bluealsa-aplay: ../../utils/aplay.c:315: buffer_size: new:8820, old:882
Jul 27 19:52:31 moode bluealsa-aplay[11150]: /usr/bin/bluealsa-aplay: ../../utils/../src/shared/ctl-client.c:103: Connecting to socket: /var/run/bluealsa/hci0
Jul 27 19:52:31 moode bluealsa-aplay[11150]: /usr/bin/bluealsa-aplay: ../../utils/../src/shared/ctl-client.c:345: Requesting PCM open for 98:10:E8:2A:E9:70
Jul 27 19:52:31 moode bluealsa-aplay[11150]: /usr/bin/bluealsa-aplay: ../../utils/aplay.c:337: pause_threshold: new:88200, old:88200
Jul 27 19:52:31 moode bluealsa-aplay[11150]: /usr/bin/bluealsa-aplay: ../../utils/aplay.c:348: Starting PCM loop
Jul 27 19:52:31 moode bluealsa[10569]: /usr/bin/bluealsa: ../../src/ctl.c:614: Received new connection: 13
Jul 27 19:52:31 moode bluealsa[10569]: /usr/bin/bluealsa: ../../src/ctl.c:627: New client accepted: 13
Jul 27 19:52:31 moode bluealsa[10569]: /usr/bin/bluealsa: ../../src/ctl.c:651: +-+-
Jul 27 19:52:31 moode bluealsa[10569]: /usr/bin/bluealsa: ../../src/ctl.c:343: PCM requested for 98:10:E8:2A:E9:70 type 1 stream 1
Jul 27 19:52:31 moode bluealsa[10569]: /usr/bin/bluealsa: ../../src/ctl.c:651: +-+-
Jul 27 19:52:31 moode bluealsa[10569]: /usr/bin/bluealsa: ../../src/bluez.c:1032: Signal: PropertiesChanged: org.bluez.MediaTransport1
Jul 27 19:52:46 moode bluealsa[10569]: /usr/bin/bluealsa: ../../src/bluez.c:1032: Signal: PropertiesChanged: org.bluez.MediaTransport1
Jul 27 19:52:46 moode bluealsa[10569]: /usr/bin/bluealsa: ../../src/transport.c:691: State transition: 0 -> 1
Jul 27 19:52:46 moode bluetoothd[10557]: /org/bluez/hci0/dev_98_10_E8_2A_E9_70/fd0: fd(22) ready
Jul 27 19:52:46 moode bluealsa[10569]: /usr/bin/bluealsa: ../../src/transport.c:835: New transport: 17 (MTU: R:672 W:672)
Jul 27 19:52:46 moode bluealsa[10569]: /usr/bin/bluealsa: ../../src/bluez.c:1032: Signal: PropertiesChanged: org.bluez.MediaTransport1
Jul 27 19:52:47 moode bluealsa[10569]: /usr/bin/bluealsa: ../../src/transport.c:691: State transition: 1 -> 2
Jul 27 19:52:47 moode bluealsa[10569]: /usr/bin/bluealsa: ../../src/transport.c:104: Created new IO thread: A2DP Sink
Jul 27 19:52:47 moode bluealsa[10569]: /usr/bin/bluealsa: ../../src/io.c:582: Starting IO loop: A2DP Sink (AAC)
Jul 27 19:52:47 moode bluealsa[10569]: /usr/bin/bluealsa: ../../src/io.c:668: Fragmented RTP packet [1]: LATM len: 42
Jul 27 19:52:47 moode bluealsa[10569]: /usr/bin/bluealsa: ../../src/io.c:668: Fragmented RTP packet [2]: LATM len: 42
Jul 27 19:52:47 moode bluealsa[10569]: /usr/bin/bluealsa: Activating RTP mark bit quirk workaround
[... cut ...]
Jul 27 19:53:29 moode bluealsa-aplay[11150]: /usr/bin/bluealsa-aplay: ../../utils/aplay.c:644: Fetching available transports
Jul 27 19:53:29 moode bluealsa[10569]: /usr/bin/bluealsa: ../../src/ctl.c:651: +-+-
Jul 27 19:53:29 moode bluealsa[10569]: /usr/bin/bluealsa: ../../src/ctl.c:581: Client closed connection: 10
Jul 27 19:53:29 moode bluealsa[10569]: /usr/bin/bluealsa: ../../src/ctl.c:651: +-+-
Jul 27 19:53:29 moode systemd[1]: Stopping BlueAlsa-Aplay...
Jul 27 19:53:29 moode systemd[1]: Stopped BlueAlsa-Aplay.
``

Arkq, 2018-07-30:
OK, so just out of curiosity, could you try current master and confirm, that it is NOT working? Because, with uncommented lines, this branch differs from master with only one thing (reading buffer is 10x bigger), which should not cause such an effect. In fact it should reduce sound fragmentation if it has any effect.

badbat75, 2018-07-30:
Ok just recompiled everything from master branch and choppy sound is back.

Arkq, 2018-07-30:
Hmm... I've pushed new testing code to the test-132 branch. Could you make 4 more tests (logs are not important, just which code works and which does not):

1. Test-132 branch without modifications.
2. Change [L301](https://github.com/Arkq/bluez-alsa/blob/test-132/utils/aplay.c#L301) to `#if 0`.
3. (revert change from 2) Modify [L317](https://github.com/Arkq/bluez-alsa/blob/test-132/utils/aplay.c#L317) according to comment.
4. Make changes from 2 and 3.

badbat75, 2018-07-30:
Bingo... so you find it.
Test 1: OK
Test 2: OK
Test 3: KO
Test 4: KO

Arkq, 2018-07-31:
> Bingo... so you find it.

It seems so, however I do not know why this buffer size causes such an issue.... Anyway, please try current test-132 branch and let me know if it is working for you. If not, more "remote" debugging will be required :).

badbat75, 2018-07-31:
Great job Arkq, now it works like a charm.

commit 69c2eb39 (2018-07-31): Read no more samples than a period size

Fixes #132
closed 2018-08-01 by Arkq