arkq/bluez-alsa #135: Multiple devices connectable

hahagu, 2018-08-02:
I know this is already once discussed [here](https://github.com/Arkq/bluez-alsa/issues/70), but I would like to bring it up again.
The issue is that, since multiple devices are connectable, it makes switching to another device very inconvenient.
For example, if you want to switch from device A to device B, and both are connected, you have to first disconnect device A, then disconnect device B, and reconnect device B just to switch to device B, while a 'normal' bluetooth speaker will just not allow device B to connect when device A is connected in the first place.
Please look into this. Thanks in advance!

borine, 2021-06-21:
Response nearly 3 years late ! :smile: 

When multiple devices are connected, `bluealsa-aplay --single-audio` plays back only the first device to send audio, the others are sent an AVRCP "pause" request. When that first device becomes idle then bluelalsa-aplay stops pausing the others, and the next to send audio will be played, any remaining paused again.

Devices that accept the pause request generally pause their player application, and the user must then press "play", or whatever, on the device to make it start playback again. For devices that do not accept pause requests their behaviour depends on the player software they are using. I don't have any that actually need a disconnect-reconnect sequence to start playback though. Often they just continue playback as soon as the bluetooth stream is unblocked.

Having said that, maybe it might be useful in some circumstances to have an option on the bluealsa service to accept only one connection at a time. So I am labelling this as a "feature request" (albeit probably too late for the OP)

hahagu, 2021-06-29:
> Response nearly 3 years late ! 😄

Ah yes, I've graduated school since then. Indeed, very fast quality response! 😂 

> Having said that, maybe it might be useful in some circumstances to have an option on the bluealsa service to accept only one connection at a time. So I am labelling this as a "feature request" (albeit probably too late for the OP)

Well, I'm still here waiting for the feature to drop!
And I'm sure other people will have use cases for this kind of a behavior.
E.g. a conventional bluetooth speaker.

Arkq, 2021-07-09:
I will add some comment to this issue - whether it's a bluez-alsa issue, and how to implement it.

Firstly some notes on what bluez-alsa (in particular any other application which talks to BlueZ via D-Bus) can do in the matter of connecting only one BT device at a time.

BlueALSA service registers A2DP endpoints (codecs) in BlueZ. When BT device connects, a single endpoint is "used" and it will not be possible to use this particular endpoint to connect more A2DP profiles. In order to "fix" that, BlueALSA registers new endpoint just after receiving a connection - so BlueZ has spare endpoint(s) for establishing new A2DP connections. Single endpoint defines a single codec. So, in case when application registers endpoints for SBC, AAC and e.g. apt-X, there is 3 endpoints in BlueZ - it will be possible to connect 3 BT devices, but these devices will have to use different codecs. User will not be able to connect two devices which supports SBC only. So with two cheap BT headsets (SBC only), it will appear as if there would be possible to connect only one BT device.

In theory, application (e.g. BlueALSA) can reject A2DP configuration when there is an incoming connection. But, that will block A2DP profile. What about other profiles? In the end one might end up with two devices connected, but only one of these will have A2DP connected. Will that satisfy you? I think that it will not, because you might have AVRCP connected from two devices, so you will receive button events like volume up/down, next/previous from both devices.

So, according to my current knowledge, if one would want to be able to connect only one BT device, the necessary changes will have to be done in BlueZ, not BlueALSA. But if I'm wrong, please correct me.

borine, 2021-09-17:
As this request appears to require changes to Bluez before it becomes feasible within BlueALSA, I see no point in leaving it open here for possibly another 3 years without any progress. If someone has a definite requirement for the feature requested here, **and** the time to be able to work with the Bluez developers to properly define the requirement (given that it impacts other profiles, not just the ones handled by BlueALSA) **and** then see it through to implementation, then please raise with the Bluez project. If the outcome of that is successful, please raise a new request here so that this project can use the new Bluez feature.

closed 2021-09-17 by borine