arkq/bluez-alsa #136: after stream audio to BT speaker, aplay does not quit

subjectxbj, 2018-08-03:
Hi Arkq,

I'm using bluealsa to stream music from my device to a JBL speaker.

I use "aplay" to stream music. generally aplay will exit after the audio is finished streaming. But sometimes, aplay just hung up even the stream is finished. This happens about 1 out of 5 times.

I put the log of alsa plugin and bluealsa together. 

Following is the log when aplay can exit normally:
···
[alexa] / # aplay /etc/alexa/sounds-bakcup/med_ui_wakesound.wav
../../../src/asound/../shared/ctl-client.c:102: Connecting to socket: /var/run/bluealsa/hci0
../../../src/asound/../shared/ctl-client.c:214: Getting transport for 00:12:6F:66:98:B7 type 1
bluealsa: ../../src/ctl.c:609: New client accepted: 13
bluealsa: ../../src/ctl.c:630: +-+-
bluealsa: ../../src/ctl.c:630: +-+-
../../../src/asound/bluealsa-pcm.c:547: Setting constraints
Playing WAVE '/etc/alexa/sounds-bakcup/med_ui_wakesound.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
../../../src/asound/bluealsa-pcm.c:295: Initializing HW
../../../src/asound/../shared/ctl-client.c:341: Requesting PCM open for 00:12:6F:66:98:B7
bluealsa: ../../src/ctl.c:342: PCM requested for 00:12:6F:66:98:B7 type 1 stream 0
bluealsa: ../../src/transport.c:836: New transport: 16 (MTU: R:672 W:895)
bluealsa: ../../src/ctl.c:630: +-+-
../../../src/asound/bluealsa-pcm.c:318: FIFO buffer size: 4096
../../../src/asound/bluealsa-pcm.c:324: Selected HW buffer: 6 periods x 16380 bytes <= 98284 bytes
../../../src/asound/bluealsa-pcm.c:339: Initializing SW
../../../src/asound/bluealsa-pcm.c:339: Initializing SW
../../../src/asound/bluealsa-pcm.c:339: Initializing SW
../../../src/asound/bluealsa-pcm.c:339: Initializing SW
../../../src/asound/bluealsa-pcm.c:355: Prepared
../../../src/asound/bluealsa-pcm.c:339: Initializing SW
bluealsa: ../../src/bluez.c:1034: Signal: PropertiesChanged: org.bluez.MediaTransport1
bluealsa: ../../src/transport.c:692: State transition: 0 -> 2
bluealsa: ../../../src/asound/bluealsa-pcm.c:226: Starting
../../src/io.c:366: Starting IO loop: A2DP Source (SBC)
../../../src/asound/../shared/ctl-client.c:409: Requesting PCM resume for 00:12:6F:66:98:B7bluealsa:
../../src/transport.c:104: Created new IO thread: A2DP Source (SBC)
bluealsa: ../../src/transport.c:692: State transition: 2 -> 2
bluealsa: ../../src/ctl.c:630: +-+-
../../../src/asound/bluealsa-pcm.c:121: Starting IO loop
../../../src/asound/../shared/ctl-client.c:434: Requesting PCM drain for 00:12:6F:66:98:B7
../../../src/asound/bluealsa-pcm.c:132: IO thread paused: 4
bluealsa: ../../src/transport.c:794: PCM drained
bluealsa: ../../src/ctl.c:630: +-+-
../../../src/asound/bluealsa-pcm.c:266: Stopping
../../../src/asound/bluealsa-pcm.c:266: Stopping
../../../src/asound/bluealsa-pcm.c:331: Freeing HW
../../../src/asound/../shared/ctl-client.c:383: Closing PCM for 00:12:6F:66:98:B7
bluealsa: ../../src/ctl.c:440: PCM close for 00:12:6F:66:98:B7 type 1 stream 0
bluealsa: ../../src/transport.c:862: Releasing transport: A2DP Source (SBC)
bluealsa: ../../src/bluez.c:1034: Signal: PropertiesChanged: org.bluez.MediaTransport1
bluealsa: ../../src/transport.c:692: State transition: 2 -> 0
bluealsa: ../../src/transport.c:890: Closing BT: 16
bluealsa: ../../src/transport.c:1014: Exiting IO thread
bluealsa: ../../src/transport.c:988: Closing PCM: 14
bluealsa: ../../src/ctl.c:630: +-+-
../../../src/asound/bluealsa-pcm.c:284: Closing plugin
[alexa] / # bluealsa: ../../src/ctl.c:580: Client closed connection: 13
bluealsa: ../../src/ctl.c:630: +-+-

[alexa] / #

···

And following is the log when "aplay" hung up and can't quit:
```
[alexa] / # aplay /etc/alexa/sounds-bakcup/med_ui_wakesound.wav
../../../src/asound/../shared/ctl-client.c:102: Connecting to socket: /var/run/bluealsa/hci0
../../../src/asound/../shared/ctl-client.c:214: Getting transport for 00:12:6F:66:98:B7 type 1
bluealsa: ../../src/ctl.c:609: New client accepted: 13
bluealsa: ../../src/ctl.c:630: +-+-
bluealsa: ../../src/ctl.c:630: +-+-
../../../src/asound/bluealsa-pcm.c:547: Setting constraints
Playing WAVE '/etc/alexa/sounds-bakcup/med_ui_wakesound.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
../../../src/asound/bluealsa-pcm.c:295: Initializing HW
../../../src/asound/../shared/ctl-client.c:341: Requesting PCM open for 00:12:6F:66:98:B7
bluealsa: ../../src/ctl.c:342: PCM requested for 00:12:6F:66:98:B7 type 1 stream 0
bluealsa: ../../src/transport.c:836: New transport: 16 (MTU: R:672 W:895)
bluealsa: ../../src/ctl.c:630: +-+-
../../../src/asound/bluealsa-pcm.c:318: FIFO buffer size: 4096
../../../src/asound/bluealsa-pcm.c:324: Selected HW buffer: 6 periods x 16380 bytes <= 98284 bytes
../../../src/asound/bluealsa-pcm.c:339: Initializing SW
../../../src/asound/bluealsa-pcm.c:339: Initializing SW
../../../src/asound/bluealsa-pcm.c:339: Initializing SW
../../../src/asound/bluealsa-pcm.c:339: Initializing SW
../../../src/asound/bluealsa-pcm.c:355: Prepared
../../../src/asound/bluealsa-pcm.c:339: Initializing SW
../../../src/asound/bluealsa-pcm.c:226: Starting
../../../src/asound/../shared/ctl-client.c:409: Requesting PCM resume for 00:12:6F:66:98:B7
bluealsa: ../../src/transport.c:692: State transition: 0 -> 2
bluealsa: ../../src/io.c:366: Starting IO loop: A2DP Source (SBC)
bluealsa: ../../src/transport.c:104: Created new IO thread: A2DP Source (SBC)
bluealsa: ../../src/ctl.c:630: +-+-
../../../src/asound/bluealsa-pcm.c:121: Starting IO loop
../../../src/asound/../shared/ctl-client.c:434: Requesting PCM drain for 00:12:6F:66:98:B7
../../../src/asound/bluealsa-pcm.c:132: IO thread paused: 4
bluealsa: ../../src/transport.c:794: PCM drained
bluealsa: ../../src/ctl.c:630: +-+-
../../../src/asound/bluealsa-pcm.c:266: Stopping
../../../src/asound/bluealsa-pcm.c:266: Stopping
../../../src/asound/bluealsa-pcm.c:331: Freeing HW
../../../src/asound/../shared/ctl-client.c:383: Closing PCM for 00:12:6F:66:98:B7
bluealsa: ../../src/ctl.c:440: PCM close for 00:12:6F:66:98:B7 type 1 stream 0
bluealsa: ../../src/transport.c:862: Releasing transport: A2DP Source (SBC)
bluealsa: ../../src/transport.c:890: Closing BT: 16
bluealsa: ../../src/transport.c:1014: Exiting IO thread
bluealsa: ../../src/bluez.c:1034: Signal: PropertiesChanged: org.bluez.MediaTransport1
bluealsa: ../../src/transport.c:692: State transition: 2 -> 2
bluealsa: ../../src/bluez.c:1034: Signal: PropertiesChanged: org.bluez.MediaTransport1
bluealsa: ../../src/transport.c:692: State transition: 2 -> 0
```
It looks like the plugin "bluealsa-pcm" is trying to send message to bluealsa to close PCM for the BT speaker, but blocked on waiting for the response.

While bluealsa received to request, but failed to end up the io thread.

Arkq, 2018-08-03:
Hi,

Logs generated by bluez-alsa project are generally OK, however in this case these logs are not sufficient. Could you provide `uftrace.data` for case when bluealsa hangs?

You will have to configure bluez-alsa project with CFLAGS=-pg option, e.g.:
`../configure CFLAGS=-pg --enable-debug --enable-aac [some other options]`

Then, run `bluealsa` as follows: `uftrace record <path_to_bluealsa> [options]`. In the current directory there will be created subdirectory named `uftrace.data` which will contain debug information. Pack it and attach to this issue. The [`uftrace`](https://github.com/namhyung/uftrace) program records all function calls, so it might be huge if you will play big audio file, so I hope, that your "test audio" is only few seconds long :D

subjectxbj, 2018-08-06:
Unforturenately my platform is alpine. the toolchain has problem building application with "-pg".

subjectxbj, 2018-08-06:
From the log,seem the history is like this:
[ctl thread]ctl_thread_cmd_pcm_close() 
[ctl thread]    _transport_release
[ctl thread]	    pthread_cancel
[ctl thread]			transport_pthread_cleanup
[ctl thread]				transport_release_bt_a2dp
[ctl thread]				t->thread = config.main_thread;
[Dbus callback]bluez_signal_transport_changed
[Dbus callback]	transport_set_state_from_string
[Dbus callback]		transport_set_state TRANSPORT_ACTIVE to TRANSPORT_ACTIVE
[Dbus callback]			io_thread_create(t->thread);
[ctl thread]		pthread_join(t->thread)                       ---> since t->thread is created in the callback, pthread_join will block.
[ctl thread!]	transport_send_signal                             ---> _transport_release is blocked, will never get chance to move to this function.
[Dbus callback]bluez_signal_transport_changed
[Dbus callback]	transport_set_state_from_string
[Dbus callback]		transport_set_state TRANSPORT_ACTIVE to TRANSPORT_IDLE
[Dbus callback]			pthread_cancel(t->thread);
[Dbus callback]			pthread_join(t->thread);

The point is "t->thread" seems not protected by a mutex, so that both the ctl thread and the Dbus callback can access it at the same time. I noticed there is a bluealsa created a lock "config.devices_mutex". ctl thread will obtain it during operation, but Dbus callback never use it. So I think Dbus callback need to lock it before make any change. What do you think?

subjectxbj, 2018-08-06:
Also I have another question:
Regarding the following operation:
```
pthread_cancel(t->thread);
pthread_join(t->thread, NULL);
```
After pthread_cancel(), t->thread is changed to "config.main_thread" in transport_pthread_cleanup() , Then the following pthread_join() will collect the TID of the "config.main_thread". Shouldn't it collect the TID of the IO thread? Is this the attended design or is it a bug here?

subjectxbj, 2018-08-06:
The difference is:
If the IO thread is terminated by DBUS callback, then the plugin can receive the response to the close request, and quit correctly. 

But if the IO thread is terminated by the plugin request, then bluealsa will have problem sending out the response.

Arkq, 2018-08-06:
Wow, thx for such an extensive debugging. I will look into this in the next few days. You might be right that responsible for this issue is cancel&join coupling, because t->thread assignment in the cleanup function was not there as a part of the "original" design. I've added it there in order to fix some bug and apparently I've introduced another.In your "history log" you've got two lines which worries me. The dbus thread sets state from active to active and then creates new thread... which should be kinda impossible, because set state from string function check wether status is NOT the same as before. However, I will check such a scenario as well.

Arkq, 2018-08-14:
Please, check whether branch [thread-cj-fix](https://github.com/Arkq/bluez-alsa/tree/thread-cj-fix) resolves this issue for you.

commit 4078f145 (2018-08-20): Fix thread cancel & join coupling

Thread cancellation cleanup routine modifies self thread ID, so one
can not use referenced t->thread value for both cancel & join calls.

Fixes #136
closed 2018-08-20 by Arkq