arkq/bluez-alsa #137: Bluealsa not working after update - error: a2dp-source profile connect failed for protocol not available

henry-shadowjet, 2018-08-24:
Hi fellow BlueALSA users,

I've been extremely happy since using Raspbian Stretch on my Raspberry Pi 3 Model B+ with the Bluealsa A2DP instead of using the buggy and slow Pulseaudio.

I've had my setup working for my Pi to play audio from my phone using the A2DP and playback through bluealsa-play.

However since the large update I did yesterday (23-08-2018), which from memory includes a kernel update and a bunch of other programs, my bluetooth stopped working properly as described above.

First I noticed that I just stopped being able to connect to the Pi. And when I try to connect using bluetoothctl from the Pi side I get "org.bluez.error.Failed". That is typically when I didn't have pulseaudio activated in the old days. I stopped getting this error since bluealsa, but now it's back!

I then tried to unpair and re-pair the bluetooth from phone to the Pi. Went thru the pairing process smoothly, except now the services available doesn't include media playback anymore (as seen by the phone).

I looked at bluetooth.service and it threw the error as follows:
a2dp-source profile connect failed for protocol not available

I tried to reinstall bluealsa and bluetooth through apt-get, still doesn't work.

I'd appreciate any help. Thanks in advance

Henry

Arkq, 2018-08-25:
Hi,I don't know how big update there was, but there is one incompatibility since last version (in bluealsa). Please, check if bluealsa is working: pgrep -a bluealsa, if not, the problem is with startup script - it might use old option not supported by bluealsa right now. Run 'bluealsa -h' for more information.-------- Original Message --------Subject: [Arkq/bluez-alsa] Bluealsa not working after update - error: a2dp-source profile connect failed for protocol not available (#137)From: henry-shadowjet To: Arkq/bluez-alsa CC: Subscribed Hi fellow BlueALSA users,
I've been extremely happy since using Raspbian Stretch on my Raspberry Pi 3 Model B+ with the Bluealsa A2DP instead of using the buggy and slow Pulseaudio.
I've had my setup working for my Pi to play audio from my phone using the A2DP and playback through bluealsa-play.
However since the large update I did yesterday (23-08-2018), which from memory includes a kernel update and a bunch of other programs, my bluetooth stopped working properly as described above.
First I noticed that I just stopped being able to connect to the Pi. And when I try to connect using bluetoothctl from the Pi side I get "org.bluez.error.Failed". That is typically when I didn't have pulseaudio activated in the old days. I stopped getting this error since bluealsa, but now it's back!
I then tried to unpair and re-pair the bluetooth from phone to the Pi. Went thru the pairing process smoothly, except now the services available doesn't include media playback anymore (as seen by the phone).
I looked at bluetooth.service and it threw the error as follows:
a2dp-source profile connect failed for protocol not available
I tried to reinstall bluealsa and bluetooth through apt-get, still doesn't work.
I'd appreciate any help. Thanks in advance
Henry

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/Arkq/bluez-alsa/issues/137",
"url": "https://github.com/Arkq/bluez-alsa/issues/137",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
},
{
"@type": "MessageCard",
"@context": "http://schema.org/extensions",
"hideOriginalBody": "false",
"originator": "AF6C5A86-E920-430C-9C59-A73278B5EFEB",
"title": "Bluealsa not working after update - error: a2dp-source profile connect failed for protocol not available (#137)",
"sections": [
{
"text": "",
"activityTitle": "**henry-shadowjet**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@henry-shadowjet",
"facts": [
{
"name": "Repository: ",
"value": "Arkq/bluez-alsa"
},
{
"name": "Issue #: ",
"value": 137
}
]
}
],
{
"name": "Add a comment",
"@type": "ActionCard",
"inputs": [
{
"isMultiLine": true,
"@type": "TextInput",
"id": "IssueComment",
"isRequired": false
}
],
"actions": [
{
"name": "Comment",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"Arkq/bluez-alsa\",\n\"issueId\": 137,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
}
]
},
{
"name": "Close issue",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueClose\",\n\"repositoryFullName\": \"Arkq/bluez-alsa\",\n\"issueId\": 137\n}"
},
{
"targets": [
{
"os": "default",
"uri": "https://github.com/Arkq/bluez-alsa/issues/137"
}
],
"@type": "OpenUri",
"name": "View on GitHub"
},
{
"name": "Unsubscribe",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 372251396\n}"
}
],
"themeColor": "26292E"
}
]

henry-shadowjet, 2018-08-25:
Hi Arkq,

Thanks for your quick response.

The funny thing is bluealsa is running. Pidof bluealsa or pgrep bluealsa both return a PID, and so does systemctl status bluealsa.service. I stopped the service and run it manually as well, it runs, but nothing changes.

I tried to also start from fresh install. The bluealsa worked, as it usually would. As soon as I updated through apt, same problem.
Now I installed a fresh copy again and just refrain from updating, but it's just not ideal :(

Thanks for your help though ;)

jbrowne200, 2018-08-27:
Also experiencing this issue, my two Raspberry Pi's were working flawlessly as bluetooth receiver's using Bluealsa, and since updating both the other day neither are working. My devices can connect to the Pi, but the connection is cancelled only seconds later and no audio streaming is possible.

Hoping this issue is resolved soon as was really enjoying the feature.

Arkq, 2018-08-28:
I think that the problem is cause by recent bluealsa changes (according to [this](https://github.com/Arkq/bluez-alsa/issues/59#issuecomment-416547840) comment). By default not all BT profiles are enabled. In order to enable A2DP Sink, one has to specify this profile on the command line. So, it seems that the startup script has to be changed from `bluealsa` to `bluealsa -p a2dp-source -p a2dp-sink`. See `bluealsa --help` for details.

henry-shadowjet, 2018-09-03:
I tried your suggestion, Arkq and it worked! Finally dared to update bluealsa again after having some time in the weekend. The -p flags really fixed it! Thanks!

Arkq, 2018-10-11:
Issue seems to be resolved, I'm closing it then.

closed 2018-10-11 by Arkq