arkq/bluez-alsa #138: socket lost signal when receive AVDTP cmd

Devilliv, 2018-08-30:
Hi @Arkq and guys!

I am using bluez-alsa with bluez-5.38 to make a2dp-sink works. Then I came across a problem, when the interval between a  START_CMD and a SUSPEND_CMD is too short, bluealsa may lose the PropertiesChanged signal, it then leads to a malfunction.

Here is the log:
normal:
bluetoothd[1345]: profiles/audio/avdtp.c:session_cb() 
bluetoothd[1345]: profiles/audio/avdtp.c:avdtp_parse_cmd() Received START_CMD
bluetoothd[1345]: profiles/audio/a2dp.c:start_ind() Sink 0x11b5c00: Start_Ind
bluetoothd[1345]: profiles/audio/avdtp.c:avdtp_ref() 0x11b6d70: ref=3
bluetoothd[1345]: profiles/audio/avdtp.c:avdtp_sep_set_state() stream state changed: OPEN -> STREAMING
bluetoothd[1345]: profiles/audio/source.c:source_set_state() State changed /org/bluez/hci0/dev_00_E0_4C_23_99_87: SOURCE_STATE_CONNECTED -> SOURCE_STATE_PLAYING
bluetoothd[1345]: profiles/audio/transport.c:transport_update_playing() /org/bluez/hci0/dev_00_E0_4C_23_99_87/fd0 State=TRANSPORT_STATE_IDLE Playing=1
bluetoothd[1345]: profiles/audio/transport.c:transport_set_state() State changed /org/bluez/hci0/dev_00_E0_4C_23_99_87/fd0: TRANSPORT_STATE_IDLE -> TRANSPORT_STATE_PENDING
/usr/bin/bluealsa: bluez.c:1032: Signal: PropertiesChanged: org.bluez.MediaTransport1
/usr/bin/bluealsa: transport.c:691: State transition: 0 -> 1
bluetoothd[1345]: profiles/audio/transport.c:media_owner_create() Owner created: sender=:1.1
bluetoothd[1345]: profiles/audio/a2dp.c:a2dp_sep_lock() SEP 0x11b5c00 locked
bluetoothd[1345]: profiles/audio/avdtp.c:avdtp_ref() 0x11b6d70: ref=4
bluetoothd[1345]: profiles/audio/a2dp.c:setup_ref() 0x11ba360: ref=1
bluetoothd[1345]: profiles/audio/avdtp.c:avdtp_unref() 0x11b6d70: ref=3
bluetoothd[1345]: profiles/audio/transport.c:media_request_create() Request created: method=TryAcquire id=4
bluetoothd[1345]: profiles/audio/transport.c:media_owner_add() Owner :1.1 Request TryAcquire
bluetoothd[1345]: profiles/audio/transport.c:media_transport_set_owner() Transport /org/bluez/hci0/dev_00_E0_4C_23_99_87/fd0 Owner :1.1
/usr/bin/bluealsa: transport.c:835: New transport: 11 (MTU: R:672 W:1008)
bluetoothd[1345]: profiles/audio/transport.c:media_owner_remove() Owner :1.1 Request TryAcquire
bluetoothd[1345]: profiles/audio/transport.c:transport_set_state() State changed /org/bluez/hci0/dev_00_E0_4C_23_99_87/fd0: TRANSPORT_STATE_PENDING -> TRANSPORT_STATE_ACTIVE
bluetoothd[1345]: profiles/audio/a2dp.c:setup_unref() 0x11ba360: ref=0
bluetoothd[1345]: profiles/audio/a2dp.c:setup_free() 0x11ba360
bluetoothd[1345]: profiles/audio/avdtp.c:avdtp_unref() 0x11b6d70: ref=2
/usr/bin/bluealsa: bluez.c:1032: Signal: PropertiesChanged: org.bluez.MediaTransport1
/usr/bin/bluealsa: transport.c:691: State transition: 1 -> 2
/usr/bin/bluealsa: transport.c:104: Created new IO thread: A2DP Sink
/usr/bin/bluealsa: io.c:225: Starting IO loop: A2DP Sink (SBC)
bluetoothd[1345]: profiles/audio/player.c:media_player_set_position() 4294967295
bluetoothd[1345]: profiles/audio/player.c:media_player_set_status() playing
bluetoothd[1345]: profiles/audio/player.c:media_player_set_duration() 295358
bluetoothd[1345]: profiles/audio/player.c:media_player_set_position() 4294967295
bluetoothd[1345]: profiles/audio/player.c:media_player_set_status() playing
bluealsa-aplay: aplay.c:443: An underrun has occurred
bluetoothd[1345]: profiles/audio/player.c:media_player_set_position() 4294967295
bluetoothd[1345]: profiles/audio/player.c:media_player_set_status() paused
bluetoothd[1345]: profiles/audio/player.c:media_player_set_duration() 295358
bluetoothd[1345]: profiles/audio/player.c:media_player_set_position() 4294967295
bluetoothd[1345]: profiles/audio/player.c:media_player_set_status() paused
bluetoothd[1345]: profiles/audio/avdtp.c:session_cb() 
bluetoothd[1345]: profiles/audio/avdtp.c:avdtp_parse_cmd() Received SUSPEND_CMD
bluetoothd[1345]: profiles/audio/a2dp.c:suspend_ind() Sink 0x11b5c00: Suspend_Ind
bluetoothd[1345]: profiles/audio/avdtp.c:avdtp_sep_set_state() stream state changed: STREAMING -> OPEN
bluetoothd[1345]: profiles/audio/source.c:source_set_state() State changed /org/bluez/hci0/dev_00_E0_4C_23_99_87: SOURCE_STATE_PLAYING -> SOURCE_STATE_CONNECTED
bluetoothd[1345]: profiles/audio/transport.c:transport_update_playing() /org/bluez/hci0/dev_00_E0_4C_23_99_87/fd0 State=TRANSPORT_STATE_ACTIVE Playing=0
bluetoothd[1345]: profiles/audio/transport.c:media_transport_remove_owner() Transport /org/bluez/hci0/dev_00_E0_4C_23_99_87/fd0 Owner :1.1
bluetoothd[1345]: profiles/audio/transport.c:media_owner_free() Owner :1.1
bluetoothd[1345]: profiles/audio/transport.c:transport_set_state() State changed /org/bluez/hci0/dev_00_E0_4C_23_99_87/fd0: TRANSPORT_STATE_ACTIVE -> TRANSPORT_STATE_IDLE
bluetoothd[1345]: profiles/audio/a2dp.c:a2dp_sep_unlock() SEP 0x11b5c00 unlocked
/usr/bin/bluealsa: bluez.c:1032: Signal: PropertiesChanged: org.bluez.MediaTransport1
/usr/bin/bluealsa: transport.c:691: State transition: 2 -> 0
/usr/bin/bluealsa: transport.c:862: Releasing transport: A2DP Sink (SBC)
/usr/bin/bluealsa: transport.c:890: Closing BT: 11
/usr/bin/bluealsa: transport.c:1014: Exiting IO thread
bluealsa-aplay: aplay.c:348: Device marked as inactive: 00:E0:4C:23:99:87

abnormal:
bluetoothd[1345]: profiles/audio/avdtp.c:session_cb() 
bluetoothd[1345]: profiles/audio/avdtp.c:avdtp_parse_cmd() Received START_CMD
bluetoothd[1345]: profiles/audio/a2dp.c:start_ind() Sink 0x11b5c00: Start_Ind
bluetoothd[1345]: profiles/audio/avdtp.c:avdtp_ref() 0x11b6d70: ref=3
bluetoothd[1345]: profiles/audio/avdtp.c:avdtp_sep_set_state() stream state changed: OPEN -> STREAMING
bluetoothd[1345]: profiles/audio/source.c:source_set_state() State changed /org/bluez/hci0/dev_00_E0_4C_23_99_87: SOURCE_STATE_CONNECTED -> SOURCE_STATE_PLAYING
bluetoothd[1345]: profiles/audio/transport.c:transport_update_playing() /org/bluez/hci0/dev_00_E0_4C_23_99_87/fd0 State=TRANSPORT_STATE_IDLE Playing=1
bluetoothd[1345]: profiles/audio/transport.c:transport_set_state() State changed /org/bluez/hci0/dev_00_E0_4C_23_99_87/fd0: TRANSPORT_STATE_IDLE -> TRANSPORT_STATE_PENDING
bluetoothd[1345]: profiles/audio/player.c:media_player_set_position() 4294967295
bluetoothd[1345]: profiles/audio/player.c:media_player_set_status() playing
bluetoothd[1345]: profiles/audio/player.c:media_player_set_duration() 295358
bluetoothd[1345]: profiles/audio/player.c:media_player_set_position() 4294967295
bluetoothd[1345]: profiles/audio/player.c:media_player_set_status() playing
bluetoothd[1345]: profiles/audio/avdtp.c:avdtp_unref() 0x11b6d70: ref=2
bluetoothd[1345]: profiles/audio/avdtp.c:session_cb() 
bluetoothd[1345]: profiles/audio/avdtp.c:avdtp_parse_resp() SUSPEND request succeeded
bluetoothd[1345]: profiles/audio/avdtp.c:avdtp_sep_set_state() stream state changed: STREAMING -> OPEN
bluetoothd[1345]: profiles/audio/source.c:source_set_state() State changed /org/bluez/hci0/dev_00_E0_4C_23_99_87: SOURCE_STATE_PLAYING -> SOURCE_STATE_CONNECTED
bluetoothd[1345]: profiles/audio/transport.c:transport_update_playing() /org/bluez/hci0/dev_00_E0_4C_23_99_87/fd0 State=TRANSPORT_STATE_PENDING Playing=0
bluetoothd[1345]: profiles/audio/transport.c:transport_set_state() State changed /org/bluez/hci0/dev_00_E0_4C_23_99_87/fd0: TRANSPORT_STATE_PENDING -> TRANSPORT_STATE_IDLE
bluetoothd[1345]: profiles/audio/a2dp.c:suspend_cfm() Sink 0x11b5c00: Suspend_Cfm

When this problem occurs, the socket connected to /var/run/bluealsa/hci0 still remains even though the source device has already disconnect.

I also try pulseaudio with bluez-5.38, it doesn't have this problem. Could you please give me some advices? Thank you!

Arkq, 2018-08-30:
In your second log there is no output from bluealsa. Have you stripped it, or it was not produced? In general bluealsa relies on DBus signals provided by bluez, please check whether such a signal was produced: `dbus-monitor --system`. If `PropertiesChanged` signal was not emitted by bluez, I'm afraid there is nothing I can do...

> when the interval between a START_CMD and a SUSPEND_CMD is too short

Is there any way to reproduce this "short interval"? I would be much easier for me if I was able to see this issue by myself as well :)

Devilliv, 2018-08-31:
HI @Arkq,
> In your second log there is no output from bluealsa. Have you stripped it, or it was not produced? 

I didn't strip it, that's what I mean 'may lose the signal'.

According to your advice, I have checked using 'dbus-monitor'. Here comes the log: (I only shows the important part of it.)

bluez & bluealsa:
`emon.debug bluetoothd[1355]: profiles/audio/avdtp.c:session_cb() 
Thu Aug 30 17:13:02 2018 daemon.debug bluetoothd[1355]: profiles/audio/avdtp.c:avdtp_parse_cmd() Received START_CMD
Thu Aug 30 17:13:02 2018 daemon.debug bluetoothd[1355]: profiles/audio/a2dp.c:start_ind() Sink 0x1669880: Start_Ind
Thu Aug 30 17:13:02 2018 daemon.debug bluetoothd[1355]: profiles/audio/avdtp.c:avdtp_ref() 0x1671e10: ref=3
Thu Aug 30 17:13:02 2018 daemon.debug bluetoothd[1355]: profiles/audio/avdtp.c:avdtp_sep_set_state() stream state changed: OPEN -> STREAMING
Thu Aug 30 17:13:02 2018 daemon.debug bluetoothd[1355]: profiles/audio/source.c:source_set_state() State changed /org/bluez/hci0/dev_00_E0_4C_23_99_87: SOURCE_STATE_CONNECTED -> SOURCE_STATE_PLAYING
Thu Aug 30 17:13:02 2018 daemon.debug bluetoothd[1355]: profiles/audio/transport.c:transport_update_playing() /org/bluez/hci0/dev_00_E0_4C_23_99_87/fd3 State=TRANSPORT_STATE_IDLE Playing=1
Thu Aug 30 17:13:02 2018 daemon.debug bluetoothd[1355]: profiles/audio/transport.c:transport_set_state() State changed /org/bluez/hci0/dev_00_E0_4C_23_99_87/fd3: TRANSPORT_STATE_IDLE -> TRANSPORT_STATE_PENDING
Thu Aug 30 17:13:02 2018 daemon.debug bluetoothd[1355]: profiles/audio/transport.c:media_owner_create() Owner created: sender=:1.2
Thu Aug 30 17:13:02 2018 daemon.debug bluetoothd[1355]: profiles/audio/a2dp.c:a2dp_sep_lock() SEP 0x1669880 locked
Thu Aug 30 17:13:02 2018 daemon.debug bluetoothd[1355]: profiles/audio/avdtp.c:avdtp_ref() 0x1671e10: ref=4
Thu Aug 30 17:13:02 2018 daemon.debug bluetoothd[1355]: profiles/audio/a2dp.c:setup_ref() 0x1674740: ref=1
Thu Aug 30 17:13:02 2018 daemon.debug bluetoothd[1355]: profiles/audio/avdtp.c:avdtp_unref() 0x1671e10: ref=3
Thu Aug 30 17:13:02 2018 daemon.debug bluetoothd[1355]: profiles/audio/transport.c:media_request_create() Request created: method=TryAcquire id=19
Thu Aug 30 17:13:02 2018 daemon.debug bluetoothd[1355]: profiles/audio/transport.c:media_owner_add() Owner :1.2 Request TryAcquire
Thu Aug 30 17:13:02 2018 daemon.debug bluetoothd[1355]: profiles/audio/transport.c:media_transport_set_owner() Transport /org/bluez/hci0/dev_00_E0_4C_23_99_87/fd3 Owner :1.2
Thu Aug 30 17:13:02 2018 daemon.debug bluetoothd[1355]: profiles/audio/transport.c:media_owner_remove() Owner :1.2 Request TryAcquire
Thu Aug 30 17:13:02 2018 daemon.debug bluetoothd[1355]: profiles/audio/transport.c:transport_set_state() State changed /org/bluez/hci0/dev_00_E0_4C_23_99_87/fd3: TRANSPORT_STATE_PENDING -> TRANSPORT_STATE_ACTIVE
Thu Aug 30 17:13:02 2018 daemon.debug bluetoothd[1355]: profiles/audio/a2dp.c:setup_unref() 0x1674740: ref=0
Thu Aug 30 17:13:02 2018 daemon.debug bluetoothd[1355]: profiles/audio/a2dp.c:setup_free() 0x1674740
Thu Aug 30 17:13:02 2018 daemon.debug bluetoothd[1355]: profiles/audio/avdtp.c:avdtp_unref() 0x1671e10: ref=2
Thu Aug 30 17:13:03 2018 daemon.debug bluetoothd[1355]: profiles/audio/avdtp.c:session_cb() 
Thu Aug 30 17:13:03 2018 daemon.debug bluetoothd[1355]: profiles/audio/avdtp.c:avdtp_parse_cmd() Received SUSPEND_CMD
Thu Aug 30 17:13:03 2018 daemon.debug bluetoothd[1355]: profiles/audio/a2dp.c:suspend_ind() Sink 0x1669880: Suspend_Ind
Thu Aug 30 17:13:03 2018 daemon.debug bluetoothd[1355]: profiles/audio/avdtp.c:avdtp_sep_set_state() stream state changed: STREAMING -> OPEN
Thu Aug 30 17:13:03 2018 daemon.debug bluetoothd[1355]: profiles/audio/source.c:source_set_state() State changed /org/bluez/hci0/dev_00_E0_4C_23_99_87: SOURCE_STATE_PLAYING -> SOURCE_STATE_CONNECTED
Thu Aug 30 17:13:03 2018 daemon.debug bluetoothd[1355]: profiles/audio/transport.c:transport_update_playing() /org/bluez/hci0/dev_00_E0_4C_23_99_87/fd3 State=TRANSPORT_STATE_ACTIVE Playing=0
Thu Aug 30 17:13:03 2018 daemon.debug bluetoothd[1355]: profiles/audio/transport.c:media_transport_remove_owner() Transport /org/bluez/hci0/dev_00_E0_4C_23_99_87/fd3 Owner :1.2
Thu Aug 30 17:13:03 2018 daemon.debug bluetoothd[1355]: profiles/audio/transport.c:media_owner_free() Owner :1.2
Thu Aug 30 17:13:03 2018 daemon.debug bluetoothd[1355]: profiles/audio/transport.c:transport_set_state() State changed /org/bluez/hci0/dev_00_E0_4C_23_99_87/fd3: TRANSPORT_STATE_ACTIVE -> TRANSPORT_STATE_IDLE
Thu Aug 30 17:13:03 2018 daemon.debug bluetoothd[1355]: profiles/audio/a2dp.c:a2dp_sep_unlock() SEP 0x1669880 unlocked
Thu Aug 30 17:13:04 2018 daemon.debug bluetoothd[1355]: profiles/audio/avdtp.c:session_cb() 
Thu Aug 30 17:13:04 2018 daemon.debug bluetoothd[1355]: profiles/audio/avdtp.c:avdtp_parse_cmd() Received START_CMD
Thu Aug 30 17:13:04 2018 daemon.debug bluetoothd[1355]: profiles/audio/a2dp.c:start_ind() Sink 0x1669880: Start_Ind
Thu Aug 30 17:13:04 2018 daemon.debug bluetoothd[1355]: profiles/audio/avdtp.c:avdtp_ref() 0x1671e10: ref=3
Thu Aug 30 17:13:04 2018 daemon.debug bluetoothd[1355]: profiles/audio/avdtp.c:avdtp_sep_set_state() stream state changed: OPEN -> STREAMING
Thu Aug 30 17:13:04 2018 daemon.debug bluetoothd[1355]: profiles/audio/source.c:source_set_state() State changed /org/bluez/hci0/dev_00_E0_4C_23_99_87: SOURCE_STATE_CONNECTED -> SOURCE_STATE_PLAYING
[... cut ...]
As we can see, a normal state change between a 'START_CMD' and a 'SUSPEND_CMD' is:
`idle -> pending -> active -> idle`
When abnormal, we can still monitor the 'pending' and 'idle' signal, but bluealsa seems not dealing with them. If that information helps?

> Is there any way to reproduce this "short interval"? 

Just my pad (android-4.4 kernel-3.4.39) will produce this "short interval" when opening the audio player, other source devices don't have this scenario. So I have no idea how to reproduce it, hah!

Thank you for your time!

Devilliv, 2018-08-31:
There is one more phenomenon may help:

After it becomes abnormal, when I kill the 'bluealsa' process, it won't succeed. Then I kill it twice, only a few logs come out from bluez:
```
bluetoothd[1597]: profiles/audio/media.c:release_endpoint() sender=:1.7 path=/A2DP/SBC/Sink/1
bluetoothd[1597]: Endpoint unregistered: sender=:1.7 path=/A2DP/SBC/Sink/1
bluetoothd[1597]: profiles/audio/media.c:media_endpoint_destroy() sender=:1.7 path=/A2DP/SBC/Sink/1
bluetoothd[1597]: profiles/audio/avdtp.c:avdtp_unregister_sep() SEP 0x136d880 unregistered: type:1 codec:0 seid:1
bluetoothd[1597]: src/adapter.c:adapter_service_remove() /org/bluez/hci0
bluetoothd[1597]: src/adapter.c:remove_uuid() sending remove uuid command for index 0
bluetoothd[1597]: src/sdpd-service.c:remove_record_from_server() Removing record with handle 0x10006
bluetoothd[1597]: profiles/audio/media.c:release_endpoint() sender=:1.7 path=/A2DP/SBC/Sink/2
bluetoothd[1597]: Endpoint unregistered: sender=:1.7 path=/A2DP/SBC/Sink/2
bluetoothd[1597]: profiles/audio/media.c:media_endpoint_destroy() sender=:1.7 path=/A2DP/SBC/Sink/2
bluetoothd[1597]: profiles/audio/avdtp.c:avdtp_unregister_sep() SEP 0x13752c0 unregistered: type:1 codec:0 seid:2
root@TinaLinux:/# bluetoothd[1597]: src/adapter.c:dev_class_changed_callback() Class: 0x000000
```

In general, bluealsa should also give some prints:
`/usr/bin/bluealsa: main.c:306: Exiting main loop`

Arkq, 2018-08-31:
> After it becomes abnormal, when I kill the 'bluealsa' process, it won't succeed.

OK, that's something. It seems, that some thread has stuck. Please, attach gdb to the stuck bluealsa process and check where each thread is waiting. It should help a lot. It is basically `info thread`, `thread <number>` and `backtrace`.

closed 2018-09-10 by Devilliv