arkq/bluez-alsa #141: HFP re-connected failed

Devilliv, 2018-09-13:
Hi @Arkq ,
It 's me again! I posted an issue here [(click)](https://github.com/Arkq/bluez-alsa/issues/138) and then wondered it was caused by too many debug log.

There is another similar problem when I was using the HFP profile. It was always success when I connected to the test board for the first time. Then if I disconnected and re-connected, it failed.  Here is the log:

connect for the first time, success:
```
/usr/bin/bluealsa: bluez.c:857: Profile method call: org.bluez.Profile1.NewConnection()
/usr/bin/bluealsa: transport.c:691: State transition: 0 -> 2
/usr/bin/bluealsa: transport.c:104: Created new IO thread: HFP Hands-Free
/usr/bin/bluealsa: ctl.c:645: Emitting notification: 0x1 => 11
bluealsa-aplay: aplay.c:642: Fetching available transports
/usr/bin/bluealsa: ctl.c:651: +-+-
/usr/bin/bluealsa: ctl.c:651: +-+-
/usr/bin/bluealsa: bluez.c:797: HFP Hands-Free configured for device 0C:D7:46:35:96:87
/usr/bin/bluealsa: transport.c:691: State transition: 0 -> 2
/usr/bin/bluealsa: transport.c:104: Created new IO thread: HFP Hands-Free
/usr/bin/bluealsa: rfcomm.c:573: Starting RFCOMM loop: HFP Hands-Free
/usr/bin/bluealsa: rfcomm.c:96: Sending AT message: SET: command:+BRSF, value:244
/usr/bin/bluealsa: io.c:1202: Starting IO loop: HFP Hands-Free
bluetoothd[1456]: Can't open input device: No such file or directory (2)
bluetoothd[1456]: AVRCP: failed to init uinput for 0C:D7:46:35:96:87
/usr/bin/bluealsa: at.c:160: AT message: RESP: command:+BRSF, value:2031
/usr/bin/bluealsa: rfcomm.c:114: HFP state transition: 0 -> 1
/usr/bin/bluealsa: at.c:160: AT message: RESP: command:, value:OK
/usr/bin/bluealsa: rfcomm.c:114: HFP state transition: 1 -> 2
/usr/bin/bluealsa: rfcomm.c:96: Sending AT message: SET: command:+BAC, value:1
/usr/bin/bluealsa: at.c:160: AT message: RESP: command:, value:OK
/usr/bin/bluealsa: rfcomm.c:114: HFP state transition: 2 -> 3
/usr/bin/bluealsa: rfcomm.c:96: Sending AT message: TEST: command:+CIND, value:(null)
/usr/bin/bluealsa: at.c:160: AT message: RESP: command:+CIND, value: ("service",(0-1)),("call",(0-1)),("callsetup",(0-3)),("battchg",(0-5)),("signal",(0-5)),("roam",(0-1)),("callheld",(0-2))
/usr/bin/bluealsa: rfcomm.c:114: HFP state transition: 3 -> 4
/usr/bin/bluealsa: at.c:160: AT message: RESP: command:, value:OK
/usr/bin/bluealsa: rfcomm.c:114: HFP state transition: 4 -> 5
/usr/bin/bluealsa: rfcomm.c:96: Sending AT message: GET: command:+CIND, value:(null)
/usr/bin/bluealsa: at.c:160: AT message: RESP: command:+CIND, value: 1,0,0,3,3,0,0
/usr/bin/bluealsa: ctl.c:651: +-+-
/usr/bin/bluealsa: rfcomm.c:114: HFP state transition: 5 -> 6
/usr/bin/bluealsa: at.c:160: AT message: RESP: command:, value:OK
/usr/bin/bluealsa: rfcomm.c:114: HFP state transition: 6 -> 7
/usr/bin/bluealsa: rfcomm.c:96: Sending AT message: SET: command:+CMER, value:3,0,0,1,0
/usr/bin/bluealsa: at.c:160: AT message: RESP: command:, value:OK
/usr/bin/bluealsa: rfcomm.c:114: HFP state transition: 7 -> 8
/usr/bin/bluealsa: rfcomm.c:114: HFP state transition: 8 -> 9
```
disconnect:
```
/usr/bin/bluealsa: rfcomm.c:801: RFCOMM disconnected: Connection reset by peer
/usr/bin/bluealsa: transport.c:914: Closing RFCOMM: 13
/usr/bin/bluealsa: transport.c:350: Freeing transport: HFP Hands-Free (N/A)
bluetoothd[1456]: Unable to get io data for Hands-Free unit: getpeername: Socket not connected (107)
```
re-connect fails:
```
/usr/bin/bluealsa: bluez.c:857: Profile method call: org.bluez.Profile1.NewConnection()
/usr/bin/bluealsa: transport.c:350: Freeing transport: HFP Hands-Free (SBC)
/usr/bin/bluealsa: transport.c:1014: Exiting IO thread

bluetoothd[1456]: Hands-Free unit replied with an error: org.freedesktop.DBus.Error.NoReply, Did not receive a reply. Possible causes include: 
the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
```
But if I removed the '--enable-debug' configure-args, the bug disappeared again. I don't know if you had  experienced that problem either. If I was making something wrong, please tell me, thank you!

Devilliv, 2018-09-18:
Hi @Arkq ,
Current master works! Thank you!
By the way, did the following commit fix that problem? (My master is now on 11 Aug.)
```
Fix thread cancel & join coupling

Thread cancellation cleanup routine modifies self thread ID, so one
can not use referenced t->thread value for both cancel & join calls.

Fixes #136
 master  v1.3.1
@Arkq
Arkq committed on 14 Aug
1 parent 302800c commit 4078f145a6d09228fc55b9509715865d778fb062
```

Arkq, 2018-09-18:
I think, that your problem was caused by calling thread join on itself which causes deadlock. On my machine, this has resulted in pthread_join() returning error. However, in your logs there is no line which might indicate such a workaround (that's why I asked for libc version). So I think that the commit which solved your issue is 51a68eb (the very last one in the master). If you have been using code before 4078f14, this might also caused such a malfunction, so in order to be sure which commit resolved this issue you will have to use `git bisect` :D

Since this issue is resolved, I'm closing it :)

closed 2018-09-18 by Arkq