arkq/bluez-alsa #142: a2dp-sink APT-X ?

github-Alex13, 2018-09-26:
hello, working a2dp-sink APT-X ?

Available BT profiles:
  - a2dp-source Advanced Audio Source (APT-X, SBC)
  - a2dp-sink   Advanced Audio Sink (SBC)

root@raspberrypi:~# bluealsa -h
Usage:
  bluealsa [OPTION]...

Options:
  -h, --help            print this help and exit
  -V, --version         print version and exit
  -S, --syslog          send output to syslog
  -i, --device=hciX     HCI device to use
  -p, --profile=NAME    enable BT profile
  --a2dp-force-mono     force monophonic sound
  --a2dp-force-audio-cd force 44.1 kHz sampling
  --a2dp-volume         control volume natively

Available BT profiles:
  - a2dp-source Advanced Audio Source (APT-X, SBC)
  - a2dp-sink   Advanced Audio Sink (SBC)
  - hfp-hf      Hands-Free (v1.7)
  - hfp-ag      Hands-Free Audio Gateway (v1.7)
  - hsp-hs      Headset (v1.2)
  - hsp-ag      Headset Audio Gateway (v1.2)

By default only output profiles are enabled, which includes A2DP Source and
HSP/HFP Audio Gateways. If one wants to enable other set of profiles, it is
required to explicitly specify all of them using `-p NAME` options.

github-Alex13, 2018-09-26:
raspberry pi zero w

Arkq, 2018-10-04:
> hello, working a2dp-sink APT-X ?

No, APT-X Sink is not available right now. It will be supported in the future, though :)

vsakson, 2019-02-23:
> > hello, working a2dp-sink APT-X ?
> 
> No, APT-X Sink is not available right now. It will be supported in the future, though :)
What about LDAC? Is LDAC sink support also planned?

gearhead, 2019-04-16:
+1 Looking forward to this as well.

lishaman, 2019-06-05:
Looking forward to supports LDAC sink as well!
I test LDAC source is ok...
hh

newmedicine, 2020-09-21:
I would really value aptX-HD support too.

Arkq, 2021-01-16:
The apt-X and apt-X HD sink (decoding) is available in the master brach since commit b11cff0c2fd436e64c98b77943bb20a2d3d0ea77 and LDAC decoding is available since commit d76252bc88f5a5ae26f679d1e45b32d205351e76.

closed 2021-01-16 by Arkq