arkq/bluez-alsa #144: Test Answer/Reject/Terminate a call in HFP

Devilliv, 2018-09-28:
hi @Arkq ,
I am interested in testing 'Answer/Reject/Terminate' behaviors when we are acting as HFP-HF. Is there any applications can do this? If not, could you please give me some advice about how to make it in a standard way? Thank you!

Arkq, 2018-10-04:
Hi,

You might try `bluealsa-rfcomm` which can be used for sending RFCOMM commands to the connected device. For more information see the HFP documentation: https://www.bluetooth.com/specifications/profiles-overview

Devilliv, 2018-10-08:
> You might try bluealsa-rfcomm

Wow, it works! Thank you!
By the way, I need to add '-lncurses' as below:
```
utils/Makefile:

376 bluealsa_rfcomm_LDADD = \
377         -L/xxxxxxxx/usr/lib -lbluetooth \
378         -lreadline -lncurses
```
or it may cause the following error:
```
usr/lib/libreadline.so: undefined reference to `tgoto'
usr/lib/libreadline.so: undefined reference to `tgetflag'
usr/lib/libreadline.so: undefined reference to `PC'
usr/lib/libreadline.so: undefined reference to `tgetstr'
usr/lib/libreadline.so: undefined reference to `UP'
usr/lib/libreadline.so: undefined reference to `tputs'
usr/lib/libreadline.so: undefined reference to `tgetent'
usr/lib/libreadline.so: undefined reference to `BC'
usr/lib/libreadline.so: undefined reference to `tgetnum'
collect2: error: ld returned 1 exit status
```
That seems to be something about the lib dependencies in my environment, does that happen in yours?

Pro-pra, 2018-10-09:
Devilliv, send example use bluealsa-rfcomm please.

Devilliv, 2018-10-09:
@Pro-pra 
You can try it on a HFP-HF side:
`bluealsa-rfcomm -i [hciX] [bdaddr]`
 'hciX' is the hci device in your system, 'bdaddr' is the address of the connected device.
Then you can send raw AT commands to the connected device, for example:
```
ATA -- answer an incoming call
AT+CHUP -- reject/terminate a call
and so on.
```

closed 2018-10-23 by Devilliv