arkq/bluez-alsa #149: bluealsa: Couldn't initialize controller thread: Bad file descriptor

Tutmanichtun, 2018-10-11:
Hi,
i got the above message when i try to start bluealsa on my RaspberryPi 3.
The Error message is the same than in issue [126](https://github.com/Arkq/bluez-alsa/issues/126), but the described solution didn't work for me. Still getting the same message.
Because i want to use the Raspberry as a Bluetooth speaker, i wrote
`BLUEALSA_CONF="-p a2dp-sink"`
into that file.
I followed [this guide](https://gist.github.com/mill1000/74c7473ee3b4a5b13f6325e9994ff84c), and it worked until a few weeks ago, i don't know what happened in the meantime. :/
Regards,
Tutmanichtun

Queequag, 2018-10-17:
It happened to me when bluealsa segfaulted (which I'm still trying to reproduce in gdb). It left the socket in /var/run/bluealsa directory.

A workaround is to check if this directory exists and if it does – remove it and try starting bluealsa again.

Tutmanichtun, 2018-10-18:
Hi,
thanks for your response. I tried deleting that folder, but it didn't work: still getting the same message. :(

Tutmanichtun, 2018-10-21:
Here is the generated Logfile:
[bluealsa.log](https://github.com/Arkq/bluez-alsa/files/2499288/bluealsa.log)
@alejandro-n-rivera : your command works, so that i can connect my phone to the pi, which is recognized as an audio device until the command is running. But i am still noch able to hear anything. :/

alejandro-n-rivera, 2018-10-21:
@Tutmanichtun While that command is running, in a separate terminal tab (ctrl+shift+T) run 
`bluealsa-aplay -vv 00:00:00:00:00:00`. If that works, the comment I made at the bottom of 
[this](https://gist.github.com/mill1000/74c7473ee3b4a5b13f6325e9994ff84c#gistcomment-2385539) thread might be useful.

Tutmanichtun, 2018-10-21:
The script works fine, thanks. :)
Just a very small minor issue: if i log into the pi via SSH, the script is being executed again and therefore no music is been played. Instead, the pi raises an error message: `bluealsa: Transport not available: /org/bluez/hci0/dev_CC_FD_17_E6_9E_FC/fd0`.

Thanks for your support. :)

closed 2018-10-21 by Tutmanichtun

Queequag, 2018-10-21:
Looking at the code there doesn't seem to be any difference between removing the directory and its contents (see `bluealsa_ctl_thread_init` function in `src/ctl.c`).

From the log file it looks like you didn't remove /var/run/bluealsa and it still contained hci0 socket created during previous bluealsa run.

`29722 mkdir("/var/run/bluealsa", 0755)  = -1 EEXIST (File exists)` – it means that the directory `/var/run/bluealsa` exists (it's not fatal, bluealsa tries to create it and if it fails with this error, it assumes it's created and continues)

`29722 bind(3, {sa_family=AF_UNIX, sun_path="/var/run/bluealsa/hci0"}, 110) = -1 EADDRINUSE (Address already in use)` – it means that the hci0 socket file exists

commit c6215713 (2018-11-06): Print verbose errors during ctl initialization

Closes #149