arkq/bluez-alsa #151: Low bitrate as a sink

motet-a, 2018-10-13:
I’m using bluez-alsa as an audio receiver. It works well with my phone (on Android 8.0) with both SBC and AAC, hovewer the out-of-the-box quality with these codecs is terrible. I mean, I think it’s a bit better than HSP/HDP but definitely way too bad for music.

I thought it was some kind of bitrate negotiation issue and I noticed that the quality increases drastically if I remove sampling rates below 44100 Hz in `bluez-a2dp.c`:

https://github.com/Arkq/bluez-alsa/blob/b743de3ad6a2d24ba5ad0585fd61d4b0eacece99/src/bluez-a2dp.c#L119-L125

https://github.com/Arkq/bluez-alsa/blob/b743de3ad6a2d24ba5ad0585fd61d4b0eacece99/src/bluez-a2dp.c#L144-L150

Still not perfect but much better without these low values. 

Then I increased the bitrate since [`AAC_INIT_BITRATE`](https://github.com/Arkq/bluez-alsa/blob/2acf4511b8fce27a88ce5ba9bb3f5e6944ba95e7/src/a2dp-codecs.h#L136-L139) seems to accept numbers up to `0x7fffff` (Is this bits per second?):

https://github.com/Arkq/bluez-alsa/blob/2acf4511b8fce27a88ce5ba9bb3f5e6944ba95e7/src/a2dp-codecs.h#L136-L139

I wrote `AAC_INIT_BITRATE(0xFFFFFF)` here:

https://github.com/Arkq/bluez-alsa/blob/b743de3ad6a2d24ba5ad0585fd61d4b0eacece99/src/bluez-a2dp.c#L135

…and then cymbals sound like real cymbals :slightly_smiling_face: 

So I’ve got a few questions:

- Am I doing it right? I know almost nothing about bluetooth.

- Why does the lowest sampling rate or bitrate get selected? I understand there might be some kind of quality vs. power consumption trade-off but could it be tweakable by the user?

Thanks for this nice library.

Arkq, 2018-10-15:
Hi,

It has never occurred to me to test AAC sink quality... :/ I've totally missed that 0xFFFF (65 kbit/s) as a maximum bitrate. However, setting this value to 0x7FFFFF (8388 kbit/s) is a little bit too much. In practice everything should work just fine, because the bitrate should be **selected** by the A2DP source, not blindly copied from the provided configuration (bluealsa blindly copies this value :/). But in theory, if source is not selecting this value consciously, one might end up with a CBR 8.4 Mbit/s which won't work with Bluetooth :D My headset uses 320 kbit/s as a max, so I think that for now such a value might be a reasonable default (it uses almost all Bluetooth A2DP bandwidth).

As a long run, few changes are required:
- dynamic selection of the maximal bitrate (based on sampling frequency and number of channels)
- potentially a mechanism for an adaptive bitrate (something like LDAC ABR)
- bluealsa expert configuration (via configuration file), which might allow setting bitrate value

I've got some (old) work in progress for configuration file support, so maybe I will go back to this feature. As for bitrate dynamic selection and ABR, right now I'm working on incorporating LDAC ABR into bluealsa, so when it will be done, I might think about AAC bitrate.

motet-a, 2018-10-21:
Thank you very much! I tested your changes over the weekend on a few devices (recent and old Android versions, and iOS 12) and it worked like a charm! I think you can close the issue :slightly_smiling_face:

Arkq, 2018-10-22:
I'm glad to hear that :)

closed 2018-10-22 by Arkq