arkq/bluez-alsa #154: getpeername: Socket not connected (107)

ilya-fedin, 2018-10-19:
Hello! I installed bluez-alsa on Void Linux and I'm trying to connect to my BT speaker, but connection fails.

Output of bluetoothd:
```
bluetoothd[2669]: Bluetooth daemon 5.50
bluetoothd[2669]: Starting SDP server
bluetoothd[2669]: Bluetooth management interface 1.14 initialized
bluetoothd[2669]: Endpoint registered: sender=:1.39 path=/A2DP/SBC/Source/1
bluetoothd[2669]: Unable to get io data for Hands-Free Voice gateway: getpeername: Socket not connected (107)
bluetoothd[2669]: Unable to get connect data for Headset Voice gateway: getpeername: Socket not connected (107)
bluetoothd[2669]: connect error: Host is down (112)
```
Output of bluealsa:
```
bluealsa: bluez.c:675: Registering endpoint: /A2DP/SBC/Source/1
bluealsa: ctl.c:571: Starting controller loop
bluealsa: bluez.c:910: Registering profile: /HSP/AudioGateway
bluealsa: bluez.c:910: Registering profile: /HFP/AudioGateway
bluealsa: main.c:302: Starting main dispatching loop
bluealsa: bluez.c:857: Profile method call: org.bluez.Profile1.NewConnection()
bluealsa: transport.c:688: State transition: 0 -> 2
bluealsa: transport.c:104: Created new IO thread: HFP Audio Gateway
bluealsa: io.c:1195: Starting IO loop: HFP Audio Gateway
bluealsa: ctl.c:675: +-+-
bluealsa: bluez.c:797: HFP Audio Gateway configured for device FF:39:92:00:0E:CE
bluealsa: transport.c:688: State transition: 0 -> 2
bluealsa: transport.c:104: Created new IO thread: HFP Audio Gateway
bluealsa: rfcomm.c:573: Starting RFCOMM loop: HFP Audio Gateway
bluealsa: rfcomm.c:803: RFCOMM disconnected: Connection reset by peer
bluealsa: transport.c:907: Closing RFCOMM: 19
bluealsa: transport.c:350: Freeing transport: HFP Audio Gateway (N/A)
```

Arkq, 2018-10-19:
To be honest, dunno what it wrong... Try to disable HFP and enable A2DP only (`bluealsa -p a2dp-source`), because it looks like bluez misbehaviour. When I've got connection issues, almost always help removing device from bluez and pairing it once more.

ilya-fedin, 2018-10-19:
Yes, it works, thank you very much! I suffered a whole month :D 
I tried only `-p a2dp-sink`, but not` -p a2dp-source`. I thought, since speaker is a sink, then I need a2dp-sink :)

closed 2018-10-20 by Arkq