arkq/bluez-alsa #155: Couldn't get BlueALSA transport

deadeert, 2018-10-19:
Hello, 

I just install everything this morning so I should have all the last versions. 

`
$sudo bluealsa &  # starting manually 
$alsamixer -D bluealsa # is working but saying that no command are available for this device. 
$ aplay -D bluealsa:HCI=hci0,DEV=C0:28:HI:DD:EN:XX,PROFILE=a2dp /tmp/92645__cgeffex__organ-ending-tune.mp3
../../../src/asound/../shared/ctl-client.c:105: Connecting to socket: /var/run/bluealsa/hci0
../../../src/asound/../shared/ctl-client.c:219: Getting transport for C0:28:8D:0E:E7:F2 type 1
bluealsa: ../../src/ctl.c:638: Received new connection: 10
bluealsa: ../../src/ctl.c:651: New client accepted: 10
bluealsa: ../../src/ctl.c:675: +-+-
bluealsa: ../../src/ctl.c:675: +-+-
ALSA lib ../../../src/asound/bluealsa-pcm.c:666:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA transport: Aucun périphérique de ce type
aplay: main:828: erreur à l'ouverture audio: Aucun périphérique de ce type
bluealsa: ../../src/ctl.c:605: Client closed connection: 10
bluealsa: ../../src/ctl.c:675: +-+-
` 
I added the following in my dbus file  (dbus-1/system.d/bluetooth.conf):
` 
 <policy user='MYUSER'>
                <allow send_destination="org.bluez"/>
  </policy>
`

Regarding the same issues tickets I tried to switch to PROFILE to sco. Same problem..  

The request seems to be sent to bluealsa but it timeouts...

Any idea of what is going one ? 

Thanks.

beginner3456789, 2018-10-20:
I think I have a similar problem. I have used the bluez-alsa for about 6 months beginning with my build from sources and then from my voidlinux dist both without pulseaudio. I think the problem started after recently updating the alsalib from 1.1.6 to 1.1.7. The current 1.3.1_1 bluez-alsa voidlinux dist package and my latest github build from source connects to my bluetooth audio devices and then gives the following error shown with mplayer:
```

Opening audio decoder: [ffmpeg] FFmpeg/libavcodec audio decoders
AUDIO: 48000 Hz, 2 ch, floatle, 128.0 kbit/4.17% (ratio: 16000->384000)
Selected audio codec: [ffaac] afm: ffmpeg (FFmpeg AAC (MPEG-2/MPEG-4 Audio))
[AO_ALSA] alsa-lib: pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM bluealsa
[AO_ALSA] Playback open error: No such file or directory
Failed to initialize audio driver 'alsa:device=bluealsa'
Could not open/initialize audio device -> no sound.
Audio: no sound
Starting playback...
exit

```

deadeert, 2018-10-22:
Hi, thanks for these information.
Can you share your files 20-bluealsa.conf & dbus/.../bluetooth.conf please? 

thansk

commit dae06998 (2018-10-23): Move ALSA add-on config file to /etc/alsa/conf.d

With the release of alsa-lib v1.1.7, the location for add-on configs has
changed to /etc/alsa/conf.d/. For alsa <1.1.7 backwards compatibility is
preserved.

Closes #158 and fixes #155
closed 2018-10-23 by Arkq