arkq/bluez-alsa #157: arecord is wait forever to data

itamarlevit, 2018-10-21:
bluez-alsa is grate
I install it with buildroot on nxp imx6
the aplay is working perfect
but when use arecord it wait for data forever
Hear is log from bluez-alsa

**Log of bluez-audio**

connect 20:17:06:AB:48:15
Attempting to connect to 20:17:06:AB:48:15
[CHG] Device 20:17:06:AB:48:15 Connected: yes
[CHG] Device 20:17:06:AB:48:15 UUIDs: 00001108-0000-1000-8000-00805f9b34fb
[CHG] Device 20:17:06:AB:48:15 UUIDs: 0000110b-0000-1000-8000-00805f9b34fb
[CHG] Device 20:17:06:AB:48:15 UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[CHG] Device 20:17:06:AB:48:15 UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Device 20:17:06:AB:48:15 UUIDs: 0000111e-0000-1000-8000-00805f9b34fb
[CHG] Device 20:17:06:AB:48:15 ServicesResolved: yes
[CHG] Device 20:17:06:AB:48:15 Paired: yes
[Miracase BT106]# bluealsa: 128.255981014: bluez.c:882: Profile method call: org.bluez.Profile1.NewConnection()
bluealsa: 128.260865348: transport.c:707: State transition: 0 -> 2
bluealsa: 128.261442014: transport.c:122: Created new IO thread: SCO: HFP Audio Gateway
bluealsa: 128.261650014: bluez.c:821: HFP Audio Gateway configured for device 20:17:06:AB:48:15
bluealsa: 128.261765681: ctl.c:674: +-+-
bluealsa: 128.261896681: transport.c:707: State transition: 0 -> 2
bluealsa: 128.262106014: io.c:1382: Starting IO loop: HFP Audio Gateway
bluealsa: 128.262324681: rfcomm.c:573: Starting RFCOMM loop: HFP Audio Gateway
bluealsa: 128.262485348: transport.c:122: Created new IO thread: RFCOMM: HFP Audio Gateway
bluealsa: 128.262572348: transport.c:707: State transition: 2 -> 2
bluealsa: 128.276079348: at.c:160: AT message: SET: command:+BRSF, value:63
bluealsa: 128.276305014: rfcomm.c:96: Sending AT message: RESP: command:+BRSF, value:992
bluealsa: 128.276522348: rfcomm.c:96: Sending AT message: RESP: command:(null), value:OK
bluealsa: 128.276662681: rfcomm.c:114: HFP state transition: 0 -> 2
bluealsa: 128.297224348: at.c:160: AT message: TEST: command:+CIND, value:(null)
bluealsa: 128.297315348: rfcomm.c:96: Sending AT message: RESP: command:+CIND, value:("call",(0,1)),("callsetup",(0-3)),("service",(0-1)),("signal",(0-5)),("roam",(0-1)),("battchg",(0-5)),("callheld",(0-2))
bluealsa: 128.297442014: rfcomm.c:96: Sending AT message: RESP: command:(null), value:OK
bluealsa: 128.297554014: rfcomm.c:114: HFP state transition: 2 -> 5
bluealsa: 128.322274348: at.c:160: AT message: GET: command:+CIND, value:(null)
bluealsa: 128.322380348: rfcomm.c:96: Sending AT message: RESP: command:+CIND, value:0,0,0,0,0,0,0
bluealsa: 128.322491681: rfcomm.c:96: Sending AT message: RESP: command:(null), value:OK
bluealsa: 128.322603681: rfcomm.c:114: HFP state transition: 5 -> 7
bluealsa: 128.342274348: at.c:160: AT message: SET: command:+CMER, value:3, 0, 0, 1
bluealsa: 128.342368681: rfcomm.c:96: Sending AT message: RESP: command:(null), value:OK
bluealsa: 128.342478348: rfcomm.c:114: HFP state transition: 7 -> 8
bluealsa: 128.342557681: rfcomm.c:114: HFP state transition: 8 -> 9
bluealsa: 128.342630014: rfcomm.c:114: HFP state transition: 9 -> 13
bluealsa: 128.342796348: ctl.c:674: +-+-
bluealsa: 128.392169348: at.c:160: AT message: SET: command:+VGS, value:13
bluealsa: 128.392266348: rfcomm.c:96: Sending AT message: RESP: command:(null), value:OK
bluealsa: 128.392549348: ctl.c:674: +-+-
bluealsa: 128.413498681: at.c:160: AT message: GET: command:+BTRH, value:(null)
bluealsa: 128.413588348: rfcomm.c:96: Sending AT message: RESP: command:(null), value:OK
bluealsa: 128.518559681: at.c:160: AT message: SET: command:+XAPL, value:0000-0000-0100,7
bluealsa: 128.518749681: rfcomm.c:96: Sending AT message: RESP: command:(null), value:+XAPL=BlueALSA,0
bluealsa: 128.535282348: bluez.c:631: Endpoint method call: org.bluez.MediaEndpoint1.SelectConfiguration()
bluealsa: 128.560285014: bluez.c:631: Endpoint method call: org.bluez.MediaEndpoint1.SetConfiguration()
bluealsa: 128.561191014: bluez.c:563: A2DP Source (SBC) configured for device 20:17:06:AB:48:15
bluealsa: 128.561296681: bluez.c:565: Configuration: channels: 2, sampling: 48000
bluealsa: 128.561391014: transport.c:707: State transition: 0 -> 0
bluealsa: 128.561522348: ctl.c:674: +-+-
bluealsa: 128.562039681: bluez.c:681: Endpoint already registered: /A2DP/MPEG24/Source/1
bluealsa: 128.562148681: bluez.c:698: Registering endpoint: /A2DP/SBC/Source/2
Connection successful
[CHG] Device C3:BB:63:1B:07:05 RSSI: -74
[Miracase BT106]# bluealsa: 133.528052348: at.c:160: AT message: SET: command:+IPHONEACCEV, value:2,1,9,2,0
bluealsa: 133.528275682: rfcomm.c:96: Sending AT message: RESP: command:(null), value:OK
bluealsa: 133.528426015: ctl.c:674: +-+-
bluealsa: 133.543136015: at.c:160: AT message: CMD: command:+CLCC, value:(null)
bluealsa: 133.543222348: Unsupported AT message: CMD: command:+CLCC, value:(null)
bluealsa: 133.543295015: rfcomm.c:96: Sending AT message: RESP: command:(null), value:ERROR
bluealsa: 133.558176015: at.c:160: AT message: SET: command:+CSRSF, value:0,0,0,1,0,0,0
bluealsa: 133.558265348: Unsupported AT message: SET: command:+CSRSF, value:0,0,0,1,0,0,0
bluealsa: 133.558337682: rfcomm.c:96: Sending AT message: RESP: command:(null), value:ERROR
[Miracase BT106]# exit

**arecord -D bluealsa:HCI=hci0,DEV=20:17:06:AB:48:15,PROFILE=sco /test.wav**

220.280717359: ../shared/ctl-client.c:105: Connecting to socket: /var/run/bluealsa/hci0
220.285050359: ../shared/ctl-client.c:219: Getting transport for 20:17:06:AB:48:15 type 2
bluealsa: 220.285142692: ctl.c:637: Received new connection: 17
bluealsa: 220.285239359: ctl.c:650: New client accepted: 17
bluealsa: 220.285280692: ctl.c:674: +-+-
bluealsa: 220.285418359: ctl.c:674: +-+-
220.295459359: bluealsa-pcm.c:546: Setting constraints
Recording WAVE '/test.wav' : Unsigned 8 bit, Rate 8000 Hz, Mono
220.306934692: bluealsa-pcm.c:294: Initializing HW
220.307038692: ../shared/ctl-client.c:375: Requesting PCM open for 20:17:06:AB:48:15
bluealsa: 220.307189025: ctl.c:352: PCM requested for 20:17:06:AB:48:15 type 2 stream 1
bluealsa: 220.307425025: 220.307420692: ctl.c:674: +-+-
bluealsa-pcm.c:323: Selected HW buffer: 4 periods x 2000 bytes == 8000 bytes
220.307499692: bluealsa-pcm.c:338: Initializing SW
220.307653359: bluealsa-pcm.c:338: Initializing SW
220.307713025: bluealsa-pcm.c:338: Initializing SW
220.307760025: bluealsa-pcm.c:354: Prepared
220.307823359: bluealsa-pcm.c:338: Initializing SW
220.309924359: bluealsa-pcm.c:226: Starting
220.309979692: ../shared/ctl-client.c:443: Requesting PCM resume for 20:17:06:AB:48:15
bluealsa: 220.310126692: transport.c:707: State transition: 2 -> 2
bluealsa: 220.310196025: ctl.c:674: +-+-
bluealsa: 220.802160025: transport.c:968: New SCO link: 18 (MTU: R:48 W:48)

**after few seconds**
bluealsa: 229.546221360: at.c:160: AT message: SET: command:+IPHONEACCEV, value:2,1,8,2,0
bluealsa: 229.546373693: rfcomm.c:96: Sending AT message: RESP: command:(null), value:OK
bluealsa: 229.546518360: ctl.c:674: +-+-
bluealsa: 229.560077693: at.c:160: AT message: SET: command:+IPHONEACCEV, value:2,1,8,2,0
bluealsa: 229.560203026: rfcomm.c:96: Sending AT message: RESP: command:(null), value:OK
bluealsa: 229.560337360: ctl.c:674: +-+-

**btmon log**
HCI Event: Synchronous Connect Compl.. (0x2c) plen 17 #531 [hci0] 434.692418
Status: Success (0x00)
Handle: 257
Address: 20:17:06:AB:48:15 (OUI 20-17-06)
Link type: eSCO (0x02)
Transmission interval: 0x0c
Retransmission window: 0x02
RX packet length: 60
TX packet length: 60
Air mode: CVSD (0x02)
-> Sop hear.

Arkq, 2019-11-04:
This issue is most likely resolved with the latest [commit](https://github.com/Arkq/bluez-alsa/commit/55c8aec1296f0756de8ae6ac334961308e9b119d) in the master. Please see whether it works for you.

closed 2019-11-04 by Arkq

itamarlevit, 2019-11-05:
Hi Arkq/bluez-alsa
Many Thanks
The problem was in the TI chip setup.
Regards, Itamar