arkq/bluez-alsa #161: No SCO profiles availables with any phones

tbultel, 2018-10-28:
Hi,
First, I have to say that on the embedded system I am working on, I am using a very old version of bluez (5.46), and that the version of bluez-alsa I have dates from March 2018...
Bumping will be done in the coming weeks but not possible at the present time.

The problem I have, is that I am not able to see any available SCO transport, with the connected
mobile phones I have been using (3 different with no better result)

I have started to dig in the code (both bluez and bluez-alsa), but if anyone has an idea of what 
could make that happen, this will save me a lot of time.

If have seen in the bluez documentation that SCO profile is provided by ofono, and yes, the ofono
service is running on the system

Best regards
Thierry Bultel

Arkq, 2018-11-06:
I assume that "Hands-Free" or/and "Headset Audio Gateway" profiles are enabled in bluealsa. Then check bluealsa logs for any information (compile bluez-alsa with `--enable-debug`). If there are no logs related to RFCOMM (you should see debug messages related to RFCOMM communication), then something is wrong on the bluez level.

tbultel, 2018-11-06:
Hi Arkadiuz,

bumping bluez-alsa to master, and bluez to 5.50, has solved almost all 
the problems I had.
It is worth to say that I got confusion due to the output of

amixer -D bluealsa

Since the list of transport is filtered with this condition (I assume 
that there must be side effects to remove it):

if (t->type == TRANSPORT_TYPE_SCO && t->codec == HFP_CODEC_UNDEFINED) {
       continue;
}
... there is a state at startup where SCO transports are not displayed. 
Making a first phone call makes them appear.

However, despite of having tried with 3 different androit phones, and a 
iPhone 6S, I have not been able to use SCO with hsp-hs only.

When enabling that profile only, the connection fails in bluetoothctl.
To have a connection, I have to one of hfp-hf and a2dp-sink or both.
Could be be due to the fact that this profile is a kind of deprecated on 
recent phones ?

Thanks !
Thierry

Arkq, 2018-11-06:
> Since the list of transport is filtered with this condition (I assume that there must be side effects to remove it). Making a first phone call makes them appear.

The side effect is, that for HFP profile codec is not selected yet, so it is impossible to tell the sampling rate. So, technically there is no transport... It is important for mSBC (not in the master), because SBC is 8 kHz, while mSBC is 16 kHz.

> However, despite of having tried with 3 different androit phones, and a iPhone 6S, I have not been able to use SCO with hsp-hs only. Could be be due to the fact that this profile is a kind of deprecated on recent phones ?

Yes, I'm aware of that, I'm not sure if Android supports HSP at all. It is quite unfortunate, because this profile is MUCH more simple and actually works in bluealsa. HFP requires a lot of "gymnastic" due to tight coupling SCO with RFCOMM and it does not work reliably (that's why it's not mentioned in the README). Making it production ready is on my todo list :)

tbultel, 2018-11-09:
Hi Arkadiuz,

> Since the list of transport is filtered with this condition (I
>     assume that there must be side effects to remove it). Making a
>     first phone call makes them appear.
>
> The side effect is, that [...]
I understand, but then, is there a way to be notified when the codec 
becomes available ? Else I do not see how to prevent to miss the first 
phone call.
> However, despite of having tried with 3 different androit phones,
>     and a iPhone 6S, I have not been able to use SCO with hsp-hs only.
>     Could be be due to the fact that this profile is a kind [...]

Yes, I have seen that. And the problem is that on my system I have Ofono 
running, that registers itself as a HFP handler in bluez.
I thus have some cases (depending on the start order I guess) where 
bluez-alsa won't successfully register to bluez, because
the role is stolen by Ofono. Are you aware of that already ?

Arkq, 2018-12-11:
> I understand, but then, is there a way to be notified when the codec becomes available ? Else I do not see how to prevent to miss the first phone call.

When codec is selected, the `BA_EVENT_TRANSPORT_CHANGED` event should be emitted, so client can subscribe for it (it's best to subscribe for `BA_EVENT_TRANSPORT_ADDED` as well). However, after some thinking, I see that the event system needs modifications. First of all some metadata should be transferred as well (e.g. for which transport given event was emitted). Also, the `CHANGED` event might be somehow misleading, because when codec is not selected, such transport is not visible... I'll try to refactor this event notifications system.

tbultel, 2018-12-12:
> I understand, but then, is there a way to be notified when the
>     codec becomes available ? Else I do not see how to prevent to miss
>     the first phone call.
>
> When codec is selected, the [...]
Cool,
If |BA_EVENT_TRANSPORT_ADDED |holds transport info, that would help the 
client apps in their logic to stop pcm loops when transports disappear.

Arkq, 2018-12-15:
Like I've said, I've changed a little bit event system. Right now, it will be possible to know for which transport given event has been emitted.

Also, I'm closing this issue, since right now it is possible to use HFP via oFono. I'll try to fix internal HFP anyway :) however with some low priority, so it might take some time...

closed 2018-12-15 by Arkq

tbultel, 2018-12-21:
Thanks ! Sorry for the delay.

I will eventually rebase to your version in the coming weeks, since we 
are now in a test phase before the CES demo.

bluez-alsa is now completely part of the audio subsystem of AGL:

https://git.automotivelinux.org/AGL/meta-agl/commit/?id=a46a78ad5f2baa90127808ad1c4d025e4b4a0e77

If it is of interest to you, please have a look at: 
https://github.com/iotbzh/bluez-alsa/tree/bluealsalib
It is 3 commits ahead to your master, I have taken the existing PR for 
having a shared library (was mandatory for our project),
plus the increased number of possible client connections, just like we 
said in another mail.

Talk you soon
Thierry