arkq/bluez-alsa #162: bluez-alsa not connecting

WilliamVJacob, 2018-10-29:
hi , I am trying to play Bluetooth audio on my Linux board but its not working properly as i get error incorrect pin or passcode please try later .i will share some more information below

- I am using buildroot
-I installed bluez5-utils and I am able connect my mobile bt address using bluetoothctl tool 
-i installed bluez-alsa  and i am able to see  bluealsa  bluealsa-aplay but i am not able to configure bluez -alsa  to bluez5
- my Bluetooth configuration files are located in /etc/Bluetooth/

if you need more info you can ask me

WilliamVJacob, 2018-10-30:
@Arkq  - I am sharing the logs below
# bluetoothctl
[NEW] Controller 0C:8C:24:67:C0:CB WIlliam-SPEAKER [default]
[NEW] Device D8:32:E3:7B:31:76 William V Jacob
[bluetooth]# power on
Changing power on succeeded
[bluetooth]# discoverable on
Changing discoverable on succeeded
[bluetooth]# pcirable on
Changing pairable on succeeded
[bluetooth]# agent NoMnputNoOutput
Agent registered
[bluetooth]# default-agent
Default agent request successful
[NEW] Device BC:54:51:72:BD:BB Galaxy J3(2017)
Request authorization
[agent] Accept pairing (yes/no): yes
[CHG] Device BC:54:51:72:BD:BB Modalias: bluetooth:v0075p0100d0200
[CHG] Device BC:54:51:72:BD:BB UUIDs: 00001105-0000-1000-8000-00805f9b34fb
[CHG] Device BC:54:51:72:BD:BB UUIDs: 0000110a-0000-1000-8000-00805f9b34fb
[CHG] Device BC:54:51:72:BD:BB UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[CHG] Device BC:54:51:72:BD:BB UUIDs: 00001112-0000-1000-8000-00805f9b34fb
[CHG] Device BC:54:51:72:BD:BB UUIDs: 00001115-0000-1000-8000-00805f9b34fb
[CHG] Device BC:54:51:72:BD:BB UUIDs: 00001116-0000-1000-8000-00805f9b34fb
[CHG] Device BC:54:51:72:BD:BB UUIDs: 0000111f-0000-1000-8000-00805f9b34fb
[CHG] Device BC:54:51:72:BD:BB UUIDs: 0000112f-0000-1000-8000-00805f9b34fb
[CHG] Device BC:54:51:72:BD:BB UUIDs: 00001132-0000-1000-8000-00805f9b34fb
[CHG] Device BC:54:51:72:BD:BB UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Device BC:54:51:72:BD:BB UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Device BC:54:51:72:BD:BB UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Device BC:54:51:72:BD:BB Paired: yes
[CHG] Device BC:54:51:72:BD:BB Connected: no
[CHG] Device D8:32:E3:7B:31:76 Connected: yes
Request authorization
[agent] Accept pairing (yes/no): yes
[CHG] Device D8:32:E3:7B:31:76 Connected: no
[bluetooth]# connect D8:32:E3:7B:31:76
Attempting to connect to D8:32:E3:7B:31:76
Failed to connect: org.bluez.Error.Failed

Pro-pra, 2018-10-30:
bluetoothctl not a part of bluez-alsa.
You need send sound to Galaxy J3 phone?

My issue:
1. Pair device (you do it)
2. run bluealsa daemon with root privelegies
3. connect to device

WilliamVJacob, 2018-10-30:
yes I know bluetoothctl is not a part of blue-alsa , its part a bluez5 . I just wanted to show the log
my bluealsa is already running  in the background  ,it is still giving me error
can you guide me on how to configure bluealsa  and I  also want to know how to check logs in bluealsa

Pro-pra, 2018-10-30:
> my bluealsa is already running in the background ,it is still giving me error

kilall bluealsa and launch from therminal with root privelegies.

WilliamVJacob, 2018-10-31:
No these things I already tried , its not working
I will share more logs for better understanding 
# ps -ef | grep blue
 1725 root     grep blue
# bluealsa -i hci0
bluealsa: Couldn't initialize controller thread: Bad file descriptor
# bluealsa -h
Usage:
  bluealsa [OPTION]...

Options:
  -h, --help            print this help and exit
  -V, --version         print version and exit
  -S, --syslog          send output to syslog
  -i, --device=hciX     HCI device to use
  -p, --profile=NAME    enable BT profile
  --a2dp-force-mono     force monophonic sound
  --a2dp-force-audio-cd force 44.1 kHz sampling
  --a2dp-volume         control volume natively

Available BT profiles:
  - a2dp-source Advanced Audio Source (SBC)
  - a2dp-sink   Advanced Audio Sink (SBC)
  - hfp-hf      Hands-Free (v1.7)
  - hfp-ag      Hands-Free Audio Gateway (v1.7)
  - hsp-hs      Headset (v1.2)
  - hsp-ag      Headset Audio Gateway (v1.2)

By default only output profiles are enabled, which includes A2DP Source and
HSP/HFP Audio Gateways. If one wants to enable other set of profiles, it is
required to explicitly specify all of them using `-p NAME` options.
# bluealsa --a2dp-force-audio-cd
bluealsa: Couldn't initialize controller thread: Bad file descriptor
# bluealsa --device=hci0
bluealsa: Couldn't initialize controller thread: Bad file descriptor
# ps -ef | grep blue
 1831 root     grep blue
# bluealsa
bluealsa: Couldn't initialize controller thread: Bad file descriptor
# bluealsa &
# bluealsa: Couldn't initialize controller thread: Bad file descriptor

[1]+  Done(1)                    bluealsa
#
#
#
# ps -ef | grep blue
 1859 root     grep blue
# bluealsa

Pro-pra, 2018-11-01:
https://github.com/Arkq/bluez-alsa/issues/149
may be help you.

Arkq, 2018-12-15:
Now bluealsa reports connection issue more verbosely. If you've still got connection problem, please reopen this issue with logs from bluealsa server (note, that there can be only one bluealsa server per hci).

closed 2018-12-15 by Arkq