arkq/bluez-alsa #165: bluez-alsa works only 1-2 times

adrianmihalko, 2018-11-04:
For some reason my bluez-alsa is very unstable, I can play something once, then hangs:

```
madrian@ubuntudev:~$ sudo aplay -D bluealsa test.wav
Playing WAVE 'test.wav' : Signed 16 bit Little Endian, Rate 22050 Hz, Mono
madrian@ubuntudev:~$ sudo aplay -D bluealsa test.wav
Playing WAVE 'test.wav' : Signed 16 bit Little Endian, Rate 22050 Hz, Mon

^CAborted by signal Interrupt...
^C
madrian@ubuntudev:~$ sudo systemctl status bluealsa.service
● bluealsa.service - BluezAlsa proxy
   Loaded: loaded (/lib/systemd/system/bluealsa.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2018-11-04 21:04:03 CET; 3min 12s ago
 Main PID: 29786 (bluealsa)
    Tasks: 4 (limit: 1111)
   CGroup: /system.slice/bluealsa.service
           └─29786 /usr/bin/bluealsa -S -p a2dp-source

Nov 04 21:04:03 ubuntudev systemd[1]: Started BluezAlsa proxy.

```

Bluezalsa is running, bluetoothctl is connected. It just won't play anything.

Arkq, 2018-11-06:
Compile bluez-alsa with `--enable-debug` and post here logs from the bluealsa.service.

Arkq, 2020-01-25:
Please try new version. If the issue still happens, reopen with more details.

closed 2020-01-25 by Arkq