arkq/bluez-alsa #166: Near immediate detection of play and pause hook - convenience switching

kiwi-ed, 2018-11-05:
Hi,
I have two devices sharing the audio card and both need to convenience switch, i.e pressing play on one pauses the other, releases its sink/card and then the "playing" one grabs the sink and renders; and visa-versa. Everything works except catching the pause / play signals Bluetooth side - problem is it's just too slow. Regarding the Bluetooth side I'm currently monitoring the transport state change to detect pause/play but with A2DP the transport state changes only when the streaming application is minimised, terminated or the music has been paused for ages going idle. If the transport is in idle then going from pause->play will immediately change the state to active - that's fine.

Admittedly there is no need to have incoming DBus pause / play signals as we're only gating an active pipe and under normal conditions that's OK. 

So in summary, I need a more reliable way to detect the point of pausing bluealsa side without relying on transport states.

Thanks.

Arkq, 2018-11-06:
I'm rather confused about this issue :) Are you using bluealsa as a bluetooth sink (you are streaming sound from the external BT device, e.g. phone), or as a source (you are streaming audio from the host to the external BT device, e.g. speaker). In the first case, it is rather not possible to detect whether audio is paused, because transport state is not related to the playback state. You have to rely on the AVRCP. In the second case you should use a mixer which will allow more then one application to share single audio device. Unfortunately, there is no mixer in the bluez-alsa project (not yet), and dmix ALSA plugin won't work....

kiwi-ed, 2018-11-07:
Hi,
Bluealsa is the sink, i.e. phone streams to it. The issue is mainly with iOS. Many iOS applications don't and don't necessarily need to provide all the AVRCP suite and after lots of experimentation that appears to be the case. The few Android apps I tested however will DBus signal play, pause and all sorts of things bluealsa side.  
So my problem is really iOS I can't see pause and pause signals only transports states so if I can force the transport bluez side to idle - problem solved! Pressing "play" on the iOS side again sets it active once more. 
1) Can the transport be forced to idle bluez side? 
2) Regarding a mixer, is there a mixer that will in no way shape or form alter the incoming stream i.e not change bit-rates, won't re-sample etc? It has to be just like using hw0:0.  

This question is not a bug just desperate for some answers.
Thanks.

unisound-ivm, 2019-04-09:
hi，kiwi
     Do you resolve the issue?
thanks

kiwi-ed, 2019-04-09:
Hi,

To a point yes. In the end I had to accept the iOS Youtube app doesn't release the transport until minimised or stopped for around a minute.  Other iOS apps are much quicker and most send the  play / pause AVRCP signals I'm after. An update of Bluez also helped. So in all.. it OK. Appreciated your help.

Regards,

Edward

Disign Consultants
t: +44 (0) 207 544 8486
w: www.disignconsultants.com

This message and any attachment are confidential and may be privileged or otherwise protected from disclosure. If you are not the intended recipient, please telephone or email the sender and delete this message and any attachment from your system. If you are not the intended recipient you must not copy this message or attachment or disclose the contents to any other person.

SuperSam56, 2019-05-01:
Hi. I just read this issue thread. I'm wondering about the remark from @Arkq in case of bluetooth sink scenario: "You have to rely on the AVRCP". Could you please give more details on how to do? because i would like to catch song name (and other metadata) played from the external BT source device directly from bluez-alsa.
Is it possible to do so ?

kiwi-ed, 2019-05-01:
Hi SS56,

It's a matter of spying and sending DBUS messages on the interface for AVRCP. I'm a bit rusty, but here's some info. So connect up something to stream BT audio and have a play. Start with something like  

----------------------------------------------------------------------------------------------------------------------------------
This Linux terminal command sends a Pause to my connected device i.e. iPad connected and playing music. 
MediaPlayer1 is the AVRCP interface on my board.

dbus-send --system --print-reply --dest=org.bluez /org/bluez/hci0/dev_A4_E9_75_CD_87_85/player2 org.bluez.MediaPlayer1.Pause
 
----------------------------------------------------------------------------------------------------------------------------------

Bit of Python:
from __future__ import print_function
import os
import sys
import dbus
import dbus.service
import dbus.mainloop.glib
try:
	from gi.repository import GObject
except ImportError:
	import gobject as GObject

import bluezutils

if __name__ == '__main__':
	dbus.mainloop.glib.DBusGMainLoop(set_as_default=True)

        print ("OK")                                                                             
        bus = dbus.SystemBus()                                             
        obj="/org/bluez/hci0/dev_A4_E9_75_CD_87_85/player2"                           
        mp = dbus.Interface(bus.get_object("org.bluez", obj),"org.bluez.MediaPlayer1")           
        #prop = dbus.Interface(bus.get_object("org.bluez", obj),"org.freedesktop.DBus.Properties")
        #properties = prop.Get("org.bluez.MediaPlayer1","Status")        
        #properties = prop.GetAll("org.bluez.MediaPlayer1")           
        #print("Properties are %s",properties)                                
        mp.Pause() 

-----------------------------------------------------------------------------------------------------------------------------------

Here's a link for a Python player that lets control your table using AVRCP commands. It's got stuff like Status, Track etc.
https://github.com/Douglas6/blueplayer/blob/master/blueplayer.py

-----------------------------------------------------------------------------------------------------------------------------------
Simple-player.py

					"xesam:title" : "Title",
					"xesam:artist" : "Artist",
					"xesam:album" : "Album",
					"xesam:genre" : "Genre",
					"xesam:trackNumber" : dbus.Int32(1),
					"mpris:length" : dbus.Int64(10000) },

infirit, 2019-05-01:
People should move to GI and use Gio.DBusProxy. Drop old and pretty much dead dbus-python.

kiwi-ed, 2019-05-01:
Yea, bit tougher on headless/displayless stuff like BT speakers like mine.. Joys of Embedded Linux.

closed 2020-10-27 by kiwi-ed