arkq/bluez-alsa #168: bluealsa: Couldn't obtain D-Bus connection error

dennishong33, 2018-11-07:
I build the as this document,
But met the following error when i run the bluealsa
/ # bluealsa &
/ # bluealsa: Couldn't obtain D-Bus connection: Could not connect: No such file or directory

Dbus library already verified using bluetoothd.
bluetoothctl can be scan, pair with bluetoothd in bluez5.

version info.
alsa-lib-1.0.28
libsndfile-1.0.28
sbc-1.3
bluez-alsa-1.3.1
Dbus-1.8.0
bluez-5.18

please help me!! 
Thank you.

dennishong33, 2018-11-07:
Now I build the Dbus-1.12.10, but it is not working.
I don't know why this error.

code is below,
address = g_dbus_address_get_for_bus_sync(G_BUS_TYPE_SYSTEM, NULL, NULL);
	if ((config.dbus = g_dbus_connection_new_for_address_sync(address,
					G_DBUS_CONNECTION_FLAGS_AUTHENTICATION_CLIENT |
					G_DBUS_CONNECTION_FLAGS_MESSAGE_BUS_CONNECTION,
					NULL, NULL, &err)) == NULL) {
		error("Couldn't obtain D-Bus connection: %s", err->message);
		return EXIT_FAILURE;
	}

dennishong33, 2018-11-12:
I dont't understand, why happen this error? 
Please let me know how to resolve this issue or need to checking?

Thank you.

Arkq, 2018-11-12:
Sorry, but I have no idea what might be wrong with your setup... It seems like there is no system bus on your host, however since bluetoothctl works, so D-Bus is working. Strange misbehavior...

dennishong33, 2018-11-13:
Thanks, reply,

My environment is embedded system,
However to run the bluealsa need the udevd?

quantenschaum, 2019-04-08:
Same here on Ubuntu 18.04. Compiled and installed as described, when running `bluealsa` I get

```
bluealsa: ../../src/ctl.c:513: Starting controller loop: hci0
bluealsa: ../../src/bluez.c:800: Registering endpoint: /org/bluez/hci0/A2DP/MPEG24/Source/1
bluealsa: ../../src/bluez.c:800: Registering endpoint: /org/bluez/hci0/A2DP/SBC/Source/1
bluealsa: ../../src/bluez.c:1064: Registering profile: /org/bluez/HSP/AudioGateway
bluealsa: ../../src/bluez.c:1064: Registering profile: /org/bluez/HFP/AudioGateway
bluealsa: ../../src/main.c:331: Acquiring D-Bus service name: org.bluealsa
bluealsa: ../../src/main.c:336: Starting main dispatching loop
bluealsa: Couldn't acquire D-Bus name: org.bluealsa
bluealsa: ../../src/main.c:340: Exiting main loop
bluealsa: ../../src/ctl.c:632: +-+-
bluealsa: ../../src/ctl.c:594: Received new connection: -1
```

system dbus is running and `bluetoothctl` is working

```
#systemctl status dbus
* dbus.service - D-Bus System Message Bus
   Loaded: loaded (/lib/systemd/system/dbus.service; static; vendor preset: enabled)
   Active: active (running) since Mon 2019-04-08 13:45:52 CEST; 29min ago
     Docs: man:dbus-daemon(1)
 Main PID: 665 (dbus-daemon)
    Tasks: 1 (limit: 4467)
   CGroup: /system.slice/dbus.service
           `-665 /usr/bin/dbus-daemon --system --address=systemd: --nofork --nopidfile --systemd-activation --syslog-only
```

abourget, 2019-04-13:
I get this on Bionic, with latest commit here

```
./build/src/bluealsa: Couldn't bind controller socket: /var/run/bluealsa/hci0: Permission denied
./build/src/bluealsa: ../../src/bluez.c:1064: Registering profile: /org/bluez/HSP/AudioGateway
./build/src/bluealsa: ../../src/bluez.c:1064: Registering profile: /org/bluez/HFP/AudioGateway
./build/src/bluealsa: ../../src/main.c:331: Acquiring D-Bus service name: org.bluealsa
./build/src/bluealsa: ../../src/main.c:336: Starting main dispatching loop
./build/src/bluealsa: Couldn't acquire D-Bus name: org.bluealsa
./build/src/bluealsa: ../../src/main.c:340: Exiting main loop
```

dbus running also.

gearhead, 2019-04-16:
Check the user value in the bluealsa.service file. In some versions of systemd, I have had to comment it out to get it to run.

quantenschaum, 2019-04-24:
I started bluealsa "manually" from the command line as root and get this error.

quantenschaum, 2019-04-24:
What systemd service file are you talking about? There is no service file.

quantenschaum, 2019-04-24:
The error disappeared after pulling then recent changes.

OtherCrashOverride, 2021-04-16:
I encountered this also.  The cause of the issue was that a required file was not present:
/etc/dbus-1/system.d/bluealsa.conf

There appears to be files that are missing when using "configure" with the "--prefix" option.  I do not know why certain files fail to get installed.

borine, 2021-04-23:
> The cause of the issue was that a required file was not present

Are you perhaps cross-compiling, or otherwise installing into a different sysroot ? I ask because the bluez-alsa configure script will by default use `pkg-config` to find the install location for d-bus configuration files, completely ignoring the --prefix option (unless you are using a very old version of bluez-alsa) . For most systems, d-bus uses `/etc/dbus-1/system.d/`.
 
For installing into a sysroot that is different from the build system root, you _can_ use --prefix, but then you most likely also need to override the install paths for external package config files such as d-bus, and possibly also alsa. The bluez-alsa configure script provides options for this: `--with-dbusconfdir=DIR --with-alsaplugindir=DIR --with-alsaconfigdir=DIR`

Alternatively, you can configure using default paths in the normal way (i.e. do not use `--prefix`), but then install with `make install DESTDIR=sysroot` which will place the whole install tree under sysroot.

OtherCrashOverride, 2021-05-02:
Yes, I was compiling as part of a build script (different sysroot) for packaging as a deb.  When compiling directly on a target, everything works.

It is just mystifying when "root" is denied permission.  Since this post was the first hit when researching the issue, I thought I would leave the relevant info here to help others that may encounter it.

borine, 2021-06-04:
Issue resolved by installing missing file as said in comment https://github.com/Arkq/bluez-alsa/issues/168#issuecomment-820951807 

With D-Bus, even `root` needs explicit permission from the D-Bus config to exchange messages with a service. That is normal, and not special to bluealsa. Installing the missing file granted that permission to root.

closed 2021-06-04 by borine

zlk12142, 2025-07-22:
My dbus is installed in /opt/bluez instead of the default path.
Therefore, an environment variable called DBUS_SYSTEM_BUS_ADDRESS must be set according to the DBUS_SYSTEM_BUS_DEFAULT_ADDRESS value in config.log from the compiled dbus-1.15.8 source folder.

export DBUS_SYSTEM_BUS_ADDRESS=unix:path=/opt/bluez/var/run/dbus/system_bus_socket
/opt/bluez-alsa/bin/bluealsa -p a2dp-source -p hsp-ag &