arkq/bluez-alsa #170: Speech Recording Failed when RPi acts as HFP source

lordet01, 2018-11-14:
Hi. Thank you for your great work! I couldn't make my own bluetooth speaker with RPi unless you published this handful project.

Currently, I am making Bluetooth speaker having 2 MEMS microphones for distant phone call with it.
my RPi-based prototype successfully paired with my iPhone7 as hands-free device which supports both HFP and A2DP like other commercial bluetooth speakers. However, recording using arecord from the device through the phone is failed, while aplay works perfectly. Is the recording functionality is not yet implemented in BlueALSA? or I am missing something? Some advice would tremendously be helpful. Best Regards!

Arkq, 2018-11-16:
HFP is not working correctly. This issue might be related to #161. In the near future there might be a lot of changes in the HFP area, so stay tuned and be patient :)

lryindra, 2018-12-11:
set the following command to record from Raspberry Pi's bluetooth:
/usr/bin/hcitool cmd 0x3F 0x01C 0x01 0x02 0x00 0x01 0x01

lordet01, 2018-12-20:
Now I solved my issue with hfp-ofono protocol which has committed recently. Thanks again for your dedication on this project.

closed 2018-12-20 by lordet01