arkq/bluez-alsa #175: Can't get it working with cmus

ciechomke, 2018-12-10:
I tried to configure cmus to work with bluealsa. My ~/.asoundrc:
```
defaults.pcm.!card PCH
defaults.pcm.!device 0
pcm.bluetooth {
        type plug
        slave.pcm {
                type bluealsa
                device "F8:DF:15:18:97:E4"
                profile "a2dp"
        }
        hint {
                show on
                description "JBL Reflect Contour 2 headphones"
        }
}
```
My default card is audio jack.
aplay -D bluetooth works fine

cmus has some options to select default output device:
```
~/.config/cmus/rc
set output_plugin=alsa
set dsp.alsa.device=default
set mixer.alsa.device=default
set mixer.alsa.channel=Master```

I've tried setting dsp.alsa.device and mixer.alsa.device to various values (bluealsa, bluetooth headset name, bluetooth device MAC) but none of them works.

@Arkq you probably use cmus (as noticed in your contributions) can you give me a little help with this?

Arkq, 2018-12-11:
My configuration is as follows:

$ cat .asoundrc
```
pcm.jabra {
	type plug
	slave.pcm {
		type bluealsa
		device "1C:48:F9:9D:81:5C"
		profile "a2dp"
		delay 20000
	}
	hint {
		show on
		description "Jabra MOVE v2.3.0"
	}
}
```
$ grep alsa ~/.config/cmus/autosave
```
set dsp.alsa.device=jabra
set mixer.alsa.channel=PCM
set mixer.alsa.device=default
set output_plugin=alsa
```

ciechomke, 2018-12-11:
Interesting. I've adjusted my configuration and now when I start playing music in cmus I hear that something is going on on headphones (I don't know how to describe it, it is similar sound when somebody connects to bluetooth and plays silence), and after few seconds cmus freezes.

Arkq, 2018-12-11:
Make sure you are not affected by the ALSA thread-safe API issue - see the [troubleshouting](https://github.com/Arkq/bluez-alsa#troubleshooting) section.

Otherwise, post versions of cmus and alsa-lib.

ciechomke, 2018-12-11:
Oh I forgot about it! Thank you dude!
Pozdrawiam :)

closed 2018-12-11 by ciechomke