arkq/bluez-alsa #176: Unknown PCM error returned

lryindra, 2018-12-11:
I have bluez-alsa working on a ubuntu 16.04 LTS where Libasound2 and alsa-utils are version 1.1.0-0

I got the following error when I setup the same config on Raspbian updated to buster release where libasound2 and alsa-utils are version 1.1.7. (also there is no alsa application available for this release):
$ aplay -D bluealsa ~/Music/piano2.wav
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM bluealsa
aplay: main:828: audio open error: No such file or directory

bluez-alsa on the buster release of Rasbian does work when I call
$ aplay -D btheadset ~/Music/piano2.wav

Here is where btheadset is defined in ~/.asoundrc:
	pcm.btheadset {
	   type plug
	   slave {
	       pcm {
	           type bluealsa
	           device XX:XX:XX:XX:XX:XX    <------replace with headphone's real MAC
	           profile "sco"
	       }   
	   }   
	   hint {
	       show on
	       description "BT Headset"
	   }   
	}
	ctl.btheadset {
	  type bluealsa
	}

lryindra, 2019-01-07:
I don't know why bluealsa is not in the pcm list (aplay --list-pcm), but this updated .asoundrc is working enough:

defaults.bluealsa.interface "hci0"
defaults.bluealsa.device "XX:XX:XX:XX:XX:XX"
defaults.bluealsa.profile "sco"
defaults.bluealsa.delay 10000

pcm.bluealsa {
  @args [HCI DEV PROFILE]
  @args.HCI {
    type string
    default "hci0"
  }
  @args.DEV {
    type string
    default "XX:XX:XX:XX:XX:XX"
  }
  @args.PROFILE {
    type string
    default "sco"
  }
  type plug
  slave {
    pcm {
      type bluealsa
      device $DEV
      profile "sco"
    }
  }
  hint {
    show on
    description "bluealsa"
  }
}
ctl.bluealsa {
  type bluealsa
}
pcm.btheadset {
  @args [DEV]
  @args.DEV {
    type string
    default "XX:XX:XX:XX:XX:XX"
  }
  type plug
  slave {
    pcm {
      type bluealsa
      device $DEV
      profile "sco"
    }
  }
  hint {
    show on
    description "BT Headset"
  }
}
ctl.btheadset {
  type bluealsa
}

leonidsah, 2021-03-23:
Thank you so much, I love you

borine, 2021-03-23:
See https://github.com/Arkq/bluez-alsa/wiki/Using-the-bluealsa-ALSA-pcm-plugin#defining-bluealsa-pcms for advice on writing .asoundrc files for use with bluealsa - in particular the note at the end of that section on not using `pcm.bluealsa` as a pcm name.

closed 2021-06-04 by borine